thesofproject/sof #3638: [BUG] Create Mux component fail

libinyang, 2020-11-24:
**Describe the bug**
When driver sent IPC to create a mux component, FW will fail. The sof-logger output:
[  3865788.750000] (        6.770833) c0 mux                     src/audio/mux/mux.c:93   ERROR mux_set_values(): mux component is not able to mix channels
[  3865804.635417] (       15.885417) c0 demux                 src/audio/component.c:135  ERROR comp_new(): unable to create the new component
[  3865811.666667] (        7.031250) c0 ipc                           src/ipc/ipc.c:186  ERROR ipc_comp_new(): component cd = NULL
[  3865818.958333] (        7.291667) c0 ipc                       src/ipc/handler.c:1131 ERROR ipc: pipe 10 comp 51 creation failed -22

**Reproduction Rate**
100%

**Expected behavior**
FW should create the mux component successfully.

**Impact**
Audio driver failed to probe

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev: 30629f3e8131072b3f97eb8ed3c27f7838ff0376
    * SOF: master: b72c7ec2df19de69efb57902edd2b277e169ae2f
2) Name of the topology file
    * Topology: sof-cml-rt711-rt1308-rt715.tplg with my patches to add the mux component.
3) Name of the platform(s) on which the bug is observed.
    * Platform: CML SDW platform

**Screenshots or console output**
kernel log:
[    3.105078] sof-audio-pci 0000:00:1f.3:  config: periods snk 2 src 2 fmt 0
[    3.105082] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    3.105240] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x30010000 size 20
[    3.105274] sof-audio-pci 0000:00:1f.3: error: create process failed
[    3.105293] sof-audio-pci 0000:00:1f.3: error: process loading failed
[    3.105315] sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 33 name : MUXDEMUX10.0 stream none reply 0
[    3.105355] sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget MUXDEMUX10.0
[    3.105380] sof-audio-pci 0000:00:1f.3: ASoC: topology: could not load header: -22
[    3.105420] sof-audio-pci 0000:00:1f.3: tplg: unload control name : 0000:00:1f.3
[    3.105430] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[    3.105517] sof-audio-pci 0000:00:1f.3: error: reply expected 0 got 12 bytes
[    3.105544] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x30020000 size 0
[    3.105588] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -22
[    3.105613] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[    3.105637] sof-audio-pci 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    3.105668] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
[    3.105729] sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[    3.105850] sof_sdw sof_sdw: snd_soc_register_card failed -22

sof-logger:
[  3865788.750000] (        6.770833) c0 mux                     src/audio/mux/mux.c:93   ERROR mux_set_values(): mux component is not able to mix channels
[  3865804.635417] (       15.885417) c0 demux                 src/audio/component.c:135  ERROR comp_new(): unable to create the new component
[  3865811.666667] (        7.031250) c0 ipc                           src/ipc/ipc.c:186  ERROR ipc_comp_new(): component cd = NULL
[  3865818.958333] (        7.291667) c0 ipc                       src/ipc/handler.c:1131 ERROR ipc: pipe 10 comp 51 creation failed -22
[  3865979.895833] (      160.937500) c0 sa                          src/lib/agent.c:65   perf sys_load peak plat 24570 cpu 193288
[  3866135.885417] (      155.989578) c0 ipc                       src/ipc/handler.c:1246 ipc: comp 0 -> free

libinyang, 2020-11-24:
https://github.com/thesofproject/sof/pull/3624 can fix this issue.

The mux is like:
define(matrix5, `ROUTE_MATRIX(10,
                             `BITS_TO_BYTE(1, 0, 0 ,0 ,0 ,0 ,0 ,0)',
                             `BITS_TO_BYTE(0, 1, 0 ,0 ,0 ,0 ,0 ,0)',
                             `BITS_TO_BYTE(0, 0, 1 ,0 ,0 ,0 ,0 ,0)',
                             `BITS_TO_BYTE(0, 0, 0 ,1 ,0 ,0 ,0 ,0)',
                             `BITS_TO_BYTE(0, 0, 0 ,0 ,0 ,0 ,0 ,0)',
                             `BITS_TO_BYTE(0, 0, 0 ,0 ,0 ,0 ,0 ,0)',
                             `BITS_TO_BYTE(0, 0, 0 ,0 ,0 ,0 ,0 ,0)',
                             `BITS_TO_BYTE(0, 0, 0 ,0 ,0 ,0 ,0 ,0)')')

define(matrix6, `ROUTE_MATRIX(11,
                             `BITS_TO_BYTE(0, 0, 0 ,0 ,1 ,0 ,0 ,0)',
                             `BITS_TO_BYTE(0, 0, 0 ,0 ,0 ,1 ,0 ,0)',
                             `BITS_TO_BYTE(0, 0, 0 ,0 ,0 ,0 ,1 ,0)',
                             `BITS_TO_BYTE(0, 0, 0 ,0 ,0 ,0 ,0 ,1)',
                             `BITS_TO_BYTE(0, 0, 0 ,0 ,0 ,0 ,0 ,0)',
                             `BITS_TO_BYTE(0, 0, 0 ,0 ,0 ,0 ,0 ,0)',
                             `BITS_TO_BYTE(0, 0, 0 ,0 ,0 ,0 ,0 ,0)',
                             `BITS_TO_BYTE(0, 0, 0 ,0 ,0 ,0 ,0 ,0)')')

dnl name, num_streams, route_matrix list
MUXDEMUX_CONFIG(mux_priv_10, 2, LIST(`  ', `matrix5,', `matrix6'))

lgirdwood, 2020-11-25:
@libinyang @juimonen what's the ETA for adding enum kcontrol to mux ? This will simplify.

juimonen, 2020-11-25:
@lgirdwood sorry about the delay... I have to re-think it a bit still. The issue I'm having is that although we are not mixing we are still using the mux/demux to pick individual channels in some cases (I think @keyonjie has these uses cases in some topologies) -> so it still makes the enums cumbersome, so we are not yet that simple as ASoC mux/demux.

lgirdwood, 2020-11-25:
@juimonen still not following why enums are complex here ?
We have 2 use cases
1) mux - (N:1) select channels from multiple sources and copy to one sink. Create enum kcontrol for each **sink** channel that lists all source channels (including a mute option).
2) demux - (1:N) select channels from 1 source to multiple sinks. Create enum kconrol for each **source** channel that list all sink channels.

i.e. for 2:1 with stereo channels we would create 2 enums (on sink) with "Mute", "source0-left", "source0-right", "source1-left", "source1-right"

juimonen, 2020-11-25:
@lgirdwood the thing you described should be here: https://github.com/juimonen/sof/commits/mux_enum

it is very poorly tested, but exactly tried to do what you we're asking.

lgirdwood, 2020-11-26:
> @lgirdwood the thing you described should be here: https://github.com/juimonen/sof/commits/mux_enum
> 
> it is very poorly tested, but exactly tried to do what you we're asking.

Please send as a PR

lgirdwood, 2021-01-14:
@juimonen any updates here ?

plbossart, 2021-06-10:
@juimonen @libinyang @lgirdwood is this still relevant? last update was in November...

plbossart, 2022-12-12:
no update in 18 months -> closing

closed 2022-12-12 by plbossart