thesofproject/sof #3639: [FEATURE] Upgrade topologies to work with alsa-lib upstream

ranj063, 2020-11-24:
**Is your feature request related to a problem? Please describe.**
SOF topologies cannot be built with upstream alsa-lib

**Describe the solution you'd like**
Modify the topologies to use the inclusive language suggested in recent alsa-lib updates

plbossart, 2020-11-30:
@ranj063 the SOF topologies should build with alsa-lib upstream, you will get a lot of messages telling that the topology needs to change but backwards compatibility is preserved. if that's not the case it's a bug. Can you clarify

I was counting on @libinyang to provide us with alsa-lib upstream packages *before* we can do the change.

Also we may want to be patient and make sure alsa-lib updates are actually propagated in most distros used for SOF development before changing topologies.

ranj063, 2020-11-30:
@plbossart sorry I had a problem with compilation because of my own changes to alsa-lib master. The SOF topologies compile just fine with a lot of messages about deprecates.

lgirdwood, 2021-01-12:
@ranj063 good now ? Can we close ?

ranj063, 2021-01-12:
@lgirdwood yeah. I think we can close this. We still use deprecates but maybe its not super important

closed 2021-01-12 by ranj063