thesofproject/sof #3640: [BUG][TGL] SSP regression on TGL platform w/ Maxim98373+RT5682 codec

keqiaozhang, 2020-11-24:
**Describe the bug**
We observed this issue on TGL chromebook w/ Maxim98373+RT5682 codec,  there's no sound can be heard when playing on speaker and also some distortion when playing on headset. This issue only observed on tgl-010-hotfix and tgl-011-stable-drop branch. bisection result shows that this issue is caused by https://github.com/thesofproject/sof/pull/3418, the SSP fix for power consumption. 

**To Reproduce**
$ speaker-test -Dhw:0,0 -t wav -c 2 
$ speaker-test -Dhw:0,1 -t wav -c 2 

**Reproduction Rate**
100%

**Expected behavior**
Both speaker and headset playback should be clean w/o distortion and silence.

**Impact**
Functional issues

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev/4f72eb13
    * SOF: https://github.com/thesofproject/sof/commit/bce9a6b1b2f7acbf1bccb23e8c67cf17e774b731 + Maxim DSM Library patch
2) Name of the topology file
    * Topology: sof-tgl-max98373-rt5682.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL-chromebook

**kernel log and trace log**
no obvious errors were found
[tgl-dmesg.txt](https://github.com/thesofproject/sof/files/5587177/tgl-dmesg.txt)
[tgl-trace.txt](https://github.com/thesofproject/sof/files/5587182/tgl-trace.txt)

mengdonglin, 2020-11-24:
@keyonjie tried to fix this regression by https://github.com/thesofproject/sof/pull/3622

mengdonglin, 2020-11-30:
Closed as #3622 merged by release branch https://github.com/thesofproject/sof/tree/tgl-011-drop-stable, RC4

closed 2020-11-30 by mengdonglin