thesofproject/sof #3650: [BUG] Update sof docker for latest alsa-lib and alsa-utils

juimonen, 2020-11-29:
**Describe the bug**
Current sof docker can't build enum topologies, because of old versions of alsa-lib and alsa-utils

**To Reproduce**
Build topology with enum control

**Reproduction Rate**
Always

**Expected behavior**
Topologies with enum controls can be built

**Impact**
Showstopper

In:
scripts/docker_build/sof_builder/Dockerfile

we have:
git clone $CLONE_DEFAULTS https://github.com/thesofproject/alsa-lib.git && \
git clone $CLONE_DEFAULTS https://github.com/thesofproject/alsa-utils.git

The above are not up-to-date with upstream

Also latest crosstools-ng needs to increase open file size, so I needed to add:
 --ulimit nofile=65536:65536 into /scripts/docker_build/sof_builder/docker-build.sh (don't know if this is just my build environment)

juimonen, 2020-11-30:
made this: https://github.com/thesofproject/sof/pull/3652

xiulipan, 2020-12-04:
@juimonen I see the alsa-lib and alsa-utils are updated to latest.
Could you try if your tplg can build with the **thesofproject/sof:20201204** image
```
docker pull thesofproject/sof:20201204
docker tag thesofproject/sof:20201204 sof
./scripts/docker-run.sh ./scripts/build-tools.sh
```

juimonen, 2020-12-08:
> docker pull thesofproject/sof:20201204
> docker tag thesofproject/sof:20201204 sof
> ./scripts/docker-run.sh ./scripts/build-tools.sh

@xiuli I tried this and it didn't compile the enum... my docker build file has direct links to alsa-lib/utild upstream... not sure if this image has the "good" versions or the update to our internal repo was bad then.

lgirdwood, 2020-12-09:
@xiulipan Btw, do we have a docker update file ? I think we have a build file that creates all the repos etc, but is there a script that does a remote update  on the container ?

xiulipan, 2020-12-10:
@lgirdwood Update local docker image with such scripts will not benefit us. It would exactly same with update your local environment. And it will also make Docker container size to increase (as old layers is kept)
To avoid container to become crazy big and different from each other, we always re-build the Docker and upload to Docker-hub. This will make sure we have a standard shared build environment.

xiulipan, 2020-12-10:
@juimonen Could you try with the Docker image I built with upstream alsa-lib?
```
docker pull thesofproject/sof:20201210
docker tag thesofproject/sof:20201210 sof
./scripts/docker-run.sh ./scripts/build-tools.sh
```

juimonen, 2021-05-21:
this is fixed, let's close

closed 2021-05-21 by juimonen