thesofproject/sof #3663: [BUG][TGL][DSM][I2S] No Data to the DSM module when the reference stream is not opened

RDharageswari, 2020-12-03:
**Describe the bug**
Following is the topology for Smart amplifier speaker & reference stream
[image]

**To Reproduce**
Start the speaker playback stream and probe the BUF1.2 which is a Feedback input module to DSM component.
No Feedback data will be captured.

**Reproduction Rate**
100%

**Expected behavior**
probe and capture the BUF1.2 to check the feedback data and BUF1.2 is expected to have fb data even when the echo reference stream is not open.

**Impact**
showstopper

**Environment**
https://github.com/thesofproject/sof/tree/tgl-011-drop-stable + Maxim DSM algo
The feedback path is "On"
localhost /sys/kernel/debug/asoc/sof-rt5682/0000:00:1f.3/dapm # cat * | grep On
BUF1.0: On  in 1 out 2
BUF1.1: On  in 9 out 2
BUF1.2: On  in 8 out 2
BUF11.1: On  in 8 out 2
MUXDEMUX11.0: On  in 8 out 2
PCM0P: On  in 1 out 2
SMART_AMP1.0: On  in 9 out 2
SSP1.IN: On  in 8 out 2
SSP1.OUT: On  in 1 out 2
Smart Amplifier Playback 0: On  in 1 out 2
On

bkokoszx, 2020-12-04:
@RDharageswari 
Hi Dhara, I think that I do not understand the issue description. Would you like to start playback stream and check BUF1.2 data without starting capture? I mean that BUF1.2 is "updated" by capture stream.

keyonjie, 2020-12-04:
Yes @RDharageswari this is known limitation, the feedback data won't be valid before the capture pipeline is activated with today's design.

RDharageswari, 2020-12-08:
Hi Keyon,
Why is there a dependency on the capture pipeline for the feedback data?
The DSM module will need this data for its processing, even during the speaker playback

keyonjie, 2020-12-08:
@RDharageswari I know your point, but as you may know, the feedback data is generated by the capture pipeline, if the capture PCM is not active, no feedback data will be generated with today's design.

To meet the requirement you mentioned, we need to redesign the topology, e.g. put the whole playback + demux + feedback as a single pipeline and the demux -> host capture as another one. This is a significant change and there might be scheduling gap with today's FW pipeline management. @bkokoszx please chime in about this.

bkokoszx, 2020-12-08:
@keyonjie 
Yes, you are right. At the moment we treat DSM pipelines as usual pipelines - we handle them (playback and capture) separately. As you said, if we wanted to get feedback data always, we should treat playback pipeline + capture_demux + capture_dai (we should split capture into two pipelines) as one pipeline during running aplay, which I think is a bit tricky at the moment. I'm not sure whether it should be handle by driver or FW.

lgirdwood, 2020-12-09:
@RDharageswar do we use PCM0C for capture or is this only ever used for DSM data ? 
If its for DSM only then it could be dropped and the smart amp could directly start the second pipeline (i.e. drop PCM0C, BUF11.0, MUXDEMUX and connect BUF11.1 to smart amp - i.e. pipe1 is SSP1.IN -> BUF11.1 -> (smartamp on pipe0)).

lgirdwood, 2021-01-14:
@RDharageswari any update ? can we close ?

RDharageswari, 2021-01-14:
Hi Liam,
Will raise as a feature request.
Will close it for now!

closed 2021-01-14 by RDharageswari