thesofproject/sof #3666: [BUG]rimage submodule f3e52cd now not on any branch

xiulipan, 2020-12-04:
Now the rimage submodule is checkout to https://github.com/thesofproject/rimage/commit/f3e52cd17e04dcc703a54501fd26e166c927e55e

But this commit shows
`This commit does not belong to any branch on this repository, and may belong to a fork outside of the repository.`

This brings trouble for rimage related update.

xiulipan, 2020-12-04:
@juimonen @lgirdwood Could you help to check where https://github.com/thesofproject/rimage/commit/f3e52cd17e04dcc703a54501fd26e166c927e55e comes from?
Could we set it to latest rimage HEAD https://github.com/thesofproject/rimage/commit/7b7c192e70c23a6024902e6781c32fc0f058c37a now

lgirdwood, 2020-12-04:
@xiulipan it should be pointing to https://github.com/thesofproject/rimage/commit/7b7c192e70c23a6024902e6781c32fc0f058c37a
It looks like https://github.com/thesofproject/rimage/commit/f3e52cd17e04dcc703a54501fd26e166c927e55e is a development commit that was squashed and deleted.

xiulipan, 2020-12-08:
@lgirdwood Update the rimage to have TGL-H support with https://github.com/thesofproject/sof/pull/3667

closed 2020-12-09 by lgirdwood