thesofproject/sof #3677: [BUG][TGL] TGLU_VOLT_SDW has no sound when playback on speaker

RuiqingX, 2020-12-08:
Describe the bug
Do playback on speaker,the speaker has no sound, but "aplay -l" and dmesg have no error, and it had been confirmed that it is not a hardware issue.

````
**** List of PLAYBACK Hardware Devices ****
card 0: Device_1 [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sofsoundwire [sof-soundwire], device 0: Headphone (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sofsoundwire [sof-soundwire], device 2: smart373-spk (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sofsoundwire [sof-soundwire], device 6: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sofsoundwire [sof-soundwire], device 7: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sofsoundwire [sof-soundwire], device 8: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sofsoundwire [sof-soundwire], device 9: HDMI4 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
````
Device
TGLU_VOLT_SDW with sh-tglu-volt-sdw-03

To Reproduce
aplay on speaker

Reproduction Rate
100%

Impact
speaker playback is impossible

Environment
Kernel: { sof-dev 47f55b1f-1}
SOF: {c8ae78ab0e4e-3}
Platform: {Volteer}

lgirdwood, 2020-12-09:
@WYL-12 I dont think https://github.com/thesofproject/sof/commit/8a8e8736a84c7557205c46add31a7577f562915d is correct, it's a commit for IMX topology. The issue may not be 100% reproducible ?

keyonjie, 2020-12-15:
@aiChaoSONG  can you help to double check the bisect result?