thesofproject/sof #3681: [BUG][CML][CI] firmware boot failure on helios in suspend-resume test

aiChaoSONG, 2020-12-10:
**Describe the bug**
In CI test report 1082 1059, fw boot failure on helios in suspend-resume test case, with below dmesg:
```
sh-cml-hel-rt5682-04 kernel: [  201.748536] sof-audio-pci 0000:00:1f.3: Firmware download successful, booting...
sh-cml-hel-rt5682-04 kernel: [  203.686858] asix 1-2.1:1.0 enx000ec668ad1e: link up, 100Mbps, full-duplex, lpa 0x41E1
sh-cml-hel-rt5682-04 kernel: [  203.771981] sof-audio-pci 0000:00:1f.3: error: firmware boot failure
sh-cml-hel-rt5682-04 kernel: [  203.772015] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
sh-cml-hel-rt5682-04 kernel: [  203.772043] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00004000
sh-cml-hel-rt5682-04 kernel: [  203.772089] sof-audio-pci 0000:00:1f.3: error: extended rom status:  0x5 0x0 0x4000 0x0 0x0 0x0 0x1811102 0x0
sh-cml-hel-rt5682-04 kernel: [  203.772112] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
sh-cml-hel-rt5682-04 kernel: [  203.772123] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware after resume -5
sh-cml-hel-rt5682-04 kernel: [  203.772147] PM: dpm_run_callback(): pci_pm_resume+0x0/0x80 returns -5
sh-cml-hel-rt5682-04 kernel: [  203.772163] PM: Device 0000:00:1f.3 failed to resume: error -5
```

**To Reproduce**
run suspend-resume test case on heliso, or do suspend resume with rtcwake

**Reproduction Rate**
issue is seen in recent several daily test report

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit:  b2c49c1c
SOF Branch: master
SOF Commit:  cc1cd8a4af1b

[dmesg.txt](https://github.com/thesofproject/sof/files/5670547/dmesg.txt)

slawblauciak, 2021-02-26:
@keyonjie @ranj063 how is the firmware boot failure condition determined? Is it based on a timeout?

lgirdwood, 2021-03-01:
> @keyonjie @ranj063 how is the firmware boot failure condition determined? Is it based on a timeout?

Yes, although the panic code looks like FW has booted.  @keyonjie can we also print the IPC message value (from FW) on a boot failure. This would at least show FW has sent boot complte.

ranj063, 2021-03-01:
> @keyonjie @ranj063 how is the firmware boot failure condition determined? Is it based on a timeout?

@slawblauciak boot failure is based on a timeout before which the driver expects the FW_READY IPC.

lgirdwood, 2021-03-05:
@aiChaoSONG we have had a lot of clocking improvements and fixed recently - have you tried latest master ?

aiChaoSONG, 2021-03-08:
@lgirdwood Thanks, this issue is not see in daily test for quite a few weeks, I will observe on more week before close.

plbossart, 2021-06-10:
@aiChaoSONG it's been 3 months since your last comment, can we close or is this issue still relevant?

aiChaoSONG, 2021-06-11:
@plbossart let's close this, fixed already

closed 2021-06-11 by aiChaoSONG