thesofproject/sof #3690: [BUG IMX] Input/output error when do simultaneous playback/capture test

dbaluta, 2020-12-15:
**Describe the bug**
Input/output error when doing simultaneous playback.capture

**To Reproduce** 

$ aplay -Dhw:3,0 -f S32_LE -c 2 -r 48000 /dev/urandom
$ arecord -Dhw:3,0 -f S32_LE -c 2 -r 48000 /dev/null

**Reproduction Rate**

100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.10
    * SOF: imx-stable-1.6-stable
2) Name of the topology file
    * Topology: sof-imx8mp-wm8960.tplg
    *
3) Name of the platform(s) on which the bug is observed.
    * Platform: i.MX8MP

Regression is introduced by https://github.com/thesofproject/sof/pull/3339

dbaluta, 2020-12-15:
@lyakh @iuliana-prodan

Regression introduced by https://github.com/thesofproject/sof/pull/3339

This happens because we reach https://github.com/thesofproject/sof/blob/master/src/schedule/dma_multi_chan_domain.c#L296
```
				*comp = dma_domain->data[i][j].task->sched_comp;
			} else if (dma_domain->data[i][j].task->sched_comp != *comp) {
				continue;
			}
```
with ``dma_domain->data[i][j] = NULL``

Our DMA controller has 32 channels but only two are used for this scenario so having ```dma_domain->data->data[i][j] = NULL```.

I wonder why this is not happening on Intel platforms. Are always all the channels started?

We have two possible solutions for this problem:

1) Check for dma_domain->data[i][j].task 

```
--- a/src/schedule/dma_multi_chan_domain.c
+++ b/src/schedule/dma_multi_chan_domain.c
@@ -293,7 +293,8 @@ static bool dma_multi_chan_domain_is_pending(struct ll_schedule_domain *domain,
                                        continue;
 
                                *comp = dma_domain->data[i][j].task->sched_comp;
-                       } else if (dma_domain->data[i][j].task->sched_comp != *comp) {
+                       } else if (!dma_domain->data[i][j].task ||
+                                  dma_domain->data[i][j].task->sched_comp != *comp) {
                                continue;
                        }
```

2) Check for channel status

```
--- a/src/schedule/dma_multi_chan_domain.c
+++ b/src/schedule/dma_multi_chan_domain.c
@@ -286,6 +286,11 @@ static bool dma_multi_chan_domain_is_pending(struct ll_schedule_domain *domain,
 
        for (i = 0; i < dma_domain->num_dma; ++i) {
                for (j = 0; j < dmas[i].plat_data.channels; ++j) {
+
+                       /* channel not running */
+                       if (dmas[i].chan[j].status != COMP_STATE_ACTIVE)
+                               continue;
```

I would go for option 2) but need some guidance on this.

lgirdwood, 2020-12-17:
> Our DMA controller has 32 channels but only two are used for this scenario so having `dma_domain->data->data[i][j].task = NULL` is legit.
> 
> I wonder why this is not happening on Intel platforms. Are always all the channels started?

No - only 1 channel is used per stream on Intel HW. However, on some platforms the DMA IRQ is shared by all the channels.
I would also prefer 2.
@dbaluta btw, any ETA for CI for IMX ? I guess you have CI for Zephyr, so you could get CI for free when your Zephyr port is done ?

iuliana-prodan, 2020-12-17:
@lgirdwood you mean that when doing a dma irq clear on a channel, this might change the status on other channels too?

One more thing, regarding #3339 - when reading the commit message I see that it 's possible to have multiple tasks on the same channel, but in the code (ll_schedule.c and dma_multi_chan_domain.c) we only look for one task per channel. Am I missing something here?

dbaluta, 2020-12-17:
> No - only 1 channel is used per stream on Intel HW. However, on some platforms the DMA IRQ is shared by all the channels.
> I would also prefer 2.
> @dbaluta btw, any ETA for CI for IMX ? I guess you [...]

Best we could do is to setup an internal CI. And now we switched to 5.10 and stable-v1.6. I will talk with Zephyr guys to see what they have in terms of CI. 

Biggest stopper is that there are some architecture patches on imxes which are not yet upstream so using topic/sof-dev is out of the question for now.

commit 849a878f (2020-12-18): schedule: dma_multi_chan: fix null pointer when scanning for tasks

When scanning all channels for a certain task we end up comparing
with a null task.
That's because we have multiple channels (e.g. 32 for i.MX) but only
some of them are used in certain scenarios. For example, for simultaneous
playback/capture we use only 2 DMA channels.
To fix this, validate task before using it.

Fixes #3690
Fixes: 647390f75e6a ("schedule: dma_multi_chan: fix a DMA channel interrupt race")
PR #3696 (2020-12-18): schedule: dma_multi_chan: fix null pointer when scanning for tasks
When scanning all channels for a certain task we end up comparing
with a null task.
That's because we have multiple channels (e.g. 32 for i.MX) but only
some of them are used in certain scenarios. For example, for simultaneous
playback/capture we use only 2 DMA channels.
To fix this, validate task before using it.

Fixes #3690
Fixes: 647390f75 ("schedule: dma_multi_chan: fix a DMA channel interrupt race")
commit 07ef8adb (2020-12-18): schedule: dma_multi_chan: fix null pointer when scanning for tasks

When scanning all channels for a certain task we end up comparing
with a null task.
That's because we have multiple channels (e.g. 32 for i.MX) but only
some of them are used in certain scenarios. For example, for simultaneous
playback/capture we use only 2 DMA channels.
To fix this, validate task before using it.

Fixes #3690
Fixes: 647390f75e6a ("schedule: dma_multi_chan: fix a DMA channel interrupt race")
closed 2020-12-18 by lgirdwood