thesofproject/sof #3702: [BUG][APL][CI]FW xrun failed on APL_UP2_NOCODEC with playback test.

WYL-12, 2020-12-21:
Describe the bug
In CI test report 1293, FW xrun failed on APL_UP2_NOCODEC with playback test, with below dmesg:

```
declare -- cmd="journalctl_cmd --since=@1608518658"
2020-12-21 02:52:56 UTC Sub-Test: [REMOTE_INFO] Test Result: PASS!
2020-12-21 02:52:56 UTC [REMOTE_INFO] pkill -TERM sof-logger
2020-12-21 02:52:57 UTC [REMOTE_ERROR] sof-logger was already dead
2020-12-21 02:52:58 UTC [REMOTE_INFO] nlines=100473 /home/ubuntu/sof-test/logs/check-playback/2020-12-21-02:44:18-3041/slogger.txt
2020-12-21 02:52:58 UTC [REMOTE_INFO] Test Result: FAIL!
```

To Reproduce
run check-playback-100times test case on APL.

Reproduction Rate
issue is seen in recent several daily test report

Environment:
Kernel Branch: topic/sof-dev
Kernel Commit: 27429439-3
SOF Branch: master
SOF Commit:eaa9a55d0b84-3
SOF Test Branch: master
SOF Test Commit: 84da71c

[dmesg.txt](https://github.com/thesofproject/sof/files/5723051/dmesg.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/5723052/slogger.txt)

mengdonglin, 2021-01-12:
Cannot reproduce this issue in CI daily test during 01/06/2021 to 01/11/2021(report ID #1478 - 1568)

slawblauciak, 2021-01-12:
Issue no longer occurs.

closed 2021-01-12 by slawblauciak