thesofproject/sof #3706: [BUG][ICL][Probe] Incorrect data in probe extracted file.

jdrzaszcz, 2020-12-22:
**Describe the bug**
When running simultaneous playback and capture on SSP interface on external loopback with probe points connected to 6 buffers, there is a problem to properly unpack the extracted probe file due to 'error: wrong state 2, err 0\n' for format:
    SAMPLING_RATE = 48000
    SAMPLE_SIZE = 32
    CONTAINER_SIZE = 32
    CHANNELS_COUNT = 8

When trying to unpack such a corrupted probe extracted file from the command line there is such an error:
[image]

**Topology**
```
                    pipe_plb
                    +-----------------------------------------+
    +----------+    | +---+  +---+  +---+  +---+  +---+  +---+|   +----+
    | Host plb |----->|Buf|->|Vol|->|Buf|->|Vol|->|Buf|->|Dai|--->|SSP0|
    +----------+    | +---+  +---+  +---+  +---+  +---+  +---+|   +----+
                    +-----------------------------------------+    |  |
                                                                   |  |
                    pipe_cap                                       |  |
                    +-----------------------------------------+    |  |
    +----------+    | +---+  +---+  +---+  +---+  +---+  +---+|   +----+
    | Host cap |<-----|Buf|<-|Vol|<-|Buf|<-|Vol|<-|Buf|<-|Dai|<----|SSP0|
    +----------+    | +---+  +---+  +---+  +---+  +---+  +---+|   +----+
                    +-----------------------------------------+
```
**To Reproduce**
Ru slim driver python tests TestMultiProbeVol48000Hz32b32b8ch from file test_01_asrc_simple_capture.py
with --iteration>=10

**Reproduction Rate**
~  30% 

**Expected behavior**
The extraction probe file should be properly unpack  and the wave data in unpacked files shouldn't have any glitch.

**Environment**
ICL RVP with SSP interface.
Commit: eaa9a55d0b844d93dc861fff727d0782886f6d66 from SOF master branch.

**Example logs**
[18_03_TestMultiProbeVol48000Hz32b32b8ch.zip](https://github.com/thesofproject/sof/files/5728742/18_03_TestMultiProbeVol48000Hz32b32b8ch.zip)

abonislawski, 2021-01-05:
After a quick check I can confirm that this is not probe app issue.
Extracted file is corrupted, this can be seen only on ICL RVP with 30% repro rate and only with high demanding format with several probe points so maybe a high load issue with DMA  in such heavily demanding scenario

lgirdwood, 2021-01-06:
@abonislawski we may want to add some trace_err/warn in probes if the data request exceeds DMA bandwidth then, this way users will know that data may be corrupted.