thesofproject/sof #3707: [cml][sdw][STRA-P]:  The volume of built-in audio output is low for 1-2 sec as come back from runtime-pm

jason77-wang, 2020-12-22:
On the STRA-P (cml soundwire audio), before runtime suspend, the audio (speaker/headphone)'s output volume is correct, after it enters runtime suspend and runtime resume, the output volume is lower than before for about 1-2 seconds,then it changes back to normal.

This issue could be reproduced with sof-firmware-v1.6 and the kernel branch sof-dev-rebase. This issue could not be reproduced on MLK OLY machine (TGL soundwire audio).

kv2019i, 2020-12-22:
Confirmed as FW topology issue, moving to sof component. It seems we lost the tags in transfers, will add them back.

singalsu, 2020-12-22:
@lihow731 Can you please test with this topology.  Check where the topology file is located and replace it with this (make a backup of file first)

```
cd /lib/firmware/intel
find . -name sof-cml-rt711-rt1308-rt715.tplg

```
[sof-cml-rt711-rt1308-rt715.zip](https://github.com/thesofproject/sof/files/5729408/sof-cml-rt711-rt1308-rt715.zip)

lihow731, 2020-12-22:
@singalsu I'll test the topology, tomorrow.

lihow731, 2020-12-23:
@singalsu 

I tested the sof-cml-rt711-rt1308-rt715.zip, this issue still exists. 

From kernel message, the ABI info: Topology: ABI 3:18:1 Kernel ABI 3:13:0
I posted the full dmesg output on https://pastebin.ubuntu.com/p/WKnsj4XVDd/ .
The video: https://drive.google.com/file/d/1LOrMj9CDJQ_z8WNPR6-pFcX_8eEV7Bzr/view?usp=sharing

mengdonglin, 2021-01-04:
@lihow731 On your CML STRA-P device, can this issue only be reproduced with built-in speaker output? Or can it also be reproduced with headset output?

lihow731, 2021-01-04:
@mengdonglin

This issue only is reproduced with built-in speaker. 
This issue can not be reproduced with headset output.

singalsu, 2021-01-04:
The patch in my opinion impacted, the start ramp is now much shorter, about 30 ms (compare to above! though should be 20) but the used MPEG audio coding may impact it. A wav capture without lossy format would be better. If there's still difference in ramp length it's caused by external amplifier, codec, etc.

[image]

singalsu, 2021-01-04:
@lihow731 @aiChaoSONG I don't have a device like this. Do you still feel the fade-in is too long for UI beep/ping/blip like sounds?  Though the gain ramp with the PR is the same as with headset so they should sound similar unless there's difference in codec/amp power-on times.

aiChaoSONG, 2021-01-05:
@singalsu I was unable to reproduced this issue on one device in PRC, I guess it may be device specific.

lihow731, 2021-01-05:
@aiChaoSONG This issue  can be reproduced on STRA-DVT2-C7 and STRA-DVT2-C8. Could you get these SKUs?
I can reproduce this issue on another platform: cyborg N 15 (CNU5-DVT1-C7). (Maybe the root cause is different)

aiChaoSONG, 2021-01-06:
@lihow731 Some summary from my side, I cannot reproduce this issue
Kernel: sof-dev-rebase-20201201, kernel deb attached in one of the above comment
SOF: v1.6
Device: DVT1-C3
Check the video: https://www.youtube.com/watch?v=PC2SIuumcqU, playback on speaker with a 997Hz audio.

As you can see from the video, when I hit enter, bunch of messages printed in the right terminal immediately. you can check how long SOF is recovered from suspend, It doesn't takes that long. On my side， from `Turning i915 HDAC power 1` to `ipc tx succeeded: 0x40020000: GLB_CTX_RESTORE` takes about 200ms

Please put below file to `/etc/modprobe.d/`, and rename it to `sof-dyndbg.conf`; open two terminals, one for aplay on speaker, and the other for dmesg monitoring like I do in the video. From the dmesg, we can know if it is SOF stuck somewhere or the aplay process stuck somewhere.
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/5773699/sof-dyndbg.conf.txt)

lihow731, 2021-01-07:
@aiChaoSONG 
Please check [this video on google driver](https://drive.google.com/file/d/14DB92fExUGbzPx0aqDmXPqV3dGibxpLj/view?usp=sharing)

kernel: Kernel: sof-dev-rebase-20201201 (in above comment)
topology: sof-cml-rt711-rt1308-rt715.zip (in above singalsu's comment)
I enabled the dyndbg that your sof-dyndbg.conf.

I use ubuntu 20.04 (LTS, but this version is a oem preload version. It is close ubuntu 20.04.1.).

keyonjie, 2021-01-11:
The issue is actually not fixed by #3708 by @lihow731, reopen it.

keyonjie, 2021-01-11:
@lihow731 it sounds better with your latest video, do you think we can close this?

lihow731, 2021-01-11:
@keyonjie if this is a device specific issue, I am fine to close it.
But, I observed this issue on other machines that is no a soundwire audio.
I hope we can find the root cause.

aiChaoSONG, 2021-01-11:
@lihow731 In the latest video you posted, I think we are aligned on the time delay, about 200ms, Do you think it is still very long for you?

lgirdwood, 2021-01-11:
codec HW will power on slowly to reduce pops and clicks. So I would expect to see some ramping an most devices. IIRC we do this in FW for some devices too. @keyonjie @aiChaoSONG

lihow731, 2021-01-12:
@lgirdwood Got it. It's an expected behavior. I agree to close this issue. Thank you.

PR #3708 (2021-01-05): Tools: Topology: Change volume demux playback ramp length to 20 ms
The earlier length 250 ms is quite long and makes very short UI
tones etc. quite silent. The 20 ms long ramp with zero crossings
detect is used in rest of this topology for playback so this is
changed to similar.

The topologies those use the changed macro are: sof-icl-rt711-rt1308-rt715-hdmi.m4, sof-apl-demux-pcm512x.m4, sof-tgl-max98357a-rt5682.m4, sof-cml-demux-rt5682.m4, and sof-tgl-nocodec-ci.m4.

Fixes: #3707
commit b8149536 (2021-01-05): Tools: Topology: Change volume demux playback ramp length to 20 ms

The earlier length 250 ms is quite long and makes very short UI
tones etc. quite silent. The 20 ms long ramp with zero crossings
detect is used in rest of this topology for playback so this is
changed to similar.

The topologies those use the changed macro are:
sof-icl-rt711-rt1308-rt715-hdmi.m4, sof-apl-demux-pcm512x.m4,
sof-tgl-max98357a-rt5682.m4, sof-cml-demux-rt5682.m4, and
sof-tgl-nocodec-ci.m4.

Fixes: #3707
closed 2021-01-14 by lgirdwood