thesofproject/sof #3711: [BUG] Dell Machine failed to load sof-tgl.ri

KaiChuan-Hsieh, 2020-12-23:
**Describe the bug**
I got a new Dell hardware. It's CPU information:
[lscpu.log](https://github.com/thesofproject/sof/files/5733650/lscpu.log)

The failed kernel log:
[journalctl-k.log](https://github.com/thesofproject/sof/files/5733653/journalctl-k.log)

Its acpi device info:
\_SB_.PC00.HDAS.SNDW.SWD0
0x000020025d071100
\_SB_.PC00.HDAS.SNDW.SWD1
0x000120025d130800
\_SB_.PC00.HDAS.SNDW.SWD2
0x000220025d130800
\_SB_.PC00.HDAS.SNDW.SWD3
0x000320025d071500

**To Reproduce**
To boot into Ubuntu.

**Reproduction Rate**
100/100

**Expected behavior**
The machine can load sof firmware successfully and audio works for internal mic/speaker and headest input/output.

**Impact**
No ALSA card can playback and record audio.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.10.0-rc5
    * SOF: current sof-tgl.ri is not workable.
2) Name of the topology file
    * Topology: sof-tgl-rt711-rt1308-rt715.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: The Dell machine product sku is 0A5E

mengdonglin, 2020-12-23:
@KaiChuan-Hsieh This device is using TGL-H silicon, and we already informed OSV that sof-tgl.ri is only for TGL-U silicon. SOF support for TGL-H is not published in SOF v1.6 or 1.6.1 atm.

xiulipan, 2020-12-25:
@mengdonglin @RanderWang It seems there is also a kernel issue here, already fixed with  https://github.com/thesofproject/linux/pull/2417

From kernel log, sof driver wants to load wrong FW binary here.
`十二 23 14:11:06 u-Precision-5760 kernel: sof-audio-pci 0000:00:1f.3: request_firmware intel/sof/sof-tgl.ri successful`

We need to check if the fix https://github.com/thesofproject/linux/pull/2417 is now merge into upstream.

PS: @KaiChuan-Hsieh where is this `5.10.0-1008-oem` kernel comes from?

KaiChuan-Hsieh, 2020-12-25:
@xiulipan 

The 5.10.0-1008-oem repository is https://code.launchpad.net/~canonical-kernel/ubuntu/+source/linux-oem/+git/focal/+ref/oem-5.10-next

The kernel works ok for https://github.com/thesofproject/linux/issues/2544
May I ask if https://github.com/thesofproject/linux/pull/2417 is required, then we have to backport it to our oem kernel?

Thanks,

xiulipan, 2020-12-25:
@KaiChuan-Hsieh I think it should be a must to support TGL-H platforms with PCI ID 0x43c8.
Now it is already unstreamed with https://github.com/torvalds/linux/commit/30ee3738f849b3f4af623c20adec73cdc4573a2e
And I do see the patch in https://github.com/torvalds/linux/blob/v5.10/sound/soc/sof/sof-pci-dev.c
So I think maybe some sync up is needed for the oem-5.10 with the upstream 5.10

xiulipan, 2020-12-25:
@KaiChuan-Hsieh Interesting, I also see the patch with https://git.launchpad.net/~canonical-kernel/ubuntu/+source/linux-oem/+git/focal/tree/sound/soc/sof/sof-pci-dev.c?h=oem-5.10-next

@libinyang If this is a issue like https://github.com/thesofproject/linux/pull/2657 that we use the fw name from ACPI table?

KaiChuan-Hsieh, 2020-12-25:
@xiulipan yes, but it doesn't try to load sof-tgl-h.ri which is not as what you expect, right? I have to check if we did add pci device id correctly.

Thanks for your notification.

xiulipan, 2020-12-25:
@KaiChuan-Hsieh Please try with #2657, sof-dev is also missing the fix.
It seems we also recently face similar issue @libinyang could you confirm?

libinyang, 2020-12-25:
@KaiChuan-Hsieh We have two patches for tgl-h. Could you please have a try? https://github.com/thesofproject/linux/pull/2650 and https://github.com/thesofproject/linux/pull/2657

KaiChuan-Hsieh, 2020-12-25:
@libinyang 

sure, but I don't have a workable sof-tgl-h.ri for TGL-H, or it will be okay by loading sof-tgl.ri with these patches?

libinyang, 2020-12-25:
@KaiChuan-Hsieh I'm not sure it is OK to load sof-tgl.ri. After applying the 2 patches, sof-tgl.ri should be renamed to sof-tgl-h.ri.

KaiChuan-Hsieh, 2020-12-25:
@libinyang yes, that's the point, I don't have sof-tgl-h.ri in hand.

libinyang, 2020-12-25:
@KaiChuan-Hsieh I think we can wait for the fw release and have the test.

KaiChuan-Hsieh, 2020-12-28:
@libinyang I found the I have to add "options snd-intel-dspcfg dsp_driver=3" for probing the device with correct driver.
I think the dspconfig table should be updated for the Multimedia audio device 8086:43c8 for using a correct driver to probe the device.

xiulipan, 2020-12-29:
@KaiChuan-Hsieh Thanks for reporting the issue with `snd-intel-dspcfg` could you also report another issue for this if we would like to let PCI `8086:43c8` to load SOF by default

KaiChuan-Hsieh, 2020-12-29:
@xiulipan Sure, here is the bug, https://github.com/thesofproject/sof/issues/3720.

xiulipan, 2020-12-29:
@KaiChuan-Hsieh Thanks! I will also move this issue to linux as we had fix in driver now.