thesofproject/sof #3713: [BUG] Acer Aspire A515-56 internal mic not working

jarek556, 2020-12-23:
**Describe the bug**
Acer Aspire 5 A515-56 internal microphone not working (silence). DMIC is detected, gain control and mute are available, but there is not sound.

**To Reproduce**
Open pavucontrol, select input: not mic activity. The same in gnome settings audio.

**Reproduction Rate**
100%

**Expected behavior**
Working mic

**Impact**
Impossible to use recording and voice communication

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: SMP Debian 5.9.6-1~bpo10+1
    * SOF: v1.6.1 (2020-12-23)
2) Name of the topology file
    * Topology: intel/sof-tplg/sof-hda-generic-2ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: {ALC255}

**dmesg**
```
[    3.463781] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.463788] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.463804] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.464262] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.464472] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.493452] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.553187] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    3.553189] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.553192] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    3.553958] sof-audio-pci 0000:00:1f.3: firmware: direct-loading firmware intel/sof/sof-tgl.ri
[    3.641456] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:6:1-53680
[    3.641457] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:17:0 Kernel ABI 3:16:0
[    3.641458] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[    3.641816] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 3 size 0x1c
[    3.641828] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 4 size 0x10
[    3.675746] sof-audio-pci 0000:00:1f.3: firmware: direct-loading firmware intel/sof-tplg/sof-hda-generic-2ch.tplg
[    3.675756] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:17:0 Kernel ABI 3:16:0
[    3.675757] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[    3.675760] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    3.676686] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    3.676688] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    3.677833] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    3.677837] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    3.678713] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    3.678716] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    3.679860] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    3.679863] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    3.679867] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    3.679870] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    3.679873] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    3.679876] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    3.680774] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    3.680776] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    3.681676] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    3.681679] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    3.682651] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    3.691161] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    3.777724] input: sof-hda-dsp Front Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
[    3.777784] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[    3.777808] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
[    3.777868] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input20

```
**arecord -l**
[... 147 lines cut ...]
│                                                        │  │             │  │             │  │             │  │                                                         │
│                                                        │  │             │  │             │  │             │  │                                                         │
│                                                        │  │             │  │             │  │             │  │                                                         │
│                                                        │▒▒│             │▒▒│             │  │             │  │                                                         │
│                                                        │▒▒│             │▒▒│             │  │             │  │                                                         │
│                                                        │▒▒│             │▒▒│             │  │             │  │                                                         │
│                                                        │▒▒│             │▒▒│             │▒▒│             │▒▒│                                                         │
│                                                        │▒▒│             │▒▒│             │▒▒│             │▒▒│                                                         │
│                                                        │▒▒│             │▒▒│             │▒▒│             │▒▒│                                                         │
│                                                        │▒▒│             │▒▒│             │▒▒│             │▒▒│                                                         │
│                                                        │▒▒│             │▒▒│             │▒▒│             │▒▒│                                                         │
│                                                        │▒▒│             │▒▒│             │▒▒│             │▒▒│                                                         │
│                                                        │▒▒│             │▒▒│             │▒▒│             │▒▒│                                                         │
│                                                       L└──┘R            └──┘             └──┘             └──┘                                                         │
│                                                      CAPTURE                                                                                                           │
│                                                       46<>46           46<>46           32<>32           32<>32                                                        │
│                                                 <     Dmic0      >   Dmic1 2nd     PGA2.0 2 Master  PGA4.0 4 Master                                                    │
│                                                                                                                                                                        │
│                                                                                                                                                                        │
│                                                                                                                                                                        │
│                                                                                                                                                                        │
│                                                                                                                                                                        │
│                                                                                                                                                                        │
└────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
```

lgirdwood, 2021-01-04:
@jarek556 can you confirm if this worked with older FW and kernel ? Can you also confirm if the capture WAV file is all 0s (except for the header).
@juimonen @singalsu any comments here ?

singalsu, 2021-01-05:
Could you please test with sof-hda-generic-4ch.tplg, just copy it over 2ch file (make backup copy of it first), and reboot. The capture with 4ch configuration needs to be four channels, e.g. "arecord -Dhw:0,6 -f dat -c 4 -d 10 rec.wav". Also check with amixer that the Dmic0 volume is 0 dB (or more gain) and unmuted. There's another similar bug reported and I suspect the connected microphones are using channels 3-4 clocks (waiting for reporter input there).

singalsu, 2021-01-05:
This issue may be similar as https://github.com/thesofproject/linux/issues/2460

lgirdwood, 2021-01-05:
Btw, do we know if these non working platforms have any "hard mic mutes" aka privacy mode ? i.e. where the mic data is gated via some HW prior to entering DSP.

jarek556, 2021-01-05:
@lgirdwood  - I have tested this notebook only in this configuration. After few days of testing I've returned it because I couldn't find solution for internal mic.

lgirdwood, 2021-01-06:
@jarek556 are you working for a distro or OEM ? It may be possible to get the audio schematic and figure out the DMIC routing to fix the issue.

GFdevelop, 2021-02-20:
I have the same issue on my laptop, the microphone is detected but in pavucontrol there isn't sound detection, the headset microphone work but when I plug it I have to manually switch to the headset microphone to disable the speaker output (speaker and headset output work at same time).

I have tried the module option `snd_hda_intel model=dual-codecs` but don't work.
How to solve?

```shell
$ grep Codec /proc/asound/card*/codec*
/proc/asound/card0/codec#0:Codec: Realtek ALC257
/proc/asound/card0/codec#2:Codec: Intel Kabylake HDMI
```

```shell
$ sudo alsa-info.sh

upload=true&script=true&cardinfo=
!!################################
!!ALSA Information Script v 0.4.65
!!################################

!!Script ran on: Sat Feb 20 16:44:09 UTC 2021

!!Linux Distribution
!!------------------

Arch Linux \r (\l) NAME="Arch Linux" PRETTY_NAME="Arch Linux" ID=arch HOME_URL="https://www.archlinux.org/" DOCUMENTATION_URL="https://wiki.archlinux.org/" SUPPORT_URL="https://bbs.archlinux.org/" BUG_REPORT_URL="https://bugs.archlinux.org/" LOGO=archlinux

!!DMI Information
!!---------------

Manufacturer:      LENOVO
Product Name:      20RR
Product Version:   Lenovo ThinkBook 13s-IML
Firmware Version:  CQCN29WW
System SKU:        LENOVO_MT_20RR_BU_idea_FM_ThinkBook 13s-IML
Board Vendor:      LENOVO
Board Name:        LNVNB161216

!!ACPI Device Status Information
!!---------------

/sys/bus/acpi/devices/ACPI0003:00/status 	 15
/sys/bus/acpi/devices/ACPI000C:00/status 	 15
/sys/bus/acpi/devices/CUST0001:00/status 	 15
/sys/bus/acpi/devices/INT0E0C:00/status 	 15
/sys/bus/acpi/devices/INT33A1:00/status 	 15
/sys/bus/acpi/devices/INT3400:00/status 	 15
/sys/bus/acpi/devices/INT3403:00/status 	 15
/sys/bus/acpi/devices/INT3403:01/status 	 15
/sys/bus/acpi/devices/INT3403:02/status 	 15
/sys/bus/acpi/devices/INT340E:00/status 	 15
/sys/bus/acpi/devices/INT34BB:00/status 	 15
/sys/bus/acpi/devices/INT3F0D:00/status 	 15
/sys/bus/acpi/devices/LIC0001:00/status 	 15
/sys/bus/acpi/devices/LNXPOWER:00/status 	 1
/sys/bus/acpi/devices/LNXPOWER:01/status 	 15
/sys/bus/acpi/devices/LNXPOWER:05/status 	 1
/sys/bus/acpi/devices/LNXPOWER:06/status 	 1
/sys/bus/acpi/devices/LNXPOWER:07/status 	 1
/sys/bus/acpi/devices/LNXPOWER:09/status 	 1
/sys/bus/acpi/devices/MSFT0001:00/status 	 15
/sys/bus/acpi/devices/NTC0702:00/status 	 15
/sys/bus/acpi/devices/PNP0103:00/status 	 15
/sys/bus/acpi/devices/PNP0B00:00/status 	 15
/sys/bus/acpi/devices/PNP0C02:00/status 	 3
/sys/bus/acpi/devices/PNP0C02:01/status 	 3
/sys/bus/acpi/devices/PNP0C02:05/status 	 3
/sys/bus/acpi/devices/PNP0C09:00/status 	 15
/sys/bus/acpi/devices/PNP0C0A:00/status 	 31
/sys/bus/acpi/devices/PNP0C0C:00/status 	 11
/sys/bus/acpi/devices/PNP0C0D:00/status 	 15
[... 1377 lines cut ...]
[    3.531805] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    3.548406] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    3.548424] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.548429] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 2
--
[    3.869283] Bluetooth: SCO socket layer initialized
[    3.875234] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC257: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:speaker
[    3.875237] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    3.875239] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    3.875240] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    3.875241] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    3.875243] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[    3.930337] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[    3.930342] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[    3.930447] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi3 overwritten
--
[    3.930479] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Alt Analog Codec Capture overwritten
[    3.941874] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input20
[    3.941900] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input21
[    3.941923] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input22
[    3.941943] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input23
[    3.941967] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input24
[    3.985956] usbcore: registered new interface driver btusb

```

singalsu, 2021-02-22:
@GFdevelop Thanks for report! There's another case with  Thinkbook 13s https://github.com/thesofproject/linux/issues/2460 . Can you please try from there the custom topology https://github.com/thesofproject/linux/files/5981682/sof-hda-generic-2ch-pdm1.zip . It needs to be copied over the default sof-hda-generic-2ch.tplg.

This is a temporary quick fix, the final outcome will likely be a kernel option.  (FYI @kv2019i ).

GFdevelop, 2021-02-22:
After one week of attemps this fix work, but the internal microphone(s) aren't muted when I plug the jack (the automute is enabled in alsamixer), headphone input/output and internal input/output work together at same time, is this a related issue?

In pavucontrol I see:
- ...
- output: hdmi3, hdmi2, hdmi1 and speaker+headphones
- input: headphone microphone and digital microphone
- ...

(in output it's a single switch for speaker and headphone but in input is a different switch)

The internal microphone is "boosted": in alsamixer I have mic boost at 0 but in pavucontrol I see a constant minimal sound detection.

Anyway thank you, it's a good job.

singalsu, 2021-02-22:
@juimonen Do you thoughts for the control issues?

There's a +20 dB gain added in topology. It could be built with 0 dB as well by editing the lines to build HDA generic topologies:

```
$ grep sof-hda-generic-2ch tools/topology/CMakeLists.txt 
	"sof-hda-generic\;sof-hda-generic-2ch\;-DCHANNELS=2\;-DHSPROC=volume\;-DDMICPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_48khz.m4\;-DDMIC16KPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_16khz.m4"

```
There's different  microphones sensitivites and the lower ones (those tolerate high SPL) sound rather silent in notebook usage. If you microphone is higher sensitivity the boost could be removed. An AGC or DRC in capture path should solve this in future.

juimonen, 2021-02-22:
@GFdevelop trying to decipher the info here...

It is kind of impossible that the inputs would be "on" at the same time (dmic & headset), simultaneous output routing is possible in hda devices.

AFAIK The "headphone mic" is something that should be visible only in Dell devices -> you should see "headset mic". I have not seen "headphone mic" ever in Lenovo devices.

So my guess is that you have something wrong in the kernel config ->so it might be that because of this you are not getting correct jack events either -> the headset playback is messed up.

juimonen, 2021-02-22:
@GFdevelop hmm I don't see this "headphone mic" in the amixer list you provided. But it is probably present in UCM configs. But it's "existence" should be tied to its presence in amixer controls... so my explanation above doesn't seem to hold -> something must be incorrect in the user space configs. Could you try to update alsa-lib, alsa-ucm-conf and pulseaudio? These are quite new features, so not sure how they are present in arch linux.

But with lenovo devices you should get jack event for both headphone insert and headset mic insert. In Ubuntu the switch is kind of co-operation between pulseaudio and UI, so I'm not sure how it supposed to work in Arch Linux. At least you should see the input/output options in  pavucontrol for you to change them manually.

GFdevelop, 2021-02-22:
I don't have any customization, my system is updated with arch repository, I have deleted `~/.config/pulse/` , when the jack is plugged in and I speak I see sound detection on headphone microphone and internal microphone, in pavucontrol input:
- Comet Lake PCH-LP cAVS Headphones Stereo Microphone
 Port: Headphones Stereo Microphone (plugged in)
- Comet Lake PCH-LP cAVS Digital Microphone
 Port: Digital Microphone

My jack it's a single combined input/output.
I'm using `xfce`, and in `xfce4-plugin-pulseaudio` I can manually select the input source but don't change the behaviour of the microphones, they listen together...in `alsamixer` I have automute in the playback view, in capture there isn't this option.