thesofproject/sof #3714: [BUG][CI][SDW] xrun and trigger pause/release ipc error in multiple-pause-resume

aiChaoSONG, 2020-12-25:
**Describe the bug**
The multiple-pause-resume will start several pipelines, and send `space` to every pipeline process to trigger pause/release simultaneously. Under this test case,  we firstly got xrun in the firmware, and failed to recover, then `trigger pause/release ipc error` is reported.

**To Reproduce**
run multiple-pause-resume.sh

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 36975c42
SOF Branch: master
SOF Commit:  eaa9a55d0b84

2) Name of the topology file
    * Topology: {sof-tgl-sdw-max98373-rt5682-4ch.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {chrome Volteer}

<details>
<summary>dmesg</summary>

```
kernel: [ 6390.722547] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
kernel: [ 6390.722740] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
kernel: [ 6390.722751] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
kernel: [ 6390.769305] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 1 cmd 4
kernel: [ 6390.769828] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
kernel: [ 6390.769832] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
kernel: [ 6390.770017] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
kernel: [ 6390.817396] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 2 dir 0 cmd 3
kernel: [ 6390.817403] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
kernel: [ 6390.817541] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
kernel: [ 6390.817551] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
kernel: [ 6390.919790] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 2 dir 0 cmd 4
kernel: [ 6390.920311] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2014001e successful
kernel: [ 6390.920314] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
kernel: [ 6390.920519] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
kernel: [ 6390.928182] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
kernel: [ 6390.928196] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
kernel: [ 6391.428175] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
kernel: [ 6391.428203] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
kernel: [ 6391.454328] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 1 cmd 3
kernel: [ 6391.454343] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
kernel: [ 6391.454537] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60060000 size 12
kernel: [ 6391.454556] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
kernel: [ 6391.454567] sof-audio-pci 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_trigger on 0000:00:1f.3: -22
kernel: [ 6391.454575]  echo: ASoC: trigger FE cmd: 3 failed: -22
kernel: [ 6391.454584] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 1 cmd 4
kernel: [ 6391.454598] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x24001e successful
kernel: [ 6391.454607] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
kernel: [ 6391.454738] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60070000 size 12
kernel: [ 6391.454744] sof-audio-pci 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_trigger on 0000:00:1f.3: -22
kernel: [ 6391.454750]  echo: ASoC: trigger FE cmd: 4 failed: -22
kernel: [ 6391.455091] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 1 cmd 3
kernel: [ 6391.455100] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
kernel: [ 6391.455419] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60060000 size 12
kernel: [ 6391.455440] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
kernel: [ 6391.455452] sof-audio-pci 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_trigger on 0000:00:1f.3: -22
kernel: [ 6391.455459]  echo: ASoC: trigger FE cmd: 3 failed: -22
kernel: [ 6391.455468] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 1 cmd 4
kernel: [ 6391.455913] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
kernel: [ 6391.455926] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
kernel: [ 6391.456098] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60070000 size 12
kernel: [ 6391.456109] sof-audio-pci 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_trigger on 0000:00:1f.3: -22
kernel: [ 6391.456116]  echo: ASoC: trigger FE cmd: 4 failed: -22
kernel: [ 6391.532260] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 2 dir 0 cmd 3
kernel: [ 6391.532267] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
kernel: [ 6391.532385] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60060000 size 12
kernel: [ 6391.532395] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
kernel: [ 6391.532398] sof-audio-pci 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_trigger on 0000:00:1f.3: -22
kernel: [ 6391.532400]  smart373-spk: ASoC: trigger FE cmd: 3 failed: -22
[... 80 lines cut ...]
kernel: [ 6392.570839] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
kernel: [ 6392.571024] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
kernel: [ 6392.571231] sof-audio-pci 0000:00:1f.3: pcm: close stream 3 dir 1
kernel: [ 6392.650087] sdw_disable_stream: subdevice #0-Playback: inconsistent state state 4
kernel: [ 6392.650100]  SDW1-Playback: sdw_trigger trigger 0 failed: -22
kernel: [ 6392.650106]  SDW1-Playback: ASoC: error at snd_soc_link_trigger on SDW1-Playback: -22
kernel: [ 6392.650111]  SDW1-Playback: ASoC: trigger BE failed -22
kernel: [ 6392.650117]  smart373-spk: ASoC: trigger FE cmd: 0 failed: -22
kernel: [ 6392.650138] sof-audio-pci 0000:00:1f.3: pcm: free stream 2 dir 0
kernel: [ 6392.650150] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
kernel: [ 6392.650420] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
kernel: [ 6392.652381] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
kernel: [ 6392.652519] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
kernel: [ 6392.652703] sof-audio-pci 0000:00:1f.3: pcm: close stream 2 dir 0
kernel: [ 6392.928091] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
kernel: [ 6392.928104] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
kernel: [ 6393.428078] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
kernel: [ 6393.428087] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG

```
</details>

[dmesg.txt](https://github.com/thesofproject/sof/files/5740870/dmesg.txt)
[logger trace.txt](https://github.com/thesofproject/sof/files/5740872/logger.trace.txt)
[logger error trace.txt](https://github.com/thesofproject/sof/files/5740875/logger.error.trace.txt)

mengdonglin, 2021-01-04:
@slawblauciak This issue may have similar root cause as https://github.com/thesofproject/sof/issues/3673

aiChaoSONG, 2021-01-05:
The commit https://github.com/thesofproject/linux/commit/2dbd71d9dc309e29720d07b3b5090cd3b1829468 increase the possibility to reproduce the issue.

aiChaoSONG, 2021-02-04:
issue always happens when we are doing pause-resume simultaneously on smart-373-spk(PCM2P), and echo((PCM3C)) pipelines. Here is how they are connected: 
[image]

plbossart, 2021-02-10:
I filed an issue for the firmware here: https://github.com/thesofproject/sof/issues/3837

marc-hb, 2021-02-11:
@plbossart is this issue different from #3837 and if yes how?

plbossart, 2021-02-11:
> @plbossart is this issue different from #3837 and if yes how?

#3714 does not suggest any root cause. 
I believe #3837 shows a problem that needs to be fixed. I can't tell you if #3837 also fixes this issue. I prefer adding separate issues with clear topics, a blanket 'underflow on machine Blah' bug report doesn't help progress IMHO.

marc-hb, 2021-02-12:
Thanks for the clarification. So let's hope fixing #3837 fixes this one too but let's keep it open in case it's different.

keyonjie, 2021-03-01:
@aiChaoSONG can you help to check if it reproduced in I2S smart_amp case?

keyonjie, 2021-03-01:
@slawblauciak can you help to check it?

aiChaoSONG, 2021-03-01:
> @aiChaoSONG can you help to check if it reproduced in I2S smart_amp case?

We have smart_amp enabled in TGL_RVP_NOCODEC, and we didn't see this issue on it. check our daily report ID 2367

aiChaoSONG, 2021-03-02:
@slawblauciak  
- build topology with sof/scripts/build-tools.sh, put it to `/lib/firmware/intel/sof-tplg/`
- build sof with sof/scripts/xtensa-build-all.sh (xcc compilier), and put FW to `/lib/firmware/intel/sof/community/`， put ldc file to `/etc/sof`
- clone sof-test to your device:  `git clone https://github.com/thesofproject/sof-test.git`
- run sof-test/env-check.sh to check requirements,
- install `expect python3-graphviz` with apt
- go to test case folder: `cd sof-test/test-case`, and run `TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-sdw-max98373-rt5682-4ch.tplg ./multiple-pause-resume.sh` (sof-logger should be in PATH, eg /usr/bin)

This test case is actually pause-resume two pipelines simultaneously,  PCM2P and PCM3C, check picture in above comment https://github.com/thesofproject/sof/issues/3714#issuecomment-773092410, this can be difficult for manual test, because you cannot have two terminal to have UI focus simultaneously. so two remote terminals are required to manually reproduce the issue.

slawblauciak, 2021-03-16:
I've been unable to reproduce the issue:

Platform: sh-tglu-volt-sdw-03 (from CI)
Firmware: fa35ee00f29cc4cf1c3b6f550e65329dbbc66811
Kernel: Linux version 5.12.0-rc2-daily-default-20210316 (jenkins@sofbld03.ostc.intel.com) (gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0, GNU ld (GNU Binutils for Ubuntu) 2.30) #1 SMP Mon Mar 15 21:19:51 UTC 2021
(don't know the exact commit, I did not change the kernel on that CI platform)

How I tried to reproduce the problem:
`sof-test/test-case/multiple-pause-resume.sh -t /lib/firmware/intel/sof-tplg/sof-tgl-sdw-max98373-rt5682.tplg`

/lib/firmware/intel/sof-tplg/sof-tgl-sdw-max98373-rt5682.tplg is a symlink pointing to /lib/firmware/intel/sof-tplg/sof-tgl-sdw-max98373-rt5682-4ch.tplg

plbossart, 2021-03-16:
@slawblauciak this happens in every daily test, see Intel test 2626 for example

the command was :
````bash
TPLG=sof-tgl-sdw-max98373-rt5682-4ch.tplg ~/sof-test/test-case/multiple-pause-resume.sh -r 25
````

````
[ 6394.265526] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[ 6394.265555] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 6394.265563] kernel:  Amplifier Reference: ASoC: trigger FE cmd: 0 failed: -22
[ 6394.265958] kernel: sdw_deprepare_stream: subdevice #0-Capture: inconsistent state state 3
[ 6394.379353] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60060000 size 12
[ 6394.379382] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 6394.379389] kernel:  Speaker: ASoC: trigger FE cmd: 3 failed: -22
[ 6394.380157] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60070000 size 12
[ 6394.380172] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 6394.380181] kernel:  Speaker: ASoC: trigger FE cmd: 4 failed: -22
[ 6394.380559] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60060000 size 12
[ 6394.380593] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 6394.380598] kernel:  Speaker: ASoC: trigger FE cmd: 3 failed: -22
[ 6394.381129] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60070000 size 12
[ 6394.381144] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 6394.381155] kernel:  Speaker: ASoC: trigger FE cmd: 4 failed: -22
[ 6394.989606] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60060000 size 12
[ 6394.989645] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 6394.989654] kernel:  Speaker: ASoC: trigger FE cmd: 3 failed: -22
[ 6394.989826] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60070000 size 12
[ 6394.989833] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 6394.989840] kernel:  Speaker: ASoC: trigger FE cmd: 4 failed: -22
[ 6394.990197] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60060000 size 12
[ 6394.990229] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 6394.990236] kernel:  Speaker: ASoC: trigger FE cmd: 3 failed: -22
[ 6394.990451] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60070000 size 12
[ 6394.990464] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 6394.990472] kernel:  Speaker: ASoC: trigger FE cmd: 4 failed: -22
[ 6395.497121] kernel: sdw_disable_stream: subdevice #0-Playback: inconsistent state state 4
[ 6395.497137] kernel:  SDW1-Playback: sdw_trigger trigger 0 failed: -22
[ 6395.497148] kernel:  SDW1-Playback: ASoC: error at soc_link_trigger on SDW1-Playback: -22
[ 6395.497157] kernel:  Speaker: ASoC: trigger FE cmd: 0 failed: -22
````

````
[12787435262.239584] (       21.354166) c0 alh-dai      4.258  src/drivers/intel/alh.c:28   alh_trigger() cmd 3
[12787435289.114584] (       26.875000) c0 ll-schedule        ./schedule/ll_schedule.c:399  task add 0xbe080180 pipe-task 
[12787435307.812500] (       18.697916) c0 ll-schedule        ./schedule/ll_schedule.c:403  task params pri 0 flags 0 start 0 period 1000
[12787435333.177084] (       25.364584) c0 ll-schedule        ./schedule/ll_schedule.c:309  new added task->start 25100905
[12787435351.197918] (       18.020834) c0 ll-schedule        ./schedule/ll_schedule.c:312  num_tasks 4 total_num_tasks 4
[12787436001.614584] (      650.416687) c0 buffer       3.16        src/audio/buffer.c:231  comp_update_buffer_produce(), no bytes to produce, source->comp.id = 20, source->comp.type = 18, sink->comp.id = 13, sink->comp.type = 21
[12787436016.822918] (       15.208333) c0 buffer       4.21        src/audio/buffer.c:231  comp_update_buffer_produce(), no bytes to produce, source->comp.id = 20, source->comp.type = 18, sink->comp.id = 19, sink->comp.type = 1
[12787436034.218750] (       17.395834) c0 buffer       4.22        src/audio/buffer.c:271  comp_update_buffer_consume(), no bytes to consume, source->comp.id = 23, source->comp.type = 2, sink->comp.id = 20, sink->comp.type = 18
[12787436056.562500] (       22.343750) c0 host         4.19          src/audio/host.c:376  no bytes to copy, 4 avail in buffer, 384 free in DMA
[12787438067.083334] (     2010.520874) c0 dw-dma                 src/drivers/dw/dma.c:1068 dw_dma_free_data_size() size is 0!
[12787438085.833334] (       18.750000) c0 dai          3.17           src/audio/dai.c:868  WARN dai_copy(): nothing to copy
[12787440047.604168] (     1961.770874) c0 buffer       3.16        src/audio/buffer.c:271  comp_update_buffer_consume(), no bytes to consume, source->comp.id = 20, source->comp.type = 18, sink->comp.id = 13, sink->comp.type = 21
[12787440063.385418] (       15.781250) c0 buffer       3.14        src/audio/buffer.c:271  comp_update_buffer_consume(), no bytes to consume, source->comp.id = 12, source->comp.type = 1, sink->comp.id = 13, sink->comp.type = 21
[... cut ...]
[12790900958.750000] (       18.177084) c0 alh-dai      4.258  src/drivers/intel/alh.c:38   alh_set_config() config->format = 0x   0
[12790900978.177084] (       19.427084) c0 dai                           src/lib/dai.c:191  dai_put type 4 index 258 new sref 1
[12790901007.812500] (       29.635416) c0 dai          3.17           src/audio/dai.c:904  dai_config() dai type = 4 index = 258
[12790901028.177084] (       20.364584) c0 dai          3.17           src/audio/dai.c:965  dai_config(), channel = 65535
````

slawblauciak, 2021-03-18:
Yeah, the "-r 25" bit was the key part to reproducing this.
So, is the topology for this case available somewhere? Because the sof-tgl-sdw-max98373-rt5682 topology in the FW repo doesn't match the pipeline IDs visible in the log. The DSM should be on pipelines 3 and 4, but in the topology in the FW repo it's on pipelines 1 and 2.

plbossart, 2021-03-18:
@slawblauciak we are using Topology: {sof-tgl-sdw-max98373-rt5682-4ch.tplg}
This is the standard topology without an actual DSM.

See attached what I pulled from test 2626 
[sof-tgl-sdw-max98373-rt5682-4ch.zip](https://github.com/thesofproject/sof/files/6164184/sof-tgl-sdw-max98373-rt5682-4ch.zip)

slawblauciak, 2021-03-18:
@plbossart where can I find the m4 for it?
Because I don't think it's based on the on in FW repository, given the pipeline ID mismatch

plbossart, 2021-03-18:
we are using the master branch https://github.com/thesofproject/sof/blob/master/tools/topology/sof-tgl-sdw-max98373-rt5682.m4

slawblauciak, 2021-03-19:
It appears that the feedback line is is not receiving enough data and is getting starved. Xrun is reported by DMA when we check for available data. Looking at the thing, it looks very similar if not basically identical to another problem we've detected with DSM working on SSP:
It was caused by SSP clock gating, when the playback line was shutdown, the SSP clock was gated. But, feedback was using the same SSP port and was still running for a short while. As a result, it was starved. I suspect the same thing might be happening here.

@plbossart how are SDW clocks managed on the kernel side? Is there a possibility a similar scenario might be happening here, where kernel is gating the clock on the port after seeing the playback pipeline stopping?

RanderWang, 2021-03-22:
@slawblauciak capture and playback don't share the same port in both master and slave. clock is off only when system set audio driver to suspend mode.

lgirdwood, 2021-05-10:
@aiChaoSONG  still an issue ?

aiChaoSONG, 2021-05-10:
@lgirdwood Still seen in daily report, let's keep this issue open

lgirdwood, 2021-05-20:
@aiChaoSONG @keyonjie does this also need a kernel fix ?

lgirdwood, 2021-05-26:
@kv2019i could this be what you are seeing too with Zephyr ?

marc-hb, 2021-05-26:
I'm guessing @lgirdwood is referring to (duplicate?) #4250

keyonjie, 2021-05-27:
> @aiChaoSONG @keyonjie does this also need a kernel fix ?

I think @slawblauciak is working on this from the FW side.

kv2019i, 2021-05-27:
@lgirdwood #4250 could be related. The problem is triggered in the same scenario (simultaneous pause), but with Zephyr build, the results are a bit different:
 - one of the pipelines is always DMIC
 - I'm using a tplg with SDW (sof-adl-rt711-4ch.tplg)
 - reproduction rate is much higher, ./multiple-pause-resume.sh with default params fails every time
 - even "./multiple-pause-resume.sh  -r 2 -l 1" failure rate is close to 100%

slawblauciak, 2021-05-31:
Having hard time reproducing this on new FW, I'm encountering this instead:
https://github.com/thesofproject/linux/issues/2805

keyonjie, 2021-06-02:
This looks not available for 1.8, have to move it to 1.9

keyonjie, 2021-06-03:
@aiChaoSONG @slawblauciak looks this is not reproduced according to the latest daily report 4346?

plbossart, 2021-06-30:
@XiaoyunWu6666 this is not tracked in Intel daily tests and you've added https://github.com/thesofproject/sof/issues/4434

closing this one for clarity.

closed 2021-06-30 by plbossart