thesofproject/sof #3717: [BUG] [Realtek ALC298] No sound from internal speakers

lo-leakage, 2020-12-28:
**Describe the bug**
No sound from internal speakers. ( headset is ok ) 

I tried the test below 
1) options snd-intel-dspcfg dsp_driver=1 : no sound 
2) options snd-intel-dspcfg dsp_driver=3 : no sound 
3) options snd-hda-intel dmic_detect=0 and options snd-hda-intel models : no sound 

I have tested on windwos10 : ok, windows driver send some i2c command data for enable maxim sound amp. 

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
100% 

**Expected behavior**
Working internal speakers 

**Impact**
Can't playback audio

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.10.0-rc1 
    * SOF: master and v1.6 tag 
2) Name of the topology file
    * Topology: sof-hda-generic-2ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TigerLake Platform

**Screenshots or console output**
ALSA info : http://alsa-project.org/db/?f=7ba6e11f4fa1cfad5518a32bfcaaba5f5aff5aff
RtHDDump : [RtHDDump_V249.txt](https://github.com/thesofproject/sof/files/5745754/RtHDDump_V249.txt)
dmesg : [dmesg.log](https://github.com/thesofproject/sof/files/5745755/dmesg.log)
lspci : [lspci-vnvn.log](https://github.com/thesofproject/sof/files/5745756/lspci-vnvn.log)

mythi, 2020-12-31:
> No sound from internal speakers. ( headset is ok )

FWIW. I had the same issue on my Asus/Tigerlake but moving to 5.11-rc1 makes the speakers working (`speaker-test`).

lo-leakage, 2021-01-04:
Thanks for your comment. 
 I just tested it, but no luck. still no sound from internal speakers.

lgirdwood, 2021-01-05:
@lo-leakage do you have any information on the i2c commands sent to maxim amp ? It looks like this is missing from the Linux driver. @plbossart @kv2019i @bardliao fyi.