thesofproject/sof #3724: [BUG] memory allocation failure in mux/demux pipeline

libinyang, 2021-01-05:
**Describe the bug**
When I tried to make smart amp work on CML stradale. I found the memory allocation failed.
This bug is a regression. And it is introduced by commit: 5d763d6

**To Reproduce**
Use the update topology which include mux/demux pipelines, it will fail to initialize the audio module.

**Reproduction Rate**
100%

**Expected behavior**
audio initialized successfully

**Impact**
audio doesn't work

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: latest kernel on https://github.com/thesofproject/linux/tree/topic/sof-dev with patch https://github.com/thesofproject/linux/pull/2514
    * SOF: https://github.com/thesofproject/sof/tree/master a2ce9d7
2) Name of the topology file
    * Topology: my local toplogy. If needed, I can upload this topology.
    * 
3) Name of the platform(s) on which the bug is observed.
    * Platform: CML, TGL (not test on other platforms, but suppose other platform have the same issue)

**Screenshots or console output**
dmesg:
[    1.814256] sof-audio-pci 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[    1.814403] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x30010000 size 20
[    1.814404] sof-audio-pci 0000:00:1f.3: error: create process failed
[    1.814405] sof-audio-pci 0000:00:1f.3: error: process loading failed
[    1.814406] sof-audio-pci 0000:00:1f.3: error: DSP failed to add widget id 0 type 33 name : MUXDEMUX10.1 stream none reply 0
[    1.814408] sof-audio-pci 0000:00:1f.3: ASoC: failed to load widget MUXDEMUX10.1
[    1.814409] sof-audio-pci 0000:00:1f.3: ASoC: topology: could not load header: -22
[    1.814419] sof-audio-pci 0000:00:1f.3: error: tplg component load failed -22
[    1.814423] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[    1.814424] sof-audio-pci 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    1.814425] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
[    1.814437] sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[    1.814505] sof_sdw sof_sdw: snd_soc_register_card failed -22

sof-logger:
[  2251877.656250] (       10.729167) c0 mux                     src/audio/mux/mux.c:168  mux_new()
[  2251910.208333] (       32.552082) c0 mux          10.43      src/audio/mux/mux.c:100  mux_set_values()
[  2252090.520833] (      180.312500) c0 component             src/audio/component.c:130  comp new 0x1fffa390U type 18 id 10.44
[  2252101.458333] (       10.937500) c0 mux                     src/audio/mux/mux.c:168  mux_new()
[  2252126.927083] (       25.468750) c0 demux                 src/audio/component.c:135  ERROR comp_new(): unable to create the new component
[  2252138.802083] (       11.875000) c0 ipc                           src/ipc/ipc.c:186  ERROR ipc_comp_new(): component cd = NULL
[  2252151.562500] (       12.760417) c0 ipc                       src/ipc/handler.c:1131 ERROR ipc: pipe 10 comp 44 creation failed -22

more log please check the attachment.

please check the attachment for the topology.

libinyang, 2021-01-05:
sof logger
[soflog.txt](https://github.com/thesofproject/sof/files/5768874/soflog.txt)

tplg
[image]

lgirdwood, 2021-01-05:
@libinyang it's likely we are asking for a block size that is no longer available.

libinyang, 2021-01-06:
> @libinyang it's likely we are asking for a block size that is no longer available.

@lrgirdwo Yes, after some simple debug, it fails at allocating memory

libinyang, 2021-01-06:
upload the tplg file.
[Uploading tplg.zip…]()

libinyang, 2021-01-06:
upload the tplg
[tplg.zip](https://github.com/thesofproject/sof/files/5773719/tplg.zip)

lgirdwood, 2021-01-06:
> > @libinyang it's likely we are asking for a block size that is no longer available.
> 
> @lrgirdwo Yes, after some simple debug, it fails at allocating memory

Then check the allocator block map and increase the heap.

closed 2021-06-03 by lgirdwood