thesofproject/sof #3743: [BUG] ALSA firmware site outdated

cujomalainey, 2021-01-14:
**Describe the bug**
ALSA firmware page holds stale outdated content on SOF
https://www.alsa-project.org/wiki/Firmware

We should delete or update this content to point to here or the github.io site

lgirdwood, 2021-02-09:
@perexg apologies, I've cant remeber the spam token to update the ALSA wiki. I want to update it to
```

== '''Sound Open Firmware''' ==

Sound Open Firmware is an open source audio DSP firmware and SDK intended to provide a generic audio firmware infrastructure and development tools to the community. The firmware is BSD licensed and is platform and architecture independent. The SDK is also open source and provides a toolchain, debugger, emulator and firmware image creation tools.

The firmware and SDK are intended for developers who are interested in audio or signal processing on modern DSPs or for developers who are interested in micro kernels running on small but 
powerful processors.

SOF currently targets the Cadence xtensa architecture DSPs found on some Intel and NXP based devices like the Minnowboard MAX. SOF is modular and generic so can be ported to other DSP architectures or host platforms.  

Please see https://thesofproject.github.io/latest/index.html for full details.
```

perexg, 2021-02-09:
Updated.

closed 2021-02-09 by cujomalainey