thesofproject/sof #3753: [BUG] Microphone fails to be detected on Thinkpad X1 Carbon 7th Generation

cinerea0, 2021-01-17:
**Describe the bug**
The built-in microphone in my Thinkpad X1 Carbon 7th Generation (X1C7) fails to be detected under PulseAudio even when using SOF firmware. Setting "options snd-intel-dspcfg dsp_driver=1" allows audio to be played but does not allow for microphone use. Setting "options snd-intel-dspcfg dsp_driver=3" does not allow either audio to be played or recorded.

**To Reproduce**
On a Thinkpad X1C7, install Void Linux, install the necessary ALSA and PulseAudio packages, and attempt to use the built-in microphone. I understand that this bug may be very hardware-specific, which may reduce the ability of others to reproduce this bug.

**Reproduction Rate**
This bug happens all the time.

**Expected behavior**
When using PulseAudio 14.0, SOF firmware 1.6.1, and alsa-ucm-conf 1.2.4, the microphone on this model of laptop should work without any further configuration.

**Impact**
This is a minor problem, since I can use an external microphone, but I would much prefer to be able to use the internal microphone because it's much more convenient.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: a60d1a8fea7579778cac14894e360a12365100e2 on branch v5.9.16
    * SOF: b77c851bc4ec1b6b552eaf1a61a66f3df4a13ab8 on branch stable-v1.6.1 in the sof-bin repository
2) Name of the topology file
    * Topology: sof-cnl-v1.6.1.ri
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel i7-10510U CPU with a glibc Linux system (not sure what was meant by 'platform')

**Screenshots or console output**
Output of `dmesg | grep audio`:
```
[    2.555127] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC285: line_outs=2 (0x14/0x17/0x0/0x0/0x0) type:speaker
[    2.555128] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    2.555128] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    2.555129] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[    2.555130] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[    2.555130] snd_hda_codec_realtek hdaudioC0D0:      Mic=0x19
[    3.756789] Modules linked in: ccm algif_aead des_generic libdes algif_skcipher cmac nf_tables nfnetlink md4 algif_hash af_alg nls_iso8859_1 nls_cp437 vfat fat uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev btusb btrtl mc btbcm btintel joydev hid_multitouch hid_generic iTCO_wdt intel_pmc_bxt iTCO_vendor_support mei_hdcp i2c_designware_platform i2c_designware_core intel_rapl_msr x86_pkg_temp_thermal intel_powerclamp coretemp snd_hda_codec_hdmi kvm_intel kvm iwlmvm irqbypass snd_hda_codec_realtek rapl snd_hda_codec_generic mac80211 intel_cstate snd_hda_intel psmouse libarc4 input_leds snd_intel_dspcfg snd_hda_codec intel_wmi_thunderbolt iwlwifi snd_hda_core wmi_bmof i2c_i801 snd_hwdep i2c_smbus mei_me intel_lpss_pci snd_pcm intel_lpss cfg80211 mei idma64 virt_dma processor_thermal_device intel_soc_dts_iosf intel_pch_thermal intel_rapl_common thermal thinkpad_acpi ledtrig_audio int3403_thermal int340x_thermal_zone ac int3400_thermal acpi_thermal_rel evdev
```

Output of `amixer contents` since I thought it was strange:
```
numid=4,iface=MIXER,name='Master Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=3,iface=MIXER,name='Master Playback Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=16351,16351
numid=2,iface=MIXER,name='Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=off
numid=1,iface=MIXER,name='Capture Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=65536,65536
```

Full dmesg log: [dmesg.log](https://github.com/thesofproject/sof/files/5825670/dmesg.log)

Full output of alsa-info.sh: [alsa_info.txt](https://github.com/thesofproject/sof/files/5825674/alsa_info.txt)

juimonen, 2021-01-18:
@cinerea0 so in you dmesg logs you don't have sof driver loaded, but the legacy hda driver. Could you try to force sof and re-generate the logs?

juimonen, 2021-01-18:
so that is  = options snd-intel-dspcfg dsp_driver=3