thesofproject/sof #3755: No microphone detected when using headphones on XPS 17 9700

sergiocaccamo, 2021-01-18:
I have Ubuntu 20.10 kernel 5.10 oem, the speakers and the microphone now seem to work on the dell xps17 9700, which is good.
As soon as I plug the headphone in the jack the audio is correctly redirected to the headphones but the microphone stops responding. 
If I look at the input, I can only select the input device "Soundwire microphone - sof-soundwire" but the microphone does nothing.
To make the microphone work again, I have to unplug the headphones and open again the sound panel (or use alsaucm reload).

Reproduce it by simply plug some headphones with microphone.

juimonen, 2021-01-18:
Do you get any pop-up window when you plug in the headphones? With headphones I understand it is a just headphone without microphone? Dell devices usually can't detect if you plug in headset (with mic) or headphone, so it might happen here that the device thinks you have plugged in a headset and tries to use the "imaginary" microphone. So what I meant by pop-up is that usually dell devices ask you did you plug headset or headphone so that the routing can be done correctly...

Anyway it could be also something different, but I guess you should see at least the 2 choices in the input device list.

My suspicion is there's something wrong with the UCM... could you try to perhaps update alsa-ucm?

juimonen, 2021-01-18:
@bardliao FYI

juimonen, 2021-01-18:
and @jason77-wang FYI also

sergiocaccamo, 2021-01-18:
I have tried now 3 headphones. 
1 beats : audio and microphone works
2 sony (bot wired, wh-1000xm3, wh-h910n): speaker working, microphone not.
the sony work on another laptop with ubuntu 14.04 and they work fine in bluetooth mode.

I have update ucm2 and tried to play a little with the configs, these are mine:

Speakers.conf
`#<sof-soundwire/RT700.conf>
<sof-soundwire/RT1308.conf>
#<sof-soundwire/RT711.conf>
#<sof-soundwire/RT5682.conf>
<sof-soundwire/RT715.conf>
#<sof-soundwire/Hdmi.conf>`

Heaphones.conf
`<sof-soundwire/RT700.conf>
#<sof-soundwire/RT1308.conf>
<sof-soundwire/RT711.conf> #only
#<sof-soundwire/RT5682.conf>
#<sof-soundwire/RT715.conf>
#<sof-soundwire/Hdmi.conf>`

juimonen, 2021-01-18:
at least the wh-h910n help guide reads that it doesn't route the microphone in the headset through the cable (I guess doesn't have 3 rings in it) -> so the use case should be as "connected headphone, using laptop microphone". So it is probably exactly the case I'm talking above...

juimonen, 2021-01-18:
@sergiocaccamo but you're not getting any pop-ups when connecting the devices with cable?

sergiocaccamo, 2021-01-18:
@juimonen no, unfortunately no pop-ups when connecting the headphones.
Also, that's what I am trying to achieve, audio from headphones but microphone from laptop. The problem is that as soon as I plug it in the headphone the microphone dies.

juimonen, 2021-01-18:
@sergiocaccamo ok good we are on the same page then 👍 

Let's wait for the others to chime in, I might have quite similar device at hand, but I have not worked with soundwire too much.

sergiocaccamo, 2021-01-18:
@juimonen thank you,sure let's wait.
I have used different ucm2 settings now but with little success. Googling around it seems I am not the only one having this issue.