thesofproject/sof #3767: [BUG] TGL topology files don't always have 4 HDMI/DP pipelines

plbossart, 2021-01-20:
**Describe the bug**
we seem to reuse the same topology files for ICL and TGL in HDaudio and SoundWire configurations, but TGL has 4 display pipelines, not 3.

examples: sof-icl-rt711-rt1308-rt715-hdmi.m4 sof-tgl-rt715-rt711-rt1308-hdmi.m4 sof-hda-generic-idisp.m4 sof-hda-generic.m4

**Expected behavior**
TGL topologies should expose all 4 pipelines. For HDaudio this may require a different matching table in the kernel to use a different topology?

@kv2019i @bardliao @RanderWang FYI

kv2019i, 2021-01-21:
We have this for generic HDA driver:
https://github.com/thesofproject/linux/issues/1639

For any new topologies, 4 iDisp should be exposed for sure, but for old ones, it requires some compatibility layers to do. Pierre you had a proposal in #1639 but I haven't yet looked at how to do it.

Note, this is only needed if you have four concurrent monitors all with audio capability, so a bit of a corner case in terms of requirements.

plbossart, 2021-01-21:
@kv2019i I think as long as the machine driver advertises support for 4 HDMI links, the topology can use 3 or 4. The problems come if the topology uses MORE resources than what the machine driver shows.

I pushed https://github.com/thesofproject/linux/pull/2709 to fix all this, we can deal with a topology update later.

lgirdwood, 2021-06-03:
@plbossart @kv2019i @slawblauciak any update - can we close ?

plbossart, 2021-06-03:
We haven't made any progress, but I am not sure if we should really care. 
IIRC the 4th HDMI link seems to be for an embedded display so the odds of having audio on that display are rather low.

kv2019i, 2021-06-03:
Ack @plbossart , let me take this on my name. The priority is appropriate for now. In most systems, the 4th pipe is reserved for the embedded display (even when it's not in use), so in practise you are limited to at max 3 concurrent pipes.

plbossart, 2022-07-20:
In hindsight it was a mistake to introduce 4 streams. We should keep existing topologies as is and move on.

closed 2022-07-20 by plbossart