thesofproject/sof #3795: Discussion: Keyword Detection Feature

dbaluta, 2021-01-28:
Hi @mrajwa @keyonjie @bkokoszx 

We are trying to integrate KWD with a custom keyword detection algorithm and we have some questions, hopefully we can help us:

[image]

Questions:

- our keyword detection algorithm needs 1s of data to work, is that possible with the current implementation?
-

mrajwa, 2021-01-28:
Hi @dbaluta,

1. Sure, the current solution can easily provide you about two seconds of history data (depending on the platform)
2. Upon detection KPB component starts to drain history data to the host component from that point the data are further copied and finally they end up in the user space application. 
3. We have a dummy detector component which mimics the actual algorithm - on the flow chart you pasted in your question it is called "Keyphrase detection algorithm", in the FW it is detect_test.c

PS: better to use number to enumerate questions ;)

paulstelian97, 2021-01-28:
> Hi @dbaluta,
>
> 1. Sure, the current solution can easily provide you about two seconds of history data (depending on the platform)
> 2. Upon detection KPB component starts to drain history data to the [...]

Our question is how does the detect_test/kwd component actually use the history inside the KPB component in its own detection algorithm.

mrajwa, 2021-01-28:
> Our question is how does the detect_test/kwd component actually use the history inside the KPB component in its own detection algorithm.

It doesn't. Its only purpose is to detect sth in the real time stream.

dbaluta, 2021-01-28:
>     1. Sure, the current solution can easily provide you about two seconds of history data (depending on the platform)

Is the history data only to be drained to Host component? We would like that the history buffer to also be accessible from our custom detection component (Keyphrased detection algorithm).

Is there any elegant way to follow in order for our custom component to buffer 1s of data?
We could have:
- internal buffering
- copy the data from KPBM somehow

mrajwa, 2021-01-28:
Yes the history data are stored for the HOST component to use. You can place some parser of history data in the KPB component maybe....the general idea is-> detector perform light weight / raw analyze of real time stream. If it senses sth it sends a notification to the KPB to drain history data of X seconds to host where that data is further analyzed.

paulstelian97, 2021-01-28:
The annoying part is that our current algorithm is still single channel so a selector is needed BUT also the host needs to receive dual channel for the long-term solution.

The NXP DSPs seem to be kinda beefy (640-800MHz depending on model) so they will take higher loads than the others. We will however still analize how well the more complex algorithms work (measure MCPS and everything) to see if we need to simplify them or not.

History buffer not being contiguous areas of memory might be interesting but I believe the algorithm can be adapted to deal with that.

Do you have any ideas to also communicate to the userspace app what keyword was detected? Current architecture is just having the DSP say that something was detected and re-running on the host if I get it correctly.

cristinailie, 2021-01-28:
Hi mrajwa,
If KPBM draining was thought only for the host than why in the code of KPBM is an array with size=no_of_clients, a register function which basically registers the relevant data of a client to KPBM component (like the client's sink for further data transfers) and why kpb_init_draining function offers an API [1] which receives as input the kpb_client data? For example the history_depth parameter used in draining is read from kpb_client structure, so a client can require an amount of data from history_buffer from KPBM.  It's true that at this moment the client is hardcoded to host since the sink buffer is not read from kpb_client structure. However it seems that was projected for multiple clients but it's not finished and now it is used only for the host. 

Do you agree with me? If yes, do you know if is there another branch which implements fully this functionality of multiple clients connecting to KPBM? 

[1] `static void kpb_init_draining(struct comp_dev *dev, struct kpb_client *cli)`

dbaluta, 2021-01-29:
I guess this won't be a problem for us.

> * advanced analyse on DSP side may affect perfrormance significantly

In our integration DSP clocks at 600-800Mhz

> * history buffer may not be continuous area of memory

We have access to a shared DRAM memory area that we can virtually make it very large. We use 8MB now but it can be extended. 

The real thing we want to figure out is that the current design allows us in a clean way to to accumulate 1s of data inside KWD component and do keyword processing on it.

mrajwa, 2021-01-29:
> The real thing we want to figure out is that the current design allows us in a clean way to to accumulate 1s of data inside KWD component and do keyword processing on it.

Then I see no problem processing that data on the DSP side

paulstelian97, 2021-01-29:
@mrajwa About the multiple clients part... Don't we need multiple clients, namely the algorithm and the host? In particular the algorithm should peek at the data in the buffer (copy without consuming) while the host actually consumes it.

mrajwa, 2021-01-29:
@paulstelian97, no. Consumers are on a different level, its more about user space clients like google, alexa etc... I have already said what I think you should do - place your algorithm inside KPB component so it can analyze whole history buffer on regular periods.

paulstelian97, 2021-01-29:
> @paulstelian97, no. Consumers are on a different level, its more about user space clients like google, alexa etc... I have already said what I think you should do - place your algorithm inside KPB component so it can analize whole history buffer on regular periods.

So, basically: we need to create a new topology which has just a single pipeline with a modified KPB component that has the algorithm integrated into it as well, and the state changes are provoked internally. Of course we still keep the regular wakeup thingy because I think adding new IPCs to actually communicate _which_ keyword was detected is probably **not** the way to go.

dbaluta, 2021-01-29:
@paulstelian97 I think this is the first option we have for now. Maybe there's a better solution which we can look at it later. 

Thanks a lot @mrajwa for your inputs!

cristinailie, 2021-02-01:
@mrajwa I followed your advice to integrate my algorithm directly in kpb. My question is now - how to trigger the keyword detection event now? I mean, could I just skip the event handling and just to call kpb_init_draining when my condition for keyword detection is accomplished? Or should I simulate that the event BEGIN_DRAIN was sent? (I don't know if this is possible from the same SOF component).

lgirdwood, 2021-02-05:
@cristinailie IIRC, there is a dummy component where you can just clap to simulate the keyword.

paulstelian97, 2021-02-08:
> @cristinailie IIRC, there is a dummy component where you can just clap to simulate the keyword.

That's what we used before, but the problem is that the component cannot intercept the data. We want another intermediate test where we still don't trust our mic's data but also we get as close as possible to how the integration itself will look like.

lgirdwood, 2021-02-09:
@paulstelian97 sound like you may need something different here, maybe a component that consume data until a kcontrol event signals that it's deteced the keyword.

dbaluta, 2021-03-16:
Closing this as we mostly figured out our way in implementing this.

closed 2021-03-16 by dbaluta