thesofproject/sof #3809: [BUG IMX] Playback and capture occasionally fails

iuliana-prodan, 2021-02-02:
**Describe the bug**
Playback and capture occasionally fails with Input/output error

**To Reproduce**
Run playback and capture simultaneously.
Or run the script below:
```
#!/bin/bash

test_aplay()
{
    while true
    do
        aplay -Dhw:1,0 -f S32_LE -c 2 -r 48000 -t raw /dev/urandom
    done
}

test_arecord()
{
    while true
    do
        arecord  -Dhw:1,0 -f S32_LE -c 2 -r 48000 -t raw /dev/null &
    done
}

test_aplay &
test_arecord
```

**Reproduction Rate**
20%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.10
    * SOF: master,  imx-stable-v1.5, imx-stable-v1.6.
2) Name of the topology file
    * Topology: sof-imx8qxp-wm8960.tplg, sof-imx8qxp-cs42888-mixer.tplg, sof-imx8mp-wm8960.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform:  i.MX8QXP/i.MX8QM/i.MX8MP

**Screenshots or console output**
```
root@imx8qmmek:~# audio_dsp_sof.sh -t 4 -c wm8960 -f p -n 50
DSP_SOF_PLAY_RECORD_WORK_TOGETHER    0  TINFO  :  test 4: DSP_SOF_PLAY_RECORD_WORK_TOGETHER 
TINFO  :  Do full hardware capacity test 
TINFO  :  Do full hardware capacity test 
TINFO  :  Start audio test, please check the HEADPHONE,hear if there is voice.
TINFO  :  Audio TEST: Total 3 combinations of formats. 
TINFO  :  No.[1] Now playing audio stream with format S16_LE, channel 2,sample rate 48000
TINFO  :  Test #1: ADC record test. Headphone is need to check the voice and quality of recording.
TINFO  :  Audio TEST: Total 3 combinations of formats. 
TINFO  :  No.[1] Now Recording audio stream with format S16_LE, channel 2,sample rate 48000
TINFO  :  No.[2] Now playing audio stream with format S24_LE, channel 2,sample rate 48000
TINFO  :  No.[3] Now playing audio stream with format S32_LE, channel 2,sample rate 48000
TINFO  :  No.[2] Now Recording audio stream with format S24_LE, channel 2,sample rate 48000
TINFO  :  Start audio test, please check the HEADPHONE,hear if there is voice.
TINFO  :  Audio TEST: Total 3 combinations of formats. 
TINFO  :  No.[4] Now playing audio stream with format S16_LE, channel 2,sample rate 48000
TINFO  :  No.[5] Now playing audio stream with format S24_LE, channel 2,sample rate 48000
TINFO  :  No.[3] Now Recording audio stream with format S32_LE, channel 2,sample rate 48000
TFAIL  :  ltpapicmd.c:188: Test #1: Playing back audio48k24b2c_sle.pcm Failed on hw:1,0
Playing raw data '/tmp/tmpfs/tmp.jWsafwNfuh/audio48k24b2c_sle.pcm' : Signed 24 bit Little Endian, Rate 48000 Hz, Stereo
aplay: pcm_write:2061: write error: Input/output error
TINFO  :  No.[6] Now playing audio stream with format S32_LE, channel 2,sample rate 48000
TINFO  :  Test #1: ADC record test. Headphone is need to check the voice and quality of recording.
TINFO  :  Audio TEST: Total 3 combinations of formats. 
TINFO  :  No.[4] Now Recording audio stream with format S16_LE, channel 2,sample rate 48000
TFAIL: Audio Playback test fails in loop time 2
TFAIL: Test Failed on below file: audio48k24b2c_sle.pcm 
TGE_LV_DAC2    1  TFAIL  :  ltpapicmd.c:188: Test FAIL
^CTFAIL: Test #1: Recording audio48k16b2c_sle.pcm Failed on hw:1,0
Recording raw data '/tmp/tmpfs/tmp.Nzsd3l5HpA/audio48k16b2c_sle.pcm' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Aborted by signal Interrupt...
^C^Caplay: no process found
arecord: no process found

root@imx8qmmek:~# ^C
```

closed 2021-03-15 by dbaluta

paulstelian97, 2021-03-16:
Is this issue fixed?

iuliana-prodan, 2021-03-16:
> Is this issue fixed?

Yes, fixed by #3867.
Another issue remains, after hundreds of playback/record we get:
```
Playing raw data '16/j16.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Playing raw data '16/j16.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Recording WAVE 'test_record.pcm' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
[  582.814139] sof-audio-of 556e8000.dsp: error: ipc timed out for 0x60050000 size 12
[  582.821802] sof-audio-of 556e8000.dsp: info: preventing DSP entering D3 state to preserve context
[  582.830906] sof-audio-of 556e8000.dsp: invalid header size 0x2000200. FW oops is bogus
[  582.838989] sof-audio-of 556e8000.dsp: error: unexpected fault 0x00000000 trace 0x00000000
[  582.847530] sof-audio-of 556e8000.dsp: error: waking up any trace sleepers
[  582.854601] sof-audio-of 556e8000.dsp: error: trace IO error
[  582.861198] sof-audio-of 556e8000.dsp: ASoC: error at snd_soc_pcm_component_trigger on 556e8000.dsp: -110
[  582.871466]  Port0: ASoC: trigger FE cmd: 0 failed: -110
Playing raw data '16/j16.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Recording WAVE 'test_record.pcm' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: pcm_write:2061: write error: Input/output error
(null)      1  TFAIL  :  ltpapicmd.c:188: Test 420 Playback failed
Recording WAVE 'test_record.pcm' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Recording WAVE 'test_record.pcm' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
```
But his will be handle in another issue.

paulstelian97, 2021-03-16:
Just checked, sometimes issues are accidentally closed by merging of pull requests on GH (the "fixes" keyword does that)

lgirdwood, 2021-03-16:
@iuliana-prodan is this working in v1.7 ? @keyonjie has created a PR pulling all the fixes ready for final release ? so please let me know if anything is missing ?

iuliana-prodan, 2021-03-16:
@lgirdwood yes, is working on v1.7. But these patches are not included in @keyonjie PR #3914
Can I push these patches on #3914 or @keyonjie will update that PR?

lgirdwood, 2021-03-16:
@iuliana-prodan just send another PR with your patches for v1.7

PR #3867 (2021-03-15): Fix playback and capture multiple runs which occasionally fails with Input/output error
This pull request fixes #3809.

The first commit updates the correct order for DAI and DMA start/stop.
Before this update, on multiple start/stop runs for playback or record or combined,
we get an underrun/overflow.
That's because the DAI makes a DMA request, before the DMA channel is
enabled.

The second commit updates the start/stop operations for SAI.
Please see a more detailed explanation in the commit message.

After the first 3 commits, after multiple playback/capture runs (~700, ~1000)
everything see [... cut]
commit 1fa1001e (2021-03-15): dai: correct the order for DAI and DMA start/stop

To stop/suspend an active DMA channel:
1. Stop the DMA service request at the peripheral first (stop the DAI);
2. Disable the hardware service request on the appropriate DMA channel.

For start/resume:
1. Enable the DMA service request on the appropriate channel;
2. Enable the DMA service request at the peripheral (enable DAI).

When the start/stop order for DMA and DAI is different, on multiple
start/stop runs for playback or record or combined, we get an
underrun/overflow.
That's because the DAI makes a DMA request, before the DMA channel is [... cut]
commit f60adbbb (2021-03-15): drivers: imx: sai: update start/stop operations

For SAI, we have the synchronous mode enabled: the transmitter is
configured for asynchronous operation and the receiver for
synchronous operation.
In this case, transmitter bit clock and frame sync are used by both
the transmitter and receiver. So, when enabling RX we need to enable TX
(if not already enabled).

Therefore, for a clear start, we first do a software reset for the current
direction, but checking the state of RX and TX.
This will reset the internal transmitter/receiver logic including the FIFO
pointers.

For capture we can disable  [... cut]
commit 4ef93fce (2021-03-15): drivers: imx: sai: update start/stop operations

For SAI, we have the synchronous mode enabled: the transmitter is
configured for asynchronous operation and the receiver for
synchronous operation.
In this case, transmitter bit clock and frame sync are used by both
the transmitter and receiver. So, when enabling RX we need to enable TX
(if not already enabled).

Therefore, for a clear start, we first do a software reset for the current
direction, but checking the state of RX and TX.
This will reset the internal transmitter/receiver logic including the FIFO
pointers.

For capture we can disable  [... cut]