thesofproject/sof #3814: [BUG] KWD w/ s24_le format pipeline will stop when DSP enters to D0i3

keqiaozhang, 2021-02-04:
**Describe the bug**
We observed this issue on firmware tag tag:v11.0-hotfix1. When testing wov with S24_LE format on TGL chrome-Volteer SDW, the KWD pipeline will stop when DSP enters to D0i3. but there're no obvious errors in kernel logs or etrace.  
 This issue is not the same as https://github.com/thesofproject/sof/issues/3800 on master branch. For https://github.com/thesofproject/sof/issues/3800, it happens with all audio format(S16_LE/S24_LE and S32_LE) and we have already found the root cause for the regression.  
But for this issue, it only happens with S24_LE format, S16_LE and S32_LE formats work well.

I did the cross check for this issue on TGL-chrome-Volteer SDW and TGL-Chrome-Delbin I2S platforms, test results are:

[image]

Since the tip kernel sof-dev has an i915 regression on Delbin platform, so we are unable to check this issue on Delbin with tip of sof-dev. but for other recipes, kwd works well with all formats.  But for Volteer, this issue happened after TGL-V10 release. so looks like this issue only happens on SDW device.

Keyon suggested me to check this issue with 2 ll scheduler fixes from bkokoszx personal branch:
https://github.com/bkokoszx/sof/commit/8430ed722590ba8e81f16a3e716ea56fd2aaf777
https://github.com/bkokoszx/sof/commit/8430ed722590ba8e81f16a3e716ea56fd2aaf777
So I applied these 2 fixes to https://github.com/thesofproject/sof/releases/tag/releases%2Ftgl%2Fv11.0-hotfix1. Indeed, it can fix this issue on Volteer.  but we still need bkokoszx to check and explain here.
 

**To Reproduce**
$ arecord -Dhw:0,5  -M -N -r 16000 -c 2 -f S24_LE --buffer-size=96000 -d 10 wov.wav -vvv

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev/6fa085f3
    * SOF: tag https://github.com/thesofproject/sof/releases/tag/releases%2Ftgl%2Fv11.0-hotfix1
2) Name of the topology file
    * Topology: sof-tgl-sdw-max98373-rt5682.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL-Chrome-Volteer-sdw

**Screenshots or console output**
```
$ arecord -Dhw:0,5  -M -N -r 16000 -c 2 -f S24_LE --buffer-size=96000 -d 10 /tmp/wov_S16_record.wav -vvv
Recording WAVE '/tmp/wov_S16_record.wav' : Signed 24 bit Little Endian, Rate 16000 Hz, Stereo
Hardware PCM card 0 'sof-soundwire' device 5 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S24_LE
  subformat    : STD
  channels     : 2
  rate         : 16000
  exact rate   : 16000 (16000/1)
  msbits       : 32
  buffer_size  : 96000
  period_size  : 24000
  period_time  : 1500000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 24000
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 96000
  silence_threshold: 0
  silence_size : 0
  boundary     : 6755399441055744000
  appl_ptr     : 0
  hw_ptr       : 0
mmap_area[0] = 0x7f079d43a000,0,64 (32)
mmap_area[1] = 0x7f079d43a000,32,64 (32)
Max peak (48000 samples): 0x001056cf ###                  12%
Max peak (16160 samples): 0x000968eb ##                   7%
Max peak (3360 samples): 0x00053d79 #                    4%
Max peak (3200 samples): 0x000b1561 ##                   8%
Max peak (3200 samples): 0x0008a087 ##                   6%
Max peak (3200 samples): 0x0007f0d8 ##                   6%
Max peak (3200 samples): 0x0005e326 #                    4%
Max peak (3200 samples): 0x00054a92 #                    4%
Max peak (3200 samples): 0x00088850 ##                   6%
Max peak (3360 samples): 0x00099d8d ##                   7%
Max peak (3200 samples): 0x0004e1cd #                    3%
Max peak (2720 samples): 0x0007f117 ##                   6%
Max peak (480 samples): 0x00026589 #                    1%
Max peak (3200 samples): 0x0006c08c ##                   5%
Max peak (3200 samples): 0x00095e0c ##                   7%
Max peak (3200 samples): 0x000ae081 ##                   8%
Max peak (3200 samples): 0x0006d961 ##                   5%
Max peak (3360 samples): 0x00070aec ##                   5%
Max peak (3200 samples): 0x0005cc6b #                    4%
Max peak (3200 samples): 0x00049476 #                    3%
Max peak (3200 samples): 0x000568d7 #                    4%
Max peak (3200 samples): 0x00056405 #                    4%
Max peak (3200 samples): 0x0006cce7 ##                   5%
Max peak (3200 samples): 0x0005376b #                    4%
Max peak (3200 samples): 0x00062fdf #                    4%
Max peak (3360 samples): 0x000b7d88 ##                   8%
Max peak (3200 samples): 0x00075629 ##                   5%
Max peak (2400 samples): 0x0006760c ##                   5%
Max peak (800 samples): 0x0007b312 ##                   6%
Max peak (3200 samples): 0x000a12cc ##                   7%
Max peak (3200 samples): 0x00064798 #                    4%
Max peak (800 samples): 0x00035a48 #                    2%
============stop here===============
```

**dmesg** 
[dmesg.txt](https://github.com/thesofproject/sof/files/5923612/dmesg.txt)
**sof-logger**
[trace.txt](https://github.com/thesofproject/sof/files/5923615/trace.txt)

keqiaozhang, 2021-02-04:
Checked with today's daily recipe, both https://github.com/thesofproject/sof/issues/3814 and https://github.com/thesofproject/sof/issues/3800 can not be reproduced.   
Looks like https://github.com/thesofproject/sof/commit/70b447b534c774d0067f7938e224af03018743ad can fix them all.

keqiaozhang, 2021-02-04:
I did a test to cherry pick below 2 patches to releases/tgl/v11.0-hotfix1 , this issue cannot be reproduced on Volteer-SDW.
https://github.com/thesofproject/sof/commit/8a5bcd9df6934b21c7a52a6cc3d9f72399fad0be https://github.com/thesofproject/sof/commit/70b447b534c774d0067f7938e224af03018743ad

lgirdwood, 2021-02-05:
@keqiaozhang thanks !
@mwasko @slawblauciak @abonislawski fyi.

mwasko, 2021-02-09:
The issue on tgl-011 will be resolved by the patchhttps://github.com/bkokoszx/sof/commit/8430ed722590ba8e81f16a3e716ea56fd2aaf777 that is a part of the fix on #3732 
Since we do not want to generate unnecessary noise and keep tgl-011 as stable as possible only the required fixes will be integrated.

slawblauciak, 2021-02-26:
Addressed by timer margin added in https://github.com/thesofproject/sof/pull/3770 on master branch. Similar change is done on the tgl-011 branch.

closed 2021-02-26 by slawblauciak