thesofproject/sof #3815: [BUG][TGL] XRUN on TGL_RVP_NOCODEC in multiple pipeline test cases

aiChaoSONG, 2021-02-04:
**Describe the bug**
run multiple pipeline test cases on TGL_RVP_NOCODEC, we will get io error in the terminal, and after check sof-logger, xrun happened

**To Reproduce**
run ` ./multiple-pipeline.sh -f p` on TGL_RVP_NOCODEC device, or run every pipeline in different terminal manually.

**Reproduction Rate**
almost 100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {topic/sof-dev 157a2621}
    * SOF: {v1.7rc1 372d3873bd55 and latest master}
2) Name of the topology file
    * Topology: {https://github.com/thesofproject/sof/blob/master/tools/topology/development/sof-tgl-nocodec-ci.m4}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {TGL_RVP_NOCODEC }
   
report ID: 2026

[meeting.txt](https://github.com/thesofproject/sof/files/5924579/meeting.txt)
[slogger_trace.txt](https://github.com/thesofproject/sof/files/5924589/slogger_trace.txt)
```
         TIMESTAMP              DELTA C# COMPONENT          LOCATION                      CONTENT
[1070077878.593750] (             nan) c0 dma-trace             src/trace/dma-trace.c:339  ERROR FW ABI 0x3012001 DBG ABI 0x5002001 tag v1.7-rc1 src hash 0xc22bf790 (ldc hash 0xc22bf790)
[1070144867.708333] (    66989.117188) c0 mn                 ./drivers/intel/ssp/mn.c:196  ERROR Can't set MCLK 0 to 38400000, it is already configured to 19200000
[1070144894.687500] (       26.979166) c0 ssp-dai      1.0   /drivers/intel/ssp/ssp.c:642  ERROR invalid mclk_rate = 38400000 for mclk_id = 0
[1070145706.197917] (      811.510437) c0 mn                 ./drivers/intel/ssp/mn.c:196  ERROR Can't set MCLK 0 to 38400000, it is already configured to 19200000
[1070146110.468750] (      404.270844) c0 dw-dma                 src/drivers/dw/dma.c:1101 ERROR dw_dma_get_data_size(): xrun detected
[1070146130.416667] (       19.947916) c0 dai          1.6            src/audio/dai.c:785  ERROR dai_report_xrun(): underrun due to no data available
[1070146150.208333] (       19.791666) c0 dai          1.6            src/audio/dai.c:675  ERROR comp_underrun(): dev->comp.id = 6, source->avail = 192, copy_bytes = 0
[1070146288.072917] (      137.864578) c0 pipe         1.7       src/audio/pipeline.c:1030 ERROR pipeline_copy(): ret = -61, start->comp.id = 6, dir = 1
[1070146312.031250] (       23.958334) c0 pipe         1.7       src/audio/pipeline.c:1220 ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[1070146389.166667] (       77.135414) c0 ssp-dai      1.1   /drivers/intel/ssp/ssp.c:642  ERROR invalid mclk_rate = 38400000 for mclk_id = 0
[1070148287.604167] (     1898.437500) c0 dw-dma                 src/drivers/dw/dma.c:1101 ERROR dw_dma_get_data_size(): xrun detected
[1070148307.343750] (       19.739584) c0 dai          3.16           src/audio/dai.c:785  ERROR dai_report_xrun(): underrun due to no data available
[1070148327.291667] (       19.947916) c0 dai          3.16           src/audio/dai.c:675  ERROR comp_underrun(): dev->comp.id = 16, source->avail = 192, copy_bytes = 0
[1070148442.604167] (      115.312500) c0 pipe         3.17      src/audio/pipeline.c:1030 ERROR pipeline_copy(): ret = -61, start->comp.id = 16, dir = 1
[1070148466.510417] (       23.906250) c0 pipe         3.17      src/audio/pipeline.c:1220 ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[1073580277.395833] (  3431811.000000) c0 host         1.0      src/audio/component.c:209  ERROR comp_set_state(): wrong state = 1, COMP_TRIGGER_STOP
[1073580298.854167] (       21.458334) c0 pipe         1.7       src/audio/pipeline.c:885  ERROR pipeline_trigger(): ret = -22, host->comp.id = 0, cmd = 0
[1073580322.343750] (       23.489584) c0 ipc                       src/ipc/handler.c:443  ERROR ipc: comp 0 trigger 0x50000 failed -22
[1073595526.250000] (    15203.906250) c0 host         3.12     src/audio/component.c:209  ERROR comp_set_state(): wrong state = 1, COMP_TRIGGER_STOP
[1073595547.916667] (       21.666666) c0 pipe         3.17      src/audio/pipeline.c:885  ERROR pipeline_trigger(): ret = -22, host->comp.id = 12, cmd = 0
[1073595571.041667] (       23.125000) c0 ipc                       src/ipc/handler.c:443  ERROR ipc: comp 12 trigger 0x50000 failed -22
```

aiChaoSONG, 2021-02-05:
d5840a920073e6712af2022ab3ec89816420e6ae / b90315f0f0dfaadf672adedc5901156a4a9297cc
Above two commits are related to SSP clk change, and caused this issue.
@keyonjie Please take a look.

keyonjie, 2021-02-05:
> [d5840a9](https://github.com/thesofproject/sof/commit/d5840a920073e6712af2022ab3ec89816420e6ae) / [b90315f](https://github.com/thesofproject/sof/commit/b90315f0f0dfaadf672adedc5901156a4a9297cc)
> Above two commits are related to SSP clk change, and caused this issue.
> @keyonjie Please take a look.

Oh, that requires the corresponding kernel PR: https://github.com/thesofproject/linux/pull/2669

keyonjie, 2021-02-05:
Oh, kernel PR is for rt5682 platforms only.

@aiChaoSONG can you help to try if https://github.com/thesofproject/sof/pull/3820 will fix this issue?

mengdonglin, 2021-02-05:
@aiChaoSONG confirmed #3820 can fix this issue. Closing.

closed 2021-02-05 by mengdonglin