thesofproject/sof #3816: [BUG][APL][v1.7rc1] XRUN on APL_UP2_NOCODEC with multiple-pause-resume test case

aiChaoSONG, 2021-02-04:
**Describe the bug**
in SOF v1.7c1 sanity test, xrun occurred on APL_UP2_NOCODEC with multiple-pause-resume test case.

**To Reproduce**
run `multiple-pause-resume.sh -r 25`

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {topic/sof-dev 157a2621}
    * SOF: {v1.7rc1 372d3873bd55}
2) Name of the topology file
    * Topology: {sof-apl-nocodec.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {APL_NOCODEC}

Report ID: 2026

[term.txt](https://github.com/thesofproject/sof/files/5924760/term.txt)

[dmesg.txt](https://github.com/thesofproject/sof/files/5924763/dmesg.txt)

[slogger_trace.txt](https://github.com/thesofproject/sof/files/5924766/slogger_trace.txt)

[slogger_error.txt](https://github.com/thesofproject/sof/files/5924771/slogger_error.txt)

aiChaoSONG, 2021-02-04:
#2749 is also a IO error issue, which happens on multiple-pipeline-capture.

lgirdwood, 2021-05-10:
@aiChaoSONG still an issue ?

aiChaoSONG, 2021-05-10:
After checking the latest several daily report, this issue is not reproduced any more, let's close this.

closed 2021-05-10 by aiChaoSONG