thesofproject/sof #3817: [FEATURE][IMPROVMENT] Add a new scheduling type for topology

iuliana-prodan, 2021-02-04:
**Is your feature request related to a problem? Please describe.**
Started from the pull request #3803, see discussion https://github.com/thesofproject/sof/pull/3803#discussion_r569524093

**Describe the solution you'd like**
The synchronous DMA_DOMAIN should be time independent (or the full_sync mode added in pull request #3803)
So, update the synchronous mode from DMA_DOMAIN and remove the time dependency for scheduling new tasks.

**Describe alternatives you've considered**
1. Add a new scheduling type for topology. 
2. Add a new DMA_DOMAIN mode (full_sync) which is not time dependent and use it as argument when initialize dma_multi_chan_domain.

**Additional context**
Why adding a new scheduling type for topology?
See issue #3802 and https://github.com/thesofproject/sof/pull/3803#discussion_r568618467

iuliana-prodan, 2021-02-04:
@lgirdwood If we choose the alternative solution 1 (add a new scheduling type for topology) I see:
- some redundant code - when creating a pipeline we use the pipeline_is_timer_driven() and based on this we choose SCHEDULE_LL_TIMER or SCHEDULE_LL_DMA. So, we should remove this, actually both LL_TIMER and LL_DMA are timer driven;
- some duplicate code - dma_multi_chan_domain should remain the same for non-timer DMA_DOMAIN, just is dma_multi_chan_domain_is_pending we should not use the task->start for scheduling;
- we'll need to add a new SOF_DOMAIN_DMA in sof_ipc_pipe_sched_time_domain (topology.h) and a new type of scheduler (scheduler.h). In both cases this will be a just a subtype of SOF_TIMER_DOMAIN_DMA, respectively of SOF_SCHEDULE_LL_DMA.

So, my proposal is to change synchronous mode from DMA_MULTI_CHAN_DOMAIN to no timer depended.
My only concern is the Intel platforms: how they use the synchronous mode for this and if they really use the task->start. I'll need to look closer on that.

lgirdwood, 2021-02-05:
@iuliana-prodan it may be clearer for everyone if you were to describe the rules for the current 2 types today and then the rules for the 3rd type. i.e for LL_TIMER (from the top of my head, so I may be missing important details).
1) LL_TIMER task are ordered and scheduled to run in same sequence on synchronous timer tick. timer is rearmed at end of work so that work is run every 1ms. Timer needs to be on same clock domain as audio interfaces. If the total length of work >1ms then it fails and recovers at next earliest tick.