thesofproject/sof #3818: [BUG IMX]

giuzan, 2021-02-04:
**Premise**

I think that the _loopback_ feature could be very useful in SOF. With _loopback_ I basically mean a _capture_ PCM from which we can read the samples that are _written_ in _playback_. You can find this functionality in ALSA, known as _ALSA Loopback_. I think that performing the loopback job inside the DSP would be very useful. Developers who want to integrate its own echo canceller block can really take advantage of this feature.
 
I think that I have found a good example of that in _sof-apl-demux-pcm512x.m4_ topology.
 
I tried to port that topology into _imx8 cs42888_. You can find the _m4_ file here https://github.com/giuzan/sof/blob/c5f4111078e052597e233d3770d773f02f18f1e2/tools/topology/sof-imx8-cs42888-demux.m4. 

The topology I build is shown in the following picture:

[image]

Unfortunately, I experienced an issue (see description below). It is the reason why I opened this [BUG]. If you think that the _loopback topic_ is beyond this issue, please let me know if it should be treated as a proposed [FEATURE].

**Describe the bug**

If you run the command
`arecord -D hw:1,1 -c 2 -r 48000 -f S16_LE record.wav`
the following error is printed
`arecord: pcm_read:2149: read error: Input/output error`

Moreover, if you arecord during an aplay, the playback stream stops until the arecord does not exit after the error.

**To Reproduce**

`arecord -D hw:1,1 -c 2 -r 48000 -f S16_LE record.wav`

or 

`aplay -D hw:1,1 file.wav & arecord -D hw:1,1 -c 2 -r 48000 -f S16_LE record.wav`

**Reproduction Rate**

100%

**Expected behavior**

The file record.wav recorded through the _capture pipeline_ should contain the samples processed by the _playback pipeline_. 

**Impact**

I cannot implement loopback for echo cancellation.

**Environment**

1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://source.codeaurora.org/external/imx/linux-imx/tree/?h=imx_5.4.47_2.2.0
    * SOF: https://github.com/giuzan/sof/commit/c5f4111078e052597e233d3770d773f02f18f1e2 (it basically is the imx-stable-v1.5 branch with topology above applied)
2) Name of the topology file
    * Topology: sof-imx8-cs42888-demux.m4
3) Name of the platform(s) on which the bug is observed.
    * Platform: i.MX8QM

lgirdwood, 2021-02-05:
@giuzan buffers cant do any processing here. i.e. your buf2.0 has 2 inputs. Shouldn't this be ```demix1.0 -> buf -> echoref2``` instead ?

giuzan, 2021-02-06:
@lgirdwood

Yes, I agree, it sounds weird to me too that a buffer has two input streams. I would expect that the only components which are allowed to have more than one stream (in input or in output) should be only: _mixer_ (that mixes N streams into one); _demux/mux_ (the first creates one stream from many, the second creates more streams from one). Am I wrong? Or in general it is allowed also for other non-buffer components?

Anyway, I took inspiration from _sof-apl-demux-pcm512x_ that has the same "double input". See picture below.

[image]

Do you think the issue I experienced is related to this kind of configuration in general or it could be limited to imx platform only?

giuzan, 2021-02-06:
@lgirdwood

About your suggestion `demix1.0 -> buf -> echoref2`

Do you mean something like that?

[image]

lgirdwood, 2021-02-09:
@giuzan so a buffer only has 1 sink and 1 source but a processing component can have N sinks and N sources (it's upto the component designer). Your pipeline above should have a buffer between ESAI0.IN and ECHO_REF2 but otherwise looks good. Your first pipeline is good if you also connect MUX to ECHO REF (i.e. echo ref has 2 inputs).  Note you will need to tell topology that it's OK for some buffer to overrun/underrun (since the user can start PCM0P and PCM0C at different times and in different orders).

giuzan, 2021-03-04:
Hi @lgirdwood and @dbaluta,

I tried the topology in the picture below. You can find the M4 here https://github.com/giuzan/sof/blob/giuzan_imx-stable-v1.6_loop/tools/topology/sof-imx8-cs42888-demux-loop.m4.

I implemented a custom component (_test loop_) that takes as sources the demux output (BUF2.2) and the DAI input (BUF2.1). My wish is that this component would couple the two 2-channels input sources in an output sink containing 2+2 channels (2 from BUF2.2 and 2 from BUF2.1). As first step I tried with a simpler scenario where the sink has only 2 channels, for the moment the DAI input samples are discarded.

Unfortunately I experienced the same input/output error above in capture scenario. And _dmesg_ shows the following errors:

```
[  101.618087] sof-audio-of 556e8000.dsp: error: ipc timed out for 0x60050000 size 12
[  101.625698] sof-audio-of 556e8000.dsp: error: waking up any trace sleepers
[  101.632639] sof-audio-of 556e8000.dsp: error: trace IO error
[  101.634108]  Capture: ASoC: trigger FE cmd: 0 failed: -110
[  102.162080] sof-audio-of 556e8000.dsp: error: ipc timed out for 0x60030000 size 12
[  102.169689] sof-audio-of 556e8000.dsp: error: waking up any trace sleepers
[  104.722083] sof-audio-of 556e8000.dsp: error: ipc timed out for 0x40010000 size 7
```

About buffer overrun/underrun, I did not be able to allow it through topology configuration. Hence I hardcoded in _buffer.c_ (just to perform the test):
```
buffer->stream.underrun_permitted = true;
buffer->stream.overrun_permitted = true;
```

[image]

----

I tried also another scenario (M4 in https://github.com/giuzan/sof/blob/giuzan_imx-stable-v1.6_loop/tools/topology/sof-testloop-imx8qm-cs42888.m4) where I used the _test loop_ component also in playback, to copy the source in two sinks. I experienced the same issue, in playback also. Could be that is there something wrong with my test component? The code is here https://github.com/giuzan/sof/blob/giuzan_imx-stable-v1.6_loop/src/audio/test_loop/test_loop.c.

[image]

In particular, I have the issue when I _connect_ the two pipelines, in other words if I remove the following lines from the the topology M4 the streaming (both playback and capture) works:

```
SectionGraph."PIPE_LOOP" {
	index "0"

	lines [
		dapm(PIPELINE_LOOPBACK_IN_BUFFER_2, PIPELINE_LOOPBACK_OUT_1)
		dapm(PIPELINE_LOOPBACK_IN_2, PIPELINE_LOOPBACK_IN_BUFFER_2)
	]
}
```

iuliana-prodan, 2021-03-04:
Hi @giuzan,

I intend to look on the loopback, also. So, please, let me give it a try and I'll come back to you.

First thing to notice on your topology is that you should have two `registrable` tasks since you have ESAI input and ESAI output.
The SCHEDULE_TIME_DOMAIN_DMA was written with the assumption that there is only one `registrable` task.

Also, after how many iterations do you get the ipc timeout?
Maybe, try to apply [this](https://github.com/thesofproject/sof/pull/3867/commits/bfaff516a04142ffad53bb12048edab03e5a71ec) commit. 

Thanks,
Iulia

giuzan, 2021-03-05:
Hi @iuliana-prodan,

Thanks for your feedback.

I get the ipc at the first iteration.

Ok, let me know after your try. In the middle, I'll try to apply the commit you linked and repeat the test.

iuliana-prodan, 2021-03-05:
> I get the ipc at the first iteration.

Great, this will be much easier to reproduce :)

lgirdwood, 2021-03-05:
@giuzan the IPC timout looks like a hang somewhere ?

giuzan, 2021-03-05:
@lgirdwood On DSP side is difficult to evaluate that, because the sof-logger crashes after the error. I cannot see the log prints before, since I guess that they are not yet flushed.

dbaluta, 2021-03-05:
@giuzan are you using 5.10 kernel. We have better infrastructure for debugging. It looks like firmware is crashing somewhere.

giuzan, 2021-03-05:
@dbaluta Yes, I guess that the timeout could be due to the firmware crash, that makes impossibile handle the IPC request.
About kernel, i'm using  imx_5.4.47-2.2.0 (https://source.codeaurora.org/external/imx/linux-imx/log/?h=imx_5.4.47_2.2.0). Do you think should I move to 5.10? The latest linux-imx BSP release isn't it imx_5.4.70_2.3.0?

giuzan, 2021-03-05:
@iuliana-prodan I tried the [fix](https://github.com/thesofproject/sof/pull/3867/commits/bfaff516a04142ffad53bb12048edab03e5a71ec) but unfortunately I have the same issue.

lgirdwood, 2021-03-08:
@giuzan any idea why logger is crashing ? It may be best to fix this first as there may be some meaningful trace item that is key to solving this (e.g. the trace item may be malformed as a result of the hang or crash).
FW may be looping due to a topology loop, or may be spinning on a lock somewhere. It seems to be topology related sine you are the only person reporting this.

giuzan, 2021-03-08:
@lgirdwood Ok thanks. At the moment I can't understand why it crashes. I will investigate on it. Anyway, do you confirm that the loopback I'm trying to arrange is, in theory, possibile?

lgirdwood, 2021-03-08:
> @lgirdwood Ok thanks. At the moment I can't understand why it crashes. I will investigate on it. Anyway, do you confirm that the loopback I'm trying to arrange is, in theory, possibile?

Yes, it may be a misconfiguration though as demux is used on production devices by Intel. It may be even worth running the pipeline copy() once and then checking the debug/trace for any obvious issues.

iuliana-prodan, 2021-03-08:
@giuzan I've checked both topologies and I believe they have some issues.
I still don't understand very well the topologies, especially the demux one, so I couldn't find the exact problem.

For both topologies, I see that the soundcards available are:
_sof-imx8-cs42888-demux-loop.tplg_
```
root@imx8qxpc0mek:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: btscoaudio [bt-sco-audio], device 0: 59040000.sai-bt-sco-pcm-wb bt-sco-pcm-wb-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofimxcs42888 [sof-imx-cs42888], device 0: Playback (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@imx8qxpc0mek:~#
```
_sof-testloop-imx8qm-cs42888.tplg_
```
root@imx8qxpc0mek:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: btscoaudio [bt-sco-audio], device 0: 59040000.sai-bt-sco-pcm-wb bt-sco-pcm-wb-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofimxcs42888 [sof-imx-cs42888], device 0: pcm_out (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@imx8qxpc0mek:~# 
```
If you want to capture on -D hw:1,1 I believe you should have another card 1 (another sub-device, like the mixer topology).
It should look something like this:
```
root@imx8qxpc0mek:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: btscoaudio [bt-sco-audio], device 0: 59040000.sai-bt-sco-pcm-wb bt-sco-pcm-wb-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofimxcs42888 [sof-imx-cs42888], device 0: Playback (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofimxcs42888 [sof-imx-cs42888], device 0: Capture (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@imx8qxpc0mek:~#
```

Using _sof-imx8-cs42888-demux-loop.tplg_ I was able to playback on hw1,0 but the capture on hw:1,1 gives errors, which is expected, there is no 1,1 sub-device. 
Here's how I ran the commands and the trace. Let me know if I've missed some steps:
```
root@imx8qxpc0mek:~# aplay -D hw:1,0 16/equinox_16b_48KHz.wav 
Playing WAVE '16/equinox_16b_48KHz.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
root@imx8qxpc0mek:~# 

root@imx8qxpc0mek:~# ./sof-logger -t -l /lib/firmware/imx/sof/sof-imx8x.ldc 2>&1 | tee trace.txt
         TIMESTAMP              DELTA C# COMPONENT          LOCATION                      CONTENT
[16266961818.072918] (             nan) c0 dma-trace             src/trace/dma-trace.c:339  ERROR FW ABI 0x3012001 DBG ABI 0x5003000 tag v1.7-rc1-116-g8057a6da753b src hash 0xef67e0c7 (ldc hash 0xef67e0c7)
[16266962194.062500] (      375.989594) c0 ll-schedule        ./schedule/ll_schedule.c:399  task add 0x92c05080 dma-trace-task <2b972272-c5b1-4b7e-926f-0fc5cb4c4690>
[16266962304.843750] (      110.781250) c0 ll-schedule        ./schedule/ll_schedule.c:403  task params pri 4 flags 0 start 500000 period 500000
[16266962507.552084] (      202.708328) c0 ll-schedule        /schedule/timer_domain.c:179  timer_domain_register domain->type 1 domain->clk 0 domain->ticks_per_ms 666000 period 500000
[16266962843.489584] (      335.937500) c0 ll-schedule        ./schedule/ll_schedule.c:309  new added task->start 333856856
[16266962958.125000] (      114.635414) c0 ll-schedule        ./schedule/ll_schedule.c:312  num_tasks 1 total_num_tasks 1
[16266964698.697918] (     1740.572876) c0 ipc                       src/ipc/handler.c:626  ipc: pm core mask 0x1 -> enable
[16266966149.531250] (     1450.833374) c0 component             src/audio/component.c:130  comp new host <8b9d100c-6d78-418f-90a3-e0e805d0852b> type 1 id 1.0
[16266966605.260418] (      455.729156) c0 dma                           src/lib/dma.c:127  dma_get() ID 1 sref = 3 busy channels 1
[16266968056.666668] (     1451.406250) c0 component             src/audio/component.c:130  comp new pga <b77e677e-5ff4-4188-af14-fba8bdbf8682> type 5 id 1.1
[16266968505.260418] (      448.593750) c0 pga          1.1   ../audio/volume/volume.c:379  vol->initial_ramp = 20, vol->ramp = 2, vol->min_value = 0, vol->max_value = 65536
[16266969763.177084] (     1257.916626) c0 component             src/audio/component.c:130  comp new demux <c4b26868-1430-470e-a089-15d1c77f851a> type 18 id 1.2
[16266969886.875000] (      123.697914) c0 mux                     src/audio/mux/mux.c:168  mux_new()
[16266970464.531250] (      577.656250) c0 demux        1.2        src/audio/mux/mux.c:100  mux_set_values()
[16266971852.135418] (     1387.604126) c0 buffer                   src/audio/buffer.c:79   buffer new size 0x300 id 1.3 flags 0x0
[16266973351.041668] (     1498.906250) c0 buffer                   src/audio/buffer.c:79   buffer new size 0x300 id 1.4 flags 0x0
[... 183 lines cut ...]
[21606871114.895836] (     1273.177124) c0 esai         6.0     src/drivers/imx/esai.c:453  esai_get_hw_params(): params->rate = 48000, fsync_rate = 48000
[21606871688.958336] (      574.062500) c0 dai          1.5            src/audio/dai.c:351  dai_playback_params() dest_dev = 4026 stream_id = 0 src_width = 4 dest_width = 4
[21606871800.260418] (      111.302086) c0 dai          1.5            src/audio/dai.c:357  dai_playback_params() fifo 0x59010000
[21606872062.135418] (      261.875000) c0 pipe         1.6       src/audio/pipeline.c:683  pipe prepare
[21606872637.916668] (      575.781250) c0 tdfb         1.1      src/audio/tdfb/tdfb.c:456  tdfb_prepare()
[21606872758.645836] (      120.729164) c0 tdfb         1.1      src/audio/component.c:438  WARN comp_get_data_blob(): blob_handler->data is not set.
[... cut ...]
[21607867079.739586] (      250.833328) c0 ipc                       src/ipc/handler.c:443  ERROR ipc: comp 0 trigger 0x50000 failed -22

```

So, I'm not getting any `ipc timed out` errors. I've used your patches on the master branch.

For the sof-logger, please be sure you've compiled it on sof/imx-stable-v1.6. 
If you have a sof-logger from sof/imx-stable-v1.5, you might experience issues. 
I've once tried a sof-logger compiled on sof/imx-stable-v1.6 on an sof/imx-stable-v1.5 branch and it didn't worked at all :(

iuliana-prodan, 2021-03-08:
One more thing for the sof-logger.
I'm also debugging an `ipc timed out` issue. I get something like this:
```
...
Playing raw data '16/j16.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Recording WAVE 'test_record.pcm' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
[  582.814139] sof-audio-of 556e8000.dsp: error: ipc timed out for 0x60050000 size 12
[  582.821802] sof-audio-of 556e8000.dsp: info: preventing DSP entering D3 state to preserve context
[  582.830906] sof-audio-of 556e8000.dsp: invalid header size 0x2000200. FW oops is bogus
[  582.838989] sof-audio-of 556e8000.dsp: error: unexpected fault 0x00000000 trace 0x00000000
[  582.847530] sof-audio-of 556e8000.dsp: error: waking up any trace sleepers
[  582.854601] sof-audio-of 556e8000.dsp: error: trace IO error
[  582.861198] sof-audio-of 556e8000.dsp: ASoC: error at snd_soc_pcm_component_trigger on 556e8000.dsp: -110
[  582.871466]  Port0: ASoC: trigger FE cmd: 0 failed: -110
Playing raw data '16/j16.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Recording WAVE 'test_record.pcm' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: pcm_write:2061: write error: Input/output error
(null)      1  TFAIL  :  ltpapicmd.c:188: Test 568 Playback failed

```
While running my test, I have the sof-logger started:
```
root@imx8qxpc0mek:~# ./sof-logger -t -l /lib/firmware/imx/sof/sof-imx8x.ldc 2>&1 | tee trace.txt
...
[19469098836.770836] (       96.666664) c0 ipc                       src/ipc/handler.c:457  handler: ipc_glb_stream_message: SOF_IPC_STREAM_PCM_PARAMS = 0x10000
[19469099490.572918] (      653.802063) c0 pipe         1.5       src/audio/pipeline.c:543  pipe params dir 0 frame_fmt 0 buffer_fmt 0 rate 48000
[19469099601.041668] (      110.468750) c0 pipe         1.5       src/audio/pipeline.c:547  pipe params stream_tag 0 channels 2 sample_valid_bytes 2 sample_container_bytes 2
[19469100828.697918] (     1227.656250) c0 dai          1.4            src/audio/dai.c:351  dai_playback_params() dest_dev = 8029 stream_id = 0 src_width = 4 dest_width = 4
[19469100967.604168] (      138.906250) c0 dai          1.4            src/audio/dai.c:357  dai_playback_params() fifo 0x59050020
[19469101452.395836] (      484.791656) c0 pipe         1.5       src/audio/pipeline.c:670  pipe prepare
[19469102562.604168] (     1110.208374) c0 edma                 src/drivers/imx/edma.c:362  EDMA: set config
[19469173324.739586] (    70762.132812) c0 ipc                       src/ipc/handler.c:1400 handler: ipc_cmd: command type 1610612736 or hex = 0x60000000
[19469173453.072918] (      128.333328) c0 ipc                       src/ipc/handler.c:1421 handler: ipc_cmd: command SOF_IPC_GLB_STREAM_MSG = 0x60000000
**error: in logger_read(), fread(..., /sys/kernel/debug/sof/trace) failed: Operation not permitted(-1)**

```
When the `ipc timed out` error occurs I see the above failure in the sof-logger.
But, next, if I check again the trace, I have the rest of it:
```
root@imx8qxpc0mek:~# ./sof-logger -t -l /lib/firmware/imx/sof/sof-imx8x.ldc 2>&1 | tee trace.txt
...
[19469173453.072918] (      128.333328) c0 ipc                       src/ipc/handler.c:1421 handler: ipc_cmd: command SOF_IPC_GLB_STREAM_MSG = 0x60000000
[19469173563.437500] (      110.364586) c0 ipc                       src/ipc/handler.c:468  handler: ipc_glb_stream_message: SOF_IPC_STREAM_TRIG_START/STOP...
[19469173683.750000] (      120.312500) c0 ipc                       src/ipc/handler.c:416  ipc: ipc_stream_trigger: comp 0 -> trigger cmd 0x40000
[19469173794.166668] (      110.416664) c0 pipe         1.5       src/audio/pipeline.c:933  pipe trigger cmd 1, time is = 338afa601f
[19469174030.052086] (      235.885422) c0 edma                 src/drivers/imx/edma.c:137  EDMA: start(15)
[19469174168.020836] (      137.968750) c0 sai          5.1      src/drivers/imx/sai.c:376  SAI: sai_trigger
[19469174260.208336] (       92.187500) c0 sai          5.1      src/drivers/imx/sai.c:32   SAI: sai_start
[19469174378.541668] (      118.333336) c0 sai          5.1      src/drivers/imx/sai.c:41   SAI: sai_start 1: dir = 0, TCSR - 0x90100000
...
```
You can check the timestamp.

giuzan, 2021-03-09:
Hi @iuliana-prodan, thanks!

About:
> If you want to capture on -D hw:1,1 I believe you should have another card 1 (another sub-device, like the mixer topology).
> ...
> Using sof-imx8-cs42888-demux-loop.tplg I was able to playback on hw1,0 but the capture on hw:1,1 gives errors, which is expected, there is no 1,1 sub-device.

The hw:1,1 device is present but, being a capture device, is made visible by `arecord -l` command:
```
root@imx8qmmek:~# arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: btscoaudio [bt-sco-audio], device 0: 59040000.sai-bt-sco-pcm-wb bt-sco-pcm-wb-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofimxcs42888 [sof-imx-cs42888], device 1: Capture (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

About your try, ok you have the same result of mine (with the first topology you playback and never capture, with the second you never playback and capture). Anyway, you didn't experience an ipc timeout error, but a generic ipc error. The difference could be related to the fact that you have applied the patches to the master branch?

About sof-logger, I confirm that I built it on imx-stable-v1.6.

lgirdwood, 2021-03-09:
@giuzan @iuliana-prodan looks like the pipeline state is inconsistent i.e. a component has an unexpected state at TRIGGER_START.
```
[21606872758.645836] (      120.729164) c0 tdfb         1.1      src/audio/component.c:438  WARN comp_get_data_blob(): blob_handler->data is not set.
[21606873354.114586] (      595.468750) c0 edma                 src/drivers/imx/edma.c:352  EDMA: set config
[21607134806.406250] (   261452.296875) c0 pipe         1.6       src/audio/pipeline.c:878  pipe trigger cmd 1
[21607135012.031250] (      205.625000) c0 tdfb         1.1      src/audio/tdfb/tdfb.c:514  tdfb_trigger(), command = 1
[21607135129.375000] (      117.343750) c0 tdfb         1.1      src/audio/component.c:190  ERROR comp_set_state(): wrong state = 1, COMP_TRIGGER_START
[21607135339.843750] (      210.468750) c0 pipe         1.6       src/audio/pipeline.c:899  ERROR pipeline_trigger(): ret = -22, host->comp.id = 0, cmd = 1
[21607135591.927086] (      252.083328) c0 ipc                       src/ipc/handler.c:443  ERROR ipc: comp 0 trigger 0x40000 failed -22
[21607866347.447918] (   730755.500000) c0 pipe         1.6       src/audio/pipeline.c:878  pipe trigger cmd 0
[21607866502.864586] (      155.416672) c0 tdfb         1.1      src/audio/tdfb/tdfb.c:514  tdfb_trigger(), command = 0
[21607866618.333336] (      115.468750) c0 tdfb         1.1      src/audio/component.c:209  ERROR comp_set_state(): wrong state = 1, COMP_TRIGGER_STOP
[21607866828.906250] (      210.572922) c0 pipe         1.6       src/audio/pipeline.c:899  ERROR pipeline_trigger(): ret = -22, host->comp.id = 0, cmd = 0
[21607867079.739586] (      250.833328) c0 ipc                       src/ipc/handler.c:443  ERROR ipc: comp 0 trigger 0x50000 failed -22
```
Stream IPC messages are notifications, this means they can be async to the pipeline copy(), could they be trashing the mailbox data ?? 
```
[19469173324.739586] (    70762.132812) c0 ipc                       src/ipc/handler.c:1400 handler: ipc_cmd: command type 1610612736 or hex = 0x60000000
[19469173453.072918] (      128.333328) c0 ipc                       src/ipc/handler.c:1421 handler: ipc_cmd: command SOF_IPC_GLB_STREAM_MSG = 0x60000000
```
Also it may be worth dumping the mailbox contents and taking a look for any obvious problems.

lgirdwood, 2021-05-10:
@dbaluta any update here ? Should we close or move to v1.9 ?

iuliana-prodan, 2021-05-10:
> @dbaluta any update here ? Should we close or move to v1.9 ?

@lgirdwood @dbaluta this won't be available for v1.8, we should move it to v1.9.

dbaluta, 2021-06-24:
This is for v1.9

giuzan, 2021-07-02:
@lgirdwood sorry for my late answer, but i was moved for a while to another project, i had not chances to work on it.

kv2019i, 2023-02-15:
@dbaluta I assigned this to "TBD" milestone, feel free to assing based on your plans.