thesofproject/sof #3827: Period size limitation (KWD_PIPE_SCH_DEADLINE_US)

cristinailie, 2021-02-08:
Hello,

I am trying to increase KWD_PIPE_SCH_DEADLINE_US value on keyword detection topology (currently set to only 20ms) and I see that for more than ~120 ms  the arecord doesn't work anymore (IPC timeout error). Is this a SOF/platform limitation and if yes why? Or this behavior is not expected and may be a bug?

I would like to increase it since I would like to run a longer task (more than 100ms) inside a component and I think the task time cannot exceed this KWD_PIPE_SCH_DEADLINE_US value, otherwise I expect to lose real-time data.

Thank you,
Cristina

dbaluta, 2021-02-08:
@cristinailie it looks SOF firmware runs out of memory.

Can you please try:

```
diff --git a/src/platform/imx8m/include/platform/lib/memory.h b/src/platform/imx8m/include/platform/lib/memory.h
index 334165fab..1a1f13f43 100644
--- a/src/platform/imx8m/include/platform/lib/memory.h
+++ b/src/platform/imx8m/include/platform/lib/memory.h
@@ -107,20 +107,20 @@
 
 /* Heap section sizes for module pool */
 #define HEAP_RT_COUNT8         0
-#define HEAP_RT_COUNT16                48
-#define HEAP_RT_COUNT32                48
-#define HEAP_RT_COUNT64                32
-#define HEAP_RT_COUNT128       32
-#define HEAP_RT_COUNT256       32
-#define HEAP_RT_COUNT512       4
-#define HEAP_RT_COUNT1024      4
-#define HEAP_RT_COUNT2048      4
-#define HEAP_RT_COUNT4096      4
+#define HEAP_RT_COUNT16                48 * 4
+#define HEAP_RT_COUNT32                48 * 4
+#define HEAP_RT_COUNT64                32 * 4
+#define HEAP_RT_COUNT128       32 * 4
+#define HEAP_RT_COUNT256       32 * 4
+#define HEAP_RT_COUNT512       4 * 4
+#define HEAP_RT_COUNT1024      4 * 4
+#define HEAP_RT_COUNT2048      4 * 4
+#define HEAP_RT_COUNT4096      4 * 4
 
 /* Heap section sizes for system runtime heap */
-#define HEAP_SYS_RT_COUNT64    128
-#define HEAP_SYS_RT_COUNT512   16
-#define HEAP_SYS_RT_COUNT1024  8
+#define HEAP_SYS_RT_COUNT64    128 * 4
+#define HEAP_SYS_RT_COUNT512   16 * 4
+#define HEAP_SYS_RT_COUNT1024  8 * 4
 
 /* Heap configuration */
 
```

dbaluta, 2021-02-08:
Closing this. As discussed the memory allocation problem arrives from a multiplication that overflows uint32_t. Long term solution is to switch some variables to uint64_t. 

Who would have imagined that uint32_t isn't enough for everyone.

closed 2021-02-08 by dbaluta

plbossart, 2021-02-08:
> Who would have imagined that uint32_t isn't enough for everyone.

It's quite common in audio to wrap-around if you play for several days... Anything that counts bytes, times or deals with sample rate needs to be handled with care.