thesofproject/sof #3831: [BUG] SOF docker image has proxy settings hardcoded for Intel network

kv2019i, 2021-02-09:
**Describe the bug**
The current SOF docker images have a hardcoded proxy set to Intel servers:
https://thesofproject.github.io/latest/getting_started/build-guide/build-with-docker.html

While testing the instructions from public network, scripts/docker-run.sh builds will fail due to this. 

** Mitigation **
Modify scripts/docker-run.sh with

````  
+     --env http_proxy \
+     --env https_proxy \
     --env CMAKE_BUILD_TYPE \
```

lgirdwood, 2021-05-10:
@fredoh9 this is now fixed right ? if so please close.

fredoh9, 2021-05-11:
@lgirdwood this is not fixed. @marc-hb I agree, we need to keep proxy-free for the docker image and set proxy when needed. I will work on it.

closed 2021-05-12 by lgirdwood