thesofproject/sof #3836: [BUG] need to align mixer and mux behavior for pause/stop

plbossart, 2021-02-10:
**Describe the bug**

The mixer and mux seem to have different behaviors to deal with cases when one or all of the sources are paused

mixer .trigger
````c
	/* don't stop mixer on pause or if at least one source is active/paused */
	if (cmd == COMP_TRIGGER_PAUSE ||
	    (cmd == COMP_TRIGGER_STOP &&
	     (mixer_source_status_count(dev, COMP_STATE_ACTIVE) ||
	     mixer_source_status_count(dev, COMP_STATE_PAUSED)))) {
		dev->state = COMP_STATE_ACTIVE;
		ret = PPL_STATUS_PATH_STOP;
	}
````

mux .trigger
````c
	/* don't stop mux if at least one source is active */
	if (mux_source_status_count(dev, COMP_STATE_ACTIVE) &&
	    (cmd == COMP_TRIGGER_PAUSE || cmd == COMP_TRIGGER_STOP)) {
		dev->state = COMP_STATE_ACTIVE;
		ret = PPL_STATUS_PATH_STOP;
	/* don't stop mux if at least one source is paused */
	} else if (mux_source_status_count(dev, COMP_STATE_PAUSED) &&
		   cmd == COMP_TRIGGER_STOP) {
		dev->state = COMP_STATE_PAUSED;
		ret = PPL_STATUS_PATH_STOP;
	}
````

The logic doesn't seem to be the same at all?

Also the mixer always generates zeroes, I don't see the behavior in the mux.

juimonen, 2021-02-10:
@plbossart isn't the behavior still the same? still this is a merge clash in mixer/mux. I copied the code exactly from mixer -> mux in my recent mux fix (to actually enable more than 1 stream input), but the mixer change pr was not merged (I guess). So the codes used to look exactly the same. Anyway should be unified. And I'm pretty sure mux doesn't produce 0 in pause...

plbossart, 2021-02-10:
@juimonen it's clearly different, in the mixer case the PAUSE trigger is translated as a COMP_STATE_ACTIVE without looking at any other conditions. I have no idea if the operator precedence is correct or if parentheses are placed in the right locations.

juimonen, 2021-02-10:
@plbossart yes if I decipher the logic correctly (I hate these multiple line logic sentences), the mixer is different exactly because it produces 0's and mux doesn't. So in that sense the mux logic should work correctly currently -> if we change it to produce 0 also we'll need to change the logic as in mixer.

plbossart, 2021-02-10:
@juimonen all it takes is to take the APL nocodec topology and replace the mixer by a mux. then start multiple streams and pause/stop them independently in different terminals.

juimonen, 2021-02-11:
@plbossart yes I was not claiming anything just for fun here... I tested again the mux/demux in hda device.

- mux with current fw -> I can play 2 streams simultaneously, play/stop/pause however I feel -> nothing bails out -> works for me (?)
- demux with current fw -> recording pipe will only record if I play something to playback pipe (gives you this "read error, no data available", so playback pipe you can play/pause/stop all the time, if you pause the playback pipe while recording -> recording pipe bails out -> for me this is how it has always worked

So the demux functionality is maybe the issue hare (?): AFAIK we did this "no data available" for the kwd pipeline, so that you can sleep against the pcm and wait for the data. So I'm not sure what would happen to it if we would produce zeros to it all the time.

However, I have nothing against implementing the zero production to it. @kv2019i @keyonjie any comments on this?

plbossart, 2021-02-11:
For playback the triggers are propagated from sources to sinks. I think the main issue is when we have multiple sources, and a trigger coming on a single one of those sources.

I am not clear on how triggers are propagated in the capture case, but what would happen if you pause one sink of a demux and not the others?

lgirdwood, 2021-05-10:
In progress, but wont hit v1.8. Move to v1.9

kv2019i, 2023-02-15:
This seems to have got stuck, assigning to TBD milestone.

closed 2025-07-24 by github-actions[bot]