thesofproject/sof #3837: [BUG] Smart Amp needs to handle pause/stop like a mixer/mux

plbossart, 2021-02-10:
**Describe the bug**

The Smart Amp by construction takes as sources the playback stream and the capture feedback stream.
When pausing or stopping the smart amp, we need to make sure the state of the feedback stream is taken into account. If the feedback is active or paused, the component needs to remain active.

This should be generalized to ALL components that take multiple sources into account.

This is a potential reason for issues with the multiple-pause-resume tests that fail on a recurring basis on TGL SoundWire chromebook.

**To Reproduce**
run multiple-pause-resume tests.

**Reproduction Rate**
100%

**Expected behavior**
the smart amp deals with pause gracefully no matter whether the playback or feedback streams are paused/stopped.

**Impact**
This does not impact ChromeOS (Pause is not used) but wrecks our CI tests, and in general keeps SOF from correctly dealing with components with multiple sources.

lgirdwood, 2021-05-10:
In progress for v1.9 as part of the codec adapter and component alignment work

kv2019i, 2023-02-15:
@ranj063 @aiChaoSONG @RanderWang  Is this old topic now covered with latest SOF main?

aiChaoSONG, 2023-02-20:
@kv2019i The smart amp now is able to pass multiple-pause-resume and multiple-pipeline stress test

lgirdwood, 2023-02-21:
@aiChaoSONG thanks, lets give it a few days in daily test and we can close this issue if smart amp is good..

closed 2023-05-10 by lgirdwood