thesofproject/sof #3842: [FEATURE] Preloading data to detect codec parameters with codec adapter

dbaluta, 2021-02-11:
Hi @mrajwa,

I want to add this Feature request in order to put my thoughts in order and to explore some options I have.

With Cadence codecs like mp3 in the initalization phase the codec needs some chunks of data from the input file in order
to figure out the codec parameters (sample rate, bit rate, channels etc).

The flow is mostly similar with what we have in Codec Adapter with this exception:

```
-> lib info
    -> read lib version
    -> read lib name, etc
-> memory allocation
-> preparation
    -> read mp3 file data from Linux kernel
    -> SET_INPUT_BYTE
    ->  INIT_PROCESS
    -> INIT_DONE_QUERY
    -> if not yet initialized go back and read more data
-> processing part
```
Now my concern is how to send the first frames in order to parse the configuration and initialize the code.

1) First option is to do this in prepare by preloading some of the mp3 frames and feed them to mp3 library
   -> Disadvantage for this is that I don't actually have acess to mp3 frames in codec_adapter component and I need to somehow preload them

2) Second option is to delay the initialization until first copy is done
    -> here I have the date but I might mess up with the codec internal state.

I will keep updating this thread with more info. If you have some ideas and time feel free to share your opinion on this.

lgirdwood, 2021-02-12:
I'm thinking that 
1) the compressed ALSA PCM should be able to trigger() and start sending data
2) the pipeline will trigger. The codec adapter would send 0s downstream until data was ready from codec.
3) the codec would keep consuming data (and load it's config) and then send decoded data downstream (i.e. codec adapter can stop sending 0s).

lgirdwood, 2021-06-03:
@dbaluta @paulstelian97 @iuliana-prodan I'm assuming this is now closed ?

paulstelian97, 2021-06-03:
> @dbaluta @paulstelian97 @iuliana-prodan I'm assuming this is now closed ?

I'm not comfortable with closing this, not at least until @dbaluta comes back from vacation and explains how it is.

Currently we're trying to delay the preparation until first copy(), and while it works, it's not exactly reliable.

dbaluta, 2021-07-01:
we've moved the params detecting in copy() function. All good. Closing now.

closed 2021-07-01 by dbaluta