thesofproject/sof #3843: [BUG][Cppcheck]

cujomalainey, 2021-02-11:
In imx edma driver there are some variables that are set but never used, this is upsetting cppcheck but they are explicitly labeled in comments. Paul you appear to ahave added some of these, do you want to suppress these errors or remove these variables?

dbaluta, 2021-02-12:
@cujomalainey care to share with us the command line you use to run `cppcheck`. Just running `cppcheck` with no arguments results in no error.

dbaluta, 2021-02-12:
Later edit:

```
$ cppcheck --enable=all --inconclusive --library=posix drivers/imx/edma.c 
Checking drivers/imx/edma.c ...
drivers/imx/edma.c:368:13: style: Variable 'handshake' is assigned a value that is never used. [unreadVariable]
  handshake = config->dest_dev;
            ^
drivers/imx/edma.c:373:13: style: Variable 'handshake' is assigned a value that is never used. [unreadVariable]
  handshake = config->src_dev;
            ^
drivers/imx/edma.c:349:17: style: Variable 'irq' is not assigned a value. [unassignedVariable]
 int handshake, irq;
                ^
nofile:0:0: information: Cppcheck cannot find all the include files (use --check-config for details) [missingIncludeSystem]
``

cujomalainey, 2021-02-12:
Looks like you found them

cujomalainey, 2021-02-12:
also, ignore divide by sizeof(pointer), those are a bug in cppcheck that I have started a topic upstream about. https://sourceforge.net/p/cppcheck/discussion/general/thread/d7881b159c/

paulstelian97, 2021-02-15:
A refactor moved the way the DAI drivers fetch the handshake and I forgot to update edma. I feel like the compiler doesn't actually generate any code for these anyway.

Still, it would be cleaner to just remove them. Thanks for notifying.

closed 2021-02-18 by dbaluta