thesofproject/sof #3852: [BUG] DSP panic during audio playback

brentlu, 2021-02-18:
**Describe the bug**
Playback stops during audio playback. Check kernel log and find there is DSP panic message.

**To Reproduce**
Running audio play for long period of time.

**Reproduction Rate**
2/10 when running 12-hour playback.

**Expected behavior**
Should not happen the DSP panic problem.

**Impact**
Need to reboot the device for audio to work again.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: chrome-v4.19
    * SOF: 68b5ab82e6d24e78ff189c65831c171bc3c9d389 (cml-011-drop-stable)
2) Name of the topology file
    * Topology: sof-cml-rt5682.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Kaisa (Puff-based Chromebox)

**Screenshots or console output**
2020-12-11T07:56:45.302277+08:00 ERR kernel: [46032.465543] sof-audio-pci 0000:00:1f.3: error : DSP panic!
2020-12-11T07:56:45.302343+08:00 ERR kernel: [46032.465577] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
2020-12-11T07:56:45.302349+08:00 ERR kernel: [46032.466594] sof-audio-pci 0000:00:1f.3: error: can't enter idle
2020-12-11T07:56:45.302352+08:00 ERR kernel: [46032.466601] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
2020-12-11T07:56:45.302358+08:00 ERR kernel: [46032.466607] sof-audio-pci 0000:00:1f.3: error: panic at src/lib/agent.c:60
2020-12-11T07:56:45.302362+08:00 ERR kernel: [46032.466613] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
2020-12-11T07:56:45.302365+08:00 ERR kernel: [46032.466621] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060725 SAR     0x00000000
2020-12-11T07:56:45.302371+08:00 ERR kernel: [46032.466628] sof-audio-pci 0000:00:1f.3: EPC1     0x00000000 EPC2     0xbe00d56e EPC3     0x00000000 EPC4    0x00000000
2020-12-11T07:56:45.302374+08:00 ERR kernel: [46032.466634] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
2020-12-11T07:56:45.302380+08:00 ERR kernel: [46032.466640] sof-audio-pci 0000:00:1f.3: EPS2     0x00060d20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
2020-12-11T07:56:45.302383+08:00 ERR kernel: [46032.466647] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
2020-12-11T07:56:45.302387+08:00 ERR kernel: [46032.466653] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe05c190
2020-12-11T07:56:45.302390+08:00 ERR kernel: [46032.466662] sof-audio-pci 0000:00:1f.3: 0xbe05c190: be013ca0 be05c1c0 be062880 00000001
2020-12-11T07:56:45.302393+08:00 ERR kernel: [46032.466672] sof-audio-pci 0000:00:1f.3: 0xbe05c194: 00000000 00000000 0000003c be062880
2020-12-11T07:56:45.302396+08:00 ERR kernel: [46032.466680] sof-audio-pci 0000:00:1f.3: 0xbe05c198: 00000000 00000000 2f637273 2f62696c
2020-12-11T07:56:45.302398+08:00 ERR kernel: [46032.466688] sof-audio-pci 0000:00:1f.3: 0xbe05c19c: c052a432 ffffffff 19bd9028 ffff94f4
2020-12-11T07:56:45.302402+08:00 ERR kernel: [46032.466696] sof-audio-pci 0000:00:1f.3: 0xbe05c1a0: 19bd9028 ffff94f4 0dead000 00000000
2020-12-11T07:56:45.302405+08:00 ERR kernel: [46032.466704] sof-audio-pci 0000:00:1f.3: 0xbe05c1a4: 81759109 ffffffff 832abea0 ffffa0f7
2020-12-11T07:56:45.302408+08:00 ERR kernel: [46032.466711] sof-audio-pci 0000:00:1f.3: 0xbe05c1a8: 1e24e3c0 ffff94f4 00000000 00000000
2020-12-11T07:56:45.302410+08:00 ERR kernel: [46032.466717] sof-audio-pci 0000:00:1f.3: 0xbe05c1ac: 1e24e3c0 ffff94f4 00000002 00000000

brentlu, 2021-02-18:
Attach the system log.
[messages.log](https://github.com/thesofproject/sof/files/5999840/messages.log)

ranj063, 2021-02-18:
@brentlu can you please enable SOF dynamic debug logs and give us the dmesg logs again when the issue happens?

lgirdwood, 2021-02-18:
@mwasko @slawblauciak @abonislawski fyi - @brentlu can you try master, we had a similar issues on TGL that has been fixed in master. If you cant try master then please checkin with @mwasko to get the correct fixes on your production branch.

plbossart, 2021-02-18:
@lgirdwood @mwasko Just curious on how long those 'drop' releases are supposed to be maintained and is there an expectation that fixes are provided on those branches, instead of asking that mature products such as CML branch off the master?

brentlu, 2021-02-22:
@ranj063 looks like it only happens to specific devices. we are asking odm to see if they can reproduce the issue and provide sof logs.

mwasko, 2021-02-22:
> @brentlu can you try master, we had a similar issues on TGL that has been fixed in master. If you cant try master then please checkin with @mwasko to get the correct fixes on your production branch.

The integration of PRs to master that address TGL long youtube playback issue is in progress. As a quick check I would suggest to build CML image from [tgl-011-drop-stable](https://github.com/thesofproject/sof/tree/tgl-011-drop-stable) branch.

> @lgirdwood @mwasko Just curious on how long those 'drop' releases are supposed to be maintained and is there an expectation that fixes are provided on those branches, instead of asking that mature products such as CML branch off the master?

@plbossart the OTC releases for upstream are build from source that is regularly branched for release from master. That include CML.

brentlu, 2021-02-22:
There is a question from google that, does following PR also necessary for CML? And could it possibly be a fix?

https://github.com/thesofproject/sof/pull/3288

plbossart, 2021-02-22:
> @plbossart the OTC releases for upstream are build from source that is regularly branched for release from master. That include CML.

@mwasko your team is the owner of those releases and signs them.
And my question remains: for how long will the 'drop-NN' branches be supported and who's in charge here?

mwasko, 2021-02-23:
> @mwasko your team is the owner of those releases and signs them.
> And my question remains: for how long will the 'drop-NN' branches be supported and who's in charge here?

'drop-NN' branches are created and maintained during an active development for given target. Once the feature set is complete the support is continued on master branch with all other target platforms.

brentlu, 2021-03-02:
We've got sof and linux log from ODM. I didn't find error messages in sof log but some perf messages.

[10067700957.968750] (721327872.000000) c0 dai          1.4       src/audio/pipeline.c:198   perf comp_copy peak plat 782 cpu 3351
[23412225958.229168] (             nan) c0 dai          1.4       src/audio/pipeline.c:198   perf comp_copy peak plat 782 cpu 3374
[42166762108.177086] (             nan) c0 host         1.0       src/audio/pipeline.c:198   perf comp_copy peak plat 1780 cpu 8358
[73905642108.177094] (             nan) c0 host         1.0       src/audio/pipeline.c:198   perf comp_copy peak plat 1780 cpu 8361
[73905642229.218750] (      121.041664) c0 SCHEDULE_LL        ./schedule/ll_schedule.c:178   perf ll_work peak plat 6749 cpu 20817

[messages.txt](https://github.com/thesofproject/sof/files/6065765/messages.txt)
[sof-log.txt](https://github.com/thesofproject/sof/files/6065766/sof-log.txt)
[image]

keyonjie, 2021-03-02:
@brentlu, as @mwasko mentioned above, we are only monitoring the SOF master + Linux topic/sof-dev recipes on CML now, and no this issue observed by our CI recently.

lgirdwood, 2021-03-05:
@brentlu can they try v1.7-rc2 (it has known bugs - which are being fixed for v1.7).

lgirdwood, 2021-06-03:
@brentlu I'm assuming this has been fixed but not updated here  ?

plbossart, 2021-06-10:
@brentlu can you confirm is this bug still happens? we will close it without new information.

brentlu, 2022-03-29:
backport dma patches to cml-008-patch-backport2 branch

closed 2022-03-29 by brentlu