thesofproject/sof #3856: [FEATURE] Expand fuzzer into codec_adapter by stubbing libraries

cujomalainey, 2021-02-19:
**Is your feature request related to a problem? Please describe.**
Once oss-fuzz can reach components, it will still not be able to properly fuzz the codec_adapter as it cannot instantiate any of the closed source libs on x86 (let alone actually having the libs in the first place) but it will still be good to fuzz the code

**Describe the solution you'd like**
Keep stubs of the libraries in the oss-fuzz folder so we can enable the codec_adapter code

**Describe alternatives you've considered**
Cannot deploy libs to actual fuzzer as they are incompatible at arch level and chain of trust regarding safety of libraries is basically non existant to third party services

**Additional context**
N/A

cujomalainey, 2021-02-19:
Blocked by #3833

lgirdwood, 2021-03-08:
@cujomalainey should be unblocked now.

cujomalainey, 2021-03-08:
ack thanks, will work on this when i get a moment

cujomalainey, 2021-08-23:
Assigning rest of google folk, if anyone has spare cycles this would be something good for us to look at. Also adding another fuzz target to hit the IPCv4 fuzzer as well would be good.