thesofproject/sof #3864: [BUG] Corrupted audio - sof-tgl-max98373-rt5682 topology with disabled MAXIM component

oleggtitoff, 2021-02-22:
**Describe the bug**
We observe that audio output is corrupted when using SOF built from the master and disabling the MAXIM component in the sof-tgl-max98373-rt5682 topology. Please refer to the [sof-tgl-max98373-rt5682_without_maxim](https://github.com/stolx/sof-custom/tree/sof-tgl-max98373-rt5682_without_maxim) branch.
We are not able to build the MAXIM component, so this version is closest to the master.

**To Reproduce**
1. git clone https://github.com/thesofproject/sof.git and make the one-line change as in the [sof-tgl-max98373-rt5682_without_maxim](https://github.com/stolx/sof-custom/tree/sof-tgl-max98373-rt5682_without_maxim)
2. Setup Xtensa environment.
3. cd sof/
    rm -rf tmp
    mkdir tmp
    cd tmp
    cmake -DTOOLCHAIN=xt -DROOT_DIR=$CORE_PATH -DMEU_OPENSSL= '' - 
    DRIMAGE_PRIVATE_KEY=/home/otitov/xcc/meu/otc_private_key_3k.pem ..
    make tgplp_defconfig
    cp ../src/arch/xtensa/configs/override/tigerlake_chrome.config override.config
    make overrideconfig
    make bin -j
    cd ..
    ./scripts/build-tools.sh -T
4. Copy sof-tgl.ri, sof-tgl.ldc and topology to the device. Reboot the device.
5. stop cras

6) **[for speakers]** Run the script speakers_pb.sh ([speakers_pb.txt](https://github.com/thesofproject/sof/files/6022684/speakers_pb.txt))
7) **[for speakers]** aplay -Dhw:0,0 /root/pink15s.wav -c2 -r48000 -fs16_le

6. **[for headphones]** Run the script headset_pb.sh ([headset_pb.txt](https://github.com/thesofproject/sof/files/6022765/headset_pb.txt))
7. **[for headphones]** aplay -Dhw:0,1 /root/pink15s.wav -c2 -r48000 -fs16_le
8. Check the output
9. Press Ctrl+C to stop the aplay
10. Repeat steps 7-9 up to 20 times
11. Result: **[for speakers]** There is a chance to hear an output only from the L channel.
Result: **[for headphones]** The output sound is corrupted.

**Reproduction Rate**
[for speakers] Output only from the L channel - 5/20.
[for headphones] The output sound is corrupted - 20/20.

**Expected behavior**
Expected output from both channels. Output expected to be not corrupted.

**Impact**
Blocking third-party integration.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: stabilize-13729.16.B-chromeos-5.4 {0569763ea3d8b9de1644e1bee6a1ae247d376a66}
    * SOF: [sof-tgl-max98373-rt5682_without_maxim](https://github.com/stolx/sof-custom/tree/sof-tgl-max98373-rt5682_without_maxim) {91038c3aadbb678010587ea6ac1c5f6e3302f35c}
2) Name of the topology file
    * Topology: {sof-tgl-max98373-rt5682.m4}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {tigerlake_chrome}

**Screenshots or console output**
logs: [logs.zip](https://github.com/thesofproject/sof/files/6022943/logs.zip)
Input/output audio files: [audio.zip](https://github.com/thesofproject/sof/files/6022891/audio.zip)

slawblauciak, 2021-02-23:
@keyonjie please verify the topology

keyonjie, 2021-02-23:
@oleggtitoff we didn't use the topology as you do to validate the dummy sample DSM component, e.g.
define(`SMART_TX_CHANNELS', 4)
define(`SMART_RX_CHANNELS', 8)
define(`SMART_FB_CHANNELS', 8)

please refer to https://github.com/thesofproject/sof/pull/3655 about how we validate that.

oleggtitoff, 2021-03-03:
Hi @keyonjie, thanks a lot! It helped with the upstream master. No issues anymore.

closed 2021-03-03 by oleggtitoff