thesofproject/sof #3871: [FEATURE] Allow topologies to manifest and sanity check component IDs

cujomalainey, 2021-02-26:
**Is your feature request related to a problem? Please describe.**
Speaking with @plbossart he suggested an idea where, with the rise of closed source and a large number of processing components, we could sanity check the topology against the firmware rather than having a cryptic IPC failure when the topology requests a component not in the firmware.

**Describe the solution you'd like**
If we provide an ID manifest we can do simple checks early rather than failing mid load.

**Describe alternatives you've considered**
Keep using what we have

**Additional context**
I think this may become more important as other closed source adopters possibly start generating IDs we will need tools to help identify contents of various blobs when we start possibly getting reports (i might be speculating here)

lgirdwood, 2021-03-08:
IIRC, the UUIDs for each components are built into an ELF section today. Rimage currently parses all the ELF sections so we could add a command line flag to rimage and tell it to dump UUIDs (including from pre-built images).
Note: this could also be done in a python tool if better, we do have the manifest tool and it may be simpler to add the UUID dump here.

lgirdwood, 2021-05-10:
Partially implemented in v1.8 as manifest now includes the list of UUIDs that are built into the FW. What's missing now is automatic generation of this data by rimage and use by kernel. Slated for v1.9

lgirdwood, 2021-06-03:
Looks like this is part of IPC4 work.

RanderWang, 2023-05-09:
@marc-hb @fredoh9 as I we discussed it in above PR7523, I decided to integrate this feature into tplgtool2.py. As I only study python less than a week, I prefer to get advice from you.  I plan to add a command option like -fw xxx.fw and move the fw parsing code into tplgytool2.py or create a module library for fw parsing ?  Thanks for your feedback

marc-hb, 2023-05-09:
There are basically 2 options:
1. either you create a brand new file `mynewtool.py` that `import`s `tplgtool2.py`, treated as a library. See demo below. Two files, one new file.
2. or extend `tplgtool2.py` with the new feature(s) you want, adding some new --options. One file, no new file.

Demo of 1., already working today:
```
$ cd sof-test/tools
$ apt-install ipython # more user-friendly

$ ipython3

In [3]: import tplgtool2 # cwd is in the defaut sys.path

In [3]: tplgtool2.TplgBinaryFormat().parse_file('/lib/firmware/intel/sof-tplg/sof-whl-rt5682.tplg')

```

RanderWang, 2023-05-29:
done

closed 2023-05-29 by RanderWang