thesofproject/sof #3885: [BUG] LG Gram laptop (2021 Tiger Lake) Internal speaker no sound

NelsonJeppesen, 2021-03-03:
**Describe the bug**
On LG Gram laptop (17z90p) (11th gen tiger lake) there is no output from internal speaker. Headphones and built-in microphone work just fine. As expected, all works in Windows

**To Reproduce**
Play any audio from any source without headphones connected

I checked `alsamixer` and everything is un-mutted with volumed maxed out

**Reproduction Rate**
100%

**Expected behavior**
I expected audio to come out of the laptop without the need of headphones

**Impact**
Major

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.11.1
    * SOF:  sof-bin/b77c851bc4ec1b6b552eaf1a61a66f3df4a13ab8 (1.6.1)
 
2) Name of the topology file
    * Topology: Unclear on what this is
3) Name of the platform(s) on which the bug is observed.
    * Platform:  
          NixOS x86_64

**Screenshots or console output**
```
% dmesg|egrep -i "(dsp|sof|snd)"
[    0.514780] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.514780] software IO TLB: mapped [mem 0x000000002ef9a000-0x0000000032f9a000] (64MB)
[    6.170758] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    6.170762] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    6.240146] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    6.240151] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    6.240164] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    6.240299] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    7.521268] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    7.573908] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    7.596903] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    7.596906] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    7.596911] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    7.696212] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:6:1-53680
[    7.696215] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:17:0 Kernel ABI 3:18:0
[    7.701377] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:17:0 Kernel ABI 3:18:0
[    7.715760] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    7.735862] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC298: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    7.735867] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    7.735868] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    7.735870] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    7.735871] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    7.735872] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x18
[    7.776752] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[    7.776759] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[    7.776874] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi3 overwritten
[    7.776878] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi2 overwritten
[    7.776882] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi1 overwritten
[    7.776885] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Codec Output Pin1 overwritten
[    7.776888] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Codec Input Pin1 overwritten
[    7.776893] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Analog Codec Playback overwritten
[    7.776897] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Digital Codec Playback overwritten
[    7.776901] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Alt Analog Codec Playback overwritten
[    7.776906] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Analog Codec Capture overwritten
[    7.776910] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Digital Codec Capture overwritten
[    7.776915] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Alt Analog Codec Capture overwritten
[    7.776922] skl_hda_dsp_generic skl_hda_dsp_generic: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
[    7.776924] skl_hda_dsp_generic skl_hda_dsp_generic: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 4
[    7.776925] skl_hda_dsp_generic skl_hda_dsp_generic: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 5
[    7.776926] skl_hda_dsp_generic skl_hda_dsp_generic: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 6
[    7.776927] skl_hda_dsp_generic skl_hda_dsp_generic: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 7
[    7.776928] skl_hda_dsp_generic skl_hda_dsp_generic: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 8
[    7.792603] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input11
[    7.792672] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input12
[    7.792751] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input13
[    7.792807] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input14
[    7.792866] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input15
```

```
lspci -v | grep -i aud
00:1f.3 Multimedia audio controller: Intel Corporation Tiger Lake-LP Smart Sound Technology Audio Controller (rev 20)
	Kernel driver in use: sof-audio-pci
```

```
% ls -la /dev/snd/
total 0
drwxr-xr-x   3 root root      340 Mar  2 19:44 .
drwxr-xr-x  19 root root     3900 Mar  2 19:44 ..
drwxr-xr-x   2 root root       60 Mar  2 19:44 by-path
crw-rw----+  1 root audio 116, 13 Mar  2 19:44 controlC0
crw-rw----+  1 root audio 116, 12 Mar  2 19:44 hwC0D0
crw-rw----+  1 root audio 116, 11 Mar  2 19:44 hwC0D2
crw-rw----+  1 root audio 116,  5 Mar  2 19:44 pcmC0D0c
crw-rw----+  1 root audio 116,  4 Mar  2 19:44 pcmC0D0p
crw-rw----+  1 root audio 116,  7 Mar  2 19:44 pcmC0D1c
crw-rw----+  1 root audio 116,  6 Mar  2 19:44 pcmC0D1p
crw-rw----+  1 root audio 116,  8 Mar  2 19:44 pcmC0D3p
crw-rw----+  1 root audio 116,  9 Mar  2 19:44 pcmC0D4p
crw-rw----+  1 root audio 116, 10 Mar  2 19:44 pcmC0D5p
crw-rw----+  1 root audio 116,  2 Mar  2 19:44 pcmC0D6c
crw-rw----+  1 root audio 116,  3 Mar  2 19:44 pcmC0D7c
crw-rw----+  1 root audio 116,  1 Mar  2 19:44 seq
crw-rw----+  1 root audio 116, 33 Mar  2 19:44 timer
```

```
% aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

plbossart, 2021-03-03:
@NelsonJeppesen this is likely a codec driver bug. Can you check if you are able to check audio on your speakers with the SOF driver disabled?
see this link for details:

https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#disable-sof-on-pci-hdaudio-devices-to-test-audio-playback

NelsonJeppesen, 2021-03-03:
@plbossart No change - audio from headphones but not built in speaker. When I plug/unplug headphones I can see headmones or speakers muted correctly in alsamixer

To verify the driver changed:
```
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: ALC298 Analog [ALC298 Analog]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 11: HDMI 5 [HDMI 5]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 12: HDMI 6 [HDMI 6]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 13: HDMI 7 [HDMI 7]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 14: HDMI 8 [HDMI 8]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 15: HDMI 9 [HDMI 9]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 16: HDMI 10 [HDMI 10]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
``

plbossart, 2021-03-03:
@NelsonJeppesen then this is indeed a Linux codec driver issue. Something is missing in the configuration of the HDaudio codec to enable speakers.