thesofproject/sof #3895: [BUG] Cannot build HIFI3 optimized PCM converter on i.MX platforms

paulstelian97, 2021-03-04:
**Describe the bug**
I'm trying to build SOF for i.MX8 platforms with Xtensa toolchain. However, I'm getting build errors with the default KConfig.

#3894 is a workaround that permits building to complete without issue. However, it disables an optimization.

**To Reproduce**
Set up internal toolchain. Then run `scripts/xtensa_build_all.sh imx8` (or `imx8m` or `imx8x`, all manifest same issue)

**Reproduction Rate**
Compile error (thus 100%)

**Expected behavior**
Should build without the workaround

**Impact**
Annoyance (can merge workaround)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: N/A
    * SOF: 29b149a8
2) Name of the topology file
    * Topology: N/A
3) Name of the platform(s) on which the bug is observed.
    * Platform: imx8 imx8x imx8m

**Screenshots or console output**
```
[ 76%] Building C object CMakeFiles/sof.dir/src/audio/pcm_converter/pcm_converter_hifi3.c.o
cc1: warnings being treated as errors
/work/repos/sof-upd/src/audio/pcm_converter/pcm_converter_hifi3.c: In function ‘pcm_convert_f_to_s16_lin’:
/work/repos/sof-upd/src/audio/pcm_converter/pcm_converter_hifi3.c:681: warning: implicit declaration of function ‘XT_ROUND_S’
CMakeFiles/sof.dir/build.make:1334: recipe for target 'CMakeFiles/sof.dir/src/audio/pcm_converter/pcm_converter_hifi3.c.o' failed
make[3]: *** [CMakeFiles/sof.dir/src/audio/pcm_converter/pcm_converter_hifi3.c.o] Error 2
make[3]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:427: recipe for target 'CMakeFiles/sof.dir/all' failed
make[2]: *** [CMakeFiles/sof.dir/all] Error 2
CMakeFiles/Makefile2:1694: recipe for target 'src/arch/xtensa/CMakeFiles/bin.dir/rule' failed
make[1]: *** [src/arch/xtensa/CMakeFiles/bin.dir/rule] Error 2
Makefile:736: recipe for target 'bin' failed
make: *** [bin] Error 2
```

paulstelian97, 2021-03-09:
@slawblauciak I posted this as a bug after an internal discussion with @dbaluta. I have submitted the workaround (it is merged in fact) and mentioned pull request there.

Who wrote the actual HIFI3 optimized code?

dbaluta, 2021-03-09:
Our config is for HIFI4 maybe it is normal that HIFI3 shouldn't be supported?

lgirdwood, 2021-03-09:
@singalsu can you comment on the HiFi support