thesofproject/sof #3915: [FEATURE] Q_CONVERT_FLOAT does not include ppm error

ShriramShastry, 2021-03-10:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**In the moment, macro Q_CONVERT_FLOAT does not  return accurate  #define value for given input of float and power .**

For Example :  Q_CONVERT_FLOAT(1.999999998137354850769043f, 29) return value 1073741824 but
round(1.999999998137354850769043*2^29) expected value is 1073741823 . we have offset of 1 i.e. error in ppm = .-0.000093133 

As an option - Few proposal 

**option -1  Add offset in the macro**
```
#define Q_CONVERT_FLOAT_PRECISION(f, qy, offset) \
	((int32_t)(((const double)f) * ((int64_t)1 << (const int)qy) + ((const double)offset)))
```

**option-2 - C code to include the offset**
```
int32_t compensate_double_error(const int32_t reference_data[], int32_t number_ref_elements, int32_t calculate_data[], int32_t num_precomputed_element)
{
    int32_t loop_index;
    double  ppm_error[6] = {0.0f, 0.0f, 0.0f, 0.0f, 0.0f, 0.0f};
    for (loop_index = 0; loop_index <
        number_ref_elements; loop_index++)
    {
        if (reference_data[loop_index] != calculate_data[loop_index])  
        {
            if ((reference_data[loop_index] - calculate_data[loop_index]) < 0)  
            {
                calculate_data[loop_index] -= calculate_data[loop_index] - reference_data[loop_index];
            }   else  {
                calculate_data[loop_index] += reference_data[loop_index] - calculate_data[loop_index];
            }
        }
        else
        {
            calculate_data[loop_index] = calculate_data[loop_index];
        }
    }
    return calculate_data[loop_index];
}

```

**option -3 Calculate ppm error in the run time in the macro-definition i.e. improve current macro to avoid offset error**

```
#define Q_CONVERT_FLOAT(f, qy) \
	((int32_t)(((const double)f) * ((int64_t)1 << (const int)qy) + 0.5))

```

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

cujomalainey, 2021-03-10:
option 4, use off target preprocessed high precision values that are manually determined and copied into code while floats are kept nearby for reference. This only works currently because this macro is exclusively used with constants currently

singalsu, 2021-03-11:
Hey, have you tried `Q_CONVERT_FLOAT(1.999999998137354850769043d, 29)` ?

singalsu, 2021-03-11:
Looks like all but val1 are correct:

```
#include <stdio.h>
#include <stdint.h>

#define Q_CONVERT_FLOAT(f, qy) \
	((int32_t)(((const double)f) * ((int64_t)1 << (const int)qy) + 0.5))

void main()
{
	int64_t val1 = Q_CONVERT_FLOAT(1.999999998137354850769043f, 29);
	int64_t val2 = Q_CONVERT_FLOAT(1.999999998137354850769043, 29);
	int64_t val3 = Q_CONVERT_FLOAT(1.999999998137354850769043d, 29);
	int64_t val4 = Q_CONVERT_FLOAT(1.999999998137354850769043L, 29);

	printf("val1=%ld\n", val1);
	printf("val2=%ld\n", val2);
	printf("val3=%ld\n", val3);
	printf("val4=%ld\n", val4);
}

```

Gives

```
val1=1073741824
val2=1073741823
val3=1073741823
val4=1073741823

```

ShriramShastry, 2021-03-12:
> Hey, have you tried `Q_CONVERT_FLOAT(1.999999998137354850769043d, 29)` ?
I tried on my debugger - this is resulting `**term does not evaluate to function taking 374 argument**`
d in expression is not evaluated and I get error 
So not sure double works

ShriramShastry, 2021-03-12:
> Looks like all but val1 are correct:
> 
> ```
> #include <stdio.h>
> #include <stdint.h>
> 
> #define Q_CONVERT_FLOAT(f, qy) \
> 	((int32_t)(((const double)f) * ((int64_t)1 << (const int)qy) + 0.5))
> 
> 
> void main()
> {
> 	int64_t val1 = Q_CONVERT_FLOAT(1.999999998137354850769043f, 29);
> 	int64_t val2 = Q_CONVERT_FLOAT(1.999999998137354850769043, 29);
> 	int64_t val3 = Q_CONVERT_FLOAT(1.999999998137354850769043d, 29);
> 	int64_t val4 = Q_CONVERT_FLOAT(1.999999998137354850769043L, 29);
> 
> 	printf("val1=%ld\n", val1);
> 	printf("val2=%ld\n", val2);
> 	printf("val3=%ld\n", val3);
> 	printf("val4=%ld\n", val4);
> }
> ```
> 
> Gives
> 
> ```
> val1=1073741824
> val2=1073741823
> val3=1073741823
> val4=1073741823
> ```
> 
> I could not find from C resources which way is recommended for double precision constants. Do you know?

Thanks a lot, i did few more experiment with different number and tried to look inside decimal conversion

```
| f | 0.4f*2 | 0.80000001192092896| f | 0.4f*2 |   1.2000000476837158|
|   | 0.4*2  | 0.80000000000000004|   | 0.6*2  |   1.2000000000000000|
| L | 0.4L*2 | 0.80000000000000004| L | 0.6L*2 |   1.2000000000000000|

```

paulstelian97, 2021-03-12:
In C, float literals are done with an "f" at the end and double has no suffix (literals are double precision by default if there is a "." in them). Is the delta too small to be covered in single precision floating point? That would explain val1.

singalsu, 2021-03-12:
Thanks for explanation @paulstelian97 . I can't think of any need to use float literals when converting to fractions.

The default double precision literals (e.g. 0.51) seem to avoid with 32 bit fraction length the rounding error. The small error would likely happen with 64 bit fractions. But the error is really here time and here with 32 bit float it was about 1/1000 of PPM.

singalsu, 2021-03-12:
I found this for C++ apparently https://en.cppreference.com/w/cpp/language/floating_literal

"L" seems to be long double, though I'm not sure it's guaranteed to be more accurate than a double literal.

What's interesting is that C++-17 defines hex fractions, pretty much what this reported macro is about.

paulstelian97, 2021-03-12:
> Thanks for explanation @paulstelian97 . I can't think of any need to use float literals when converting to fractions.
> 
> The default double precision literals (e.g. 0.51) seem to avoid with 32 bit fraction length the rounding error. The small error would likely happen with 64 bit fractions. But the error is really tiny, with 32 bit float it was about 1/1000 of PPM.

To be clear, when converting from floating point to fixed point 32-bit, one ulp of the single precision translates to 256 of the signed int (or 128 or something). You need double precision if you want to cover differences of 1 in the fixed point format.

Long double is wider than double on x86_32 at least (where it's 80 bits, padded to 128). On other architectures it can be equivalent to double (I honestly did not try on x86_64 or on ARM to see what happens). Usually it's not worth dealing with. You either go double precision or software multi-precision numbers, long double is an intermediate that isn't needed.

The HIFI4 core in NXP boards supports single precision maths but not double precision [I'm 99% certain, though I still need to look at the config to be fully certain]. Single precision literals are useful in that situation. If you have double precision literals in an expression the entire calculation lifts to double precision and is done in-library rather than on-hardware.

Double precision literals when assigned directly (or casted to) single precision values can be done so at compile time so that works.

paulstelian97, 2021-03-12:
To add to this. 2 - one ulp of the single precision gives 1.99999988079. Only 6 nines after the decimal point. The literal in the example has 8 of them, which means that the floating point number itself is correctly rounded up to 2.0f even before the conversion proper starts. So the bug isn't in Q_CONVERT_FLOAT but in the fact that the single precision floating point literal is actually exactly equal to 2.0f.

singalsu, 2021-04-28:
@ShriramShastry This should be OK to close, right?

ShriramShastry, 2021-04-28:
Thanks , yes it can be closed.

closed 2021-04-28 by ShriramShastry