thesofproject/sof #3923: [BUG] Non-48 kHz rate playback via SRC pipeline into hda-generic headset wrong pitch and wrong length

singalsu, 2021-03-11:
**Describe the bug**
Playback at 48 kHz rate sounds correct and playback time is as expected, e.g. 10s length clip takes 10s time to play. If 32 kHz rate clip is played it takes about 48/32 amount of time, 10s takes 15s. The pitch is lower.

The source and sink buffer parameters show e.g. correct 32 and 48 kHz in params(), prepare() and copy(). SRC seems to get correct configuration. The amount from frames copied looks like DAI sink takes only 32 frames per copy. Much like the DAI would operate at that rate instead of fixed 48 kHz.

**To Reproduce**
Currently there is no topology with SRC for hda-generic. Such can be done with edit:

```
diff --git a/tools/topology/development/CMakeLists.txt b/tools/topology/development/CMakeLists.txt
index 11e462ae1..a775e2c8c 100644
--- a/tools/topology/development/CMakeLists.txt
+++ b/tools/topology/development/CMakeLists.txt
@@ -28,6 +28,7 @@ set(TPLGS_UP
        "sof-cml-rt5682\;sof-cml-eq-iir-rt5682\;-DPLATFORM=cml\;-DHSEARPROC=eq-iir-volume\;-DDMICPROC=eq-iir-volume\;-DDMIC16KPROC=eq-iir-volume"
        "sof-glk-da7219\;sof-glk-eq-da7219\;-DDMICPROC=eq-iir-volume"
        "sof-hda-generic\;sof-hda-generic-tdfb_50mm-2ch\;-DCHANNELS=2\;-DHSPROC=volume\;-DDMIC16KPROC=tdfb-eq-iir-volume\;-DDMIC16KPROC_FILTER1=tdfb/coef_line2_50mm_pm10deg_16khz.m4\;-DDMICPROC=tdfb-eq-iir-volume\;-DDMICPROC_FILTER1=tdfb/coef_line2_50mm_pm10deg_48khz.m4\;-DDMICPROC_FILTER2=eq_iir_coef_highpass_40hz_20db_48khz.m4\;-DDMIC16KPROC_FILTER2=eq_iir_coef_highpass_40hz_20db_16khz.m4"
+       "sof-hda-generic\;sof-hda-generic-2ch-src\;-DCHANNELS=2\;-DHSPROC=src-volume\;-DDMICPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_48khz.m4\;-DDMIC16KPROC_FILTER1=eq_iir_coef_highpass_40hz_20db_16khz.m4"
 )
 
 add_custom_target(dev_topologies ALL)

The built sof-hda-generic-2ch-src.tplg needs to be copied over default sof-hda-generic-2ch.tplg. The SRC is placed to headset playback pipeline.

```

**Reproduction Rate**
100%

**Expected behavior**
Playback sounds correct and playback time matches clip length.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

singalsu, 2021-03-11:
I just checked that this does not happen with sof-cml-rt1011-rt5682 platform with I2S headset and speaker codecs. It seems to be HDA specific, I haven't tried SDW. I wonder if the Linux HDA driver actually configures the HDA DAI to 32 kHz rate with topology with PCM capability other than min/max 48 kHz. The SRC pipeline sets PCM capability to 8 - 192 kHz.

plbossart, 2021-03-11:
@singalsu I am confused, are you trying to change the output sample rate, or just allow for 32kHz input to be played at the fixed 48kHz rate?

singalsu, 2021-03-12:
@plbossart  The latter, I was assuming HDA DAI runs at 48 kHz.

The purpose of the pipe-src-volume-playback.m4 is to provide a PCM that supports 8 - 192 kHz playback into a fixed rate 48 kHz DAI. If the HDA DAI is configured to stream rate by the Linux driver then this is not a bug. The SRC playback pipeline should not be used with HDA if it does internally the resampling or re-configuring of codec to ALSA device sample rate.

plbossart, 2021-03-12:
Now I am completely confused. I thought we could only play at 48kHz with HDaudio, it's been a recurring issue with e.g. HDMI.

singalsu, 2021-03-16:
I just did a new test. I dropped SRC from playback pipeline but preserved the PCM capability 8 - 192 kHz. I also added trace output to hda_get_hw_params() to print params->rate and params->channels.

If I attempt play 32 or 48 kHz content the hda_get_hw_params() reports 48 kHz DAI configuration. There's trace output:

```
[235818230.708333] (       12.958333) c0 dai          1.5            src/audio/dai.c:308  ERROR dai_verify_params(): pcm rate parameter 44100 does not match hardware rate 48000
[235818246.291667] (       15.583333) c0 dai          1.5            src/audio/dai.c:454  ERROR dai_params(): pcm params verification failed.
[235818261.458333] (       15.166667) c0 pipe         1.6       src/audio/pipeline.c:576  ERROR pipeline_params(): ret = -22, host->comp.id = 1
[235818278.916667] (       17.458334) c0 ipc                       src/ipc/handler.c:276  ERROR ipc: pipe 1 comp 1 params failed -22

```
I'll assign this myself study more what happens in kernel and FW when SRC is included to pipeline.

singalsu, 2021-03-16:
I commented out the -EINVAL return in line 309 in dai_verify_params() that stopped params() walk to above error. Now I can play cleanly without SRC 32 kHz to HDA codec headphone out while hda_get_hw_params() reports 48 kHz. Playback of 10s clip takes 10s real-time.

- 16 kHz plays and sounds OK
- 32 kHz plays and sounds OK
- 44.1 kHz results to "ERROR hda-dmac: 7 chan 0 - buffer not DMA aligned 0x2d0". So even if we could fix the issue that caused from info at  hda_get_hw_params() there's other things preventing to allow other than 48 kHz rates playback without SRC.
- 48 kHz as expected works OK

singalsu, 2021-03-16:
@kv2019i @plbossart Kai suggested to check from hdac_hda.c the sample rate. I made this quick add:

```
diff --git a/sound/soc/codecs/hdac_hda.c b/sound/soc/codecs/hdac_hda.c
index 390dd6c7f6a5..614ef6848f3f 100644
--- a/sound/soc/codecs/hdac_hda.c
+++ b/sound/soc/codecs/hdac_hda.c
@@ -219,6 +219,11 @@ static int hdac_hda_dai_hw_params(struct snd_pcm_substream *substream,
                                                 params_format(params),
                                                 maxbps,
                                                 0);
+       dev_err(dai->dev,
+               "Hello HDA, rate=%d, ch=%d, format=%d, maxbps=%d\n",
+               params_rate(params), params_channels(params),
+               params_format(params), maxbps);
+
        if (!format_val) {
                dev_err(dai->dev,
                        "invalid format_val, rate=%d, ch=%d, format=%d, maxbps=%d\n",

```
When I play 16 kHz content I get now to kernel log:

`kernel: snd_hda_codec_realtek ehdaudio0D0: Hello HDA, rate=16000, ch=2, format=10, maxbps=24
`

kv2019i, 2021-03-18:
This is very interesting. I think we do have a gap here. Basicly we have no way to inform the kernel how the sampling rate is changed in the pipeline. This is a not a problem if the DAI/BE has a fixed rate. But if that rate can be programmed, then things will go wrong.

I was looking at how others do it in ASoC and there are not a lot of example. Closest matching is the "convert_rate" OF property of ASOC simple-card (soc/generic/simple-card-utils.c:asoc_simple_parse_convert()). Basicly if the HW/DSP pipeline has a SRC, convert rate is set to the SRC output rate, and that is configured in the fixup, so that the codec DAI gets programmed with correct parameters.

It would seem we are lacking means to specify this in SOF topology. I.e. if we program a SRC in a certain way, we'd need to be able to express similar constraint as the convert-rate OF property.

FYI @plbossart @keyonjie @ranj063

plbossart, 2021-03-18:
Yep, we have a gap in that the kernel doesn't know what the SRC does. the hw_params are either set to the FE values (i.e. defined by the PCM rate), or by the DAI configuration via the fixup. if the SRC does something else via its own control channels, then things can go sideways.

singalsu, 2021-03-18:
@plbossart In this case SRC worked as pipeline instructed, no bug there. The problem is that kernel didn't update to FW DAI the actual sample rate. It wasn't the initialized 48 kHz that pipeline propagated. If SRC had got the correct rate from DAI in pipeline walk it would have configured itself to 1:1 copy.

You and Kai are right that there's no control from topology what should happen.