thesofproject/sof #3932: [BUG][CHT] loud crackle from speakers at boot, which does not happen with SST driver

jwrdegoede, 2021-03-17:
This has been discussed before, but because the crackle became an itch to scratch for me I've ran some more tests and I' now filing an issue for this. Specifically with a 5.12-rc2 kernel on a HP Pavilion X2 10-p002nd with a ALC3276 (rt5640) codec. This is the same device where the channel-swap issue also most easily reproduced. It is as if this device has more randomness in non-initialized memory, or has something else going on which makes it more susceptible to these bugs. 

When using the SOF driver there is a loud crackle on boot when userspace opens the device for the first time and a slightly less loud crackle on subsequent opens.

When using the SST driver there is no audible crackle. This happens both with vanilla SOF-1.6.1 firmware and with the test 1.7 build which I've been using which contains the fix for the channel-swap issue.

I guess that either somehow the FIFOs / DSP I2S DMA ring-buffers contain non silence (random data?) which gets send to the codec when powering up, or that maybe there is some difference in the power-up sequence or timing between SOF/SST.

One thing is very clear to me now, after running multiple tests, this definitely is something which only happens when using the SOF driver and which goes away when switching back to the SST driver.

Interestingly enough so far I've not been able to reproduce this on BYT devices, I've only seen this on CHT devices so far. But I've only tried on BYT device sofar, the HP Pavilion X2 10-n000nd which is the BYT based previous version of the CHT based HP Pavilion X2 10-p002nd where I'm very clearly hearing the crackle with the SOF driver. I deliberately used this BYT model because it is very close to its CHT counter-part.

plbossart, 2021-03-17:
@jwrdegoede There are indeed major differences in the power sequence between SST and SOF drivers. The former requires the DSP to be on and start transfers of zero data, whereas for SOF we rely on DAPM/DPCM.

There's been a long standing problem that some codecs require the bclock and fsync to be stable for a while before sending data. This happens by construction with SST, but not for SOF.

Could you try and hack something where you add e.g. 50ms of silence to the beginning of the initial track that gets played?

jwrdegoede, 2021-03-18:
> Could you try and hack something where you add e.g. 50ms of silence to the beginning of the initial track that gets played?

The crackle happens when pulseaudio starts, at this point in time nothing is playing, it is merely opening
the audio device. I guess it may play some silence at start to determine some hw params ?

plbossart, 2021-03-18:
@jwrdegoede if this is PulseAudio only opening devices and setting-up hw_params, then this points to a DAPM issue. If PulseAudio starts streaming then it's an SOF issue.

jwrdegoede, 2021-03-19:
> @jwrdegoede if this is PulseAudio only opening devices and setting-up hw_params, then this points to a DAPM issue. If PulseAudio starts streaming then it's an SOF issue.

So what is the best way to test this?

Would disabling pulseaudio and then after a fresh boot do something like this:

```
aplay --dump-hw-params -D"hw:1,0" -r48000 -c2 -fS24_LE test.raw
```

With a zero byte sized test.raw work to test the opening + setting up hw_params

And maybe also repeat this (after a fresh boot) with a test.raw files with zeros ?

lgirdwood, 2021-03-20:
@jwrdegoede yes, that would be a good test. You can play /dev/zero as silence if needed.
Fwiw, I would also try an arecord after boot too, i.e. to check if general codec power on sequence is causing the crackle or is ir DAC path only.

jwrdegoede, 2021-03-31:
Sorry I've not had time run the discussed tests yet, but I did make one interesting observation yesterday while working on getting the audio an a Dell Venue 10 5055 Pro to work. This is a Bay Trail CR device with a rt5672 codec.

Since this is a Bay Trail CR device SSP0 must be used (SSP2 is not routed on the SoCs pins), but the machine-driver atm only supports SSP2, so the audio was always being send to SSP2 with both the SOF and SST drivers, leading to silence.

Except that when using the SOF driver there was a crackle from the speakers, which is not there with the SST driver. Since we were using the wrong I2S output here, I think this rules out the FIFO flush theory (where there is some random data in the FIFO being send at startup).

Which leaves us with the power-sequence theory.  So it seems that there is something different in the SOF drivers power-sequence which is causing these loud crackle sounds the first time the sound-card is opened from userspace.

I have a hunch that SOF powers-up the codec even if userspace is just "probing" where as SST only powers it up when there is actual streaming of audio going on, not sure about that at all though.

 Anyways I believe that this observation answers @plbossart's question " if this is PulseAudio only opening devices and setting-up hw_params, then this points to a DAPM issue. If PulseAudio starts streaming then it's an SOF issue." and it seems that this is a DAPM issue.

lgirdwood, 2021-03-31:
@jwrdegoede thanks for the update, I think @plbossart suggestion that the SSP clocks need to be running during codec initialisation is probably the reason for the crackle. I've come across codecs in the past that need the clocks running otherwise they wont clock lock or power ON/OFF correctly.

bardliao, 2021-04-01:
> @jwrdegoede thanks for the update, I think @plbossart suggestion that the SSP clocks need to be running during codec initialisation is probably the reason for the crackle. I've come across codecs in the past that need the clocks running otherwise they wont clock lock or power ON/OFF correctly.

I am pretty sure that rt5640 needs clock in the initialization stage. Please correct me if I am wrong @oder-chiou

jwrdegoede, 2021-04-01:
> How difficult is it to try and enable SSP0 for this device and retry?

Not that difficult and I was planning on doing that anyway since it is necessary to get working sound.

When using a machine-driver which properly supports SSP0 not much changes, still a crackle on boot (boot on a cold start and from a reboot) when gdm shows the greeter (so the first time the sound-server starts).

And the crackle still goes away when switching back to SST.

Unrelated, but interesting: I noticed that there already is a sof-byt-rt5670-ssp0.tplg file, even though the machine driver does not support this. I was expecting that I also needed to do a patch for the tplg file.

plbossart, 2021-04-01:
@jwrdegoede what happens if you disable the sound server, and play with aplay -Dhw:x.y later?
Trying to decorrelate the driver probe from use.

Also all BYT topologies are identical, so there's no difference to be expected with a name change.

plbossart, 2021-06-10:
Adding the BYT label since it's really the same as CHT.

We might be able to solve this problem with the bitclock early start done for cAVS platforms.

mengdonglin, 2022-06-21:
Move BYT and CHT platform to maintenance mode and SOF v2.2 will be the last upstream release with support for this platform. There will be no more FW bug fixing effort on this platform.

closed 2022-06-21 by mengdonglin