thesofproject/sof #3937: [BUG][BYT] SOF hangs, logger shows "dw_dma_start(): dma 1 channel 1 not ready" error

jwrdegoede, 2021-03-19:
While running some tests on a HP Pavilion X2 10-n000nd, I noticed that audio would stop working when I simply increase / decrease the volume a couple of times using the volume up/down keys on the keyboard.
When changing the volume like this GNOME will play a "plop"(ish) sound on every keypress both as an auditory acknowledge and so that the user can (briefly) hear the new volume.

I think the problem is that pressing the keys with the right interval results in the device being opened - closed a number of times in quick succession (once for each plop sound) and the SOF stack cannot handle this it seems.

The HP Pavilion X2 10-n000nd uses a Bay Trail Z3736F SoC with an ALC5642 (rt5640) codec.

I'm seeing this with Fedora 34 beta using pipewire, running a 5.12-rc3 kernel. I believe that pipewire is likely a factor here. Both because it will use 24/32 bits samples when available instead of 16 bits; and I guess it also has different behavior wrt opening / closing the device when not in use. AFAIK pulseaudio waits 5 seconds with closing the device after the last audio has been played.

I've seen this both with the SOF 1.7 test firmware build which I've gotten from one of the SOF devs to test the channel-swap issue seen on BYT/CHT, as well as with a vanilla 1.6.1 from sof-bin.

With the 1.7 test build, I got a bunch of kernel errors:

```
[  866.594276] sof-audio-acpi-intel-byt 80860F28:00: error: ipc timed out for 0x60010000 size 108
[  866.594326] sof-audio-acpi-intel-byt 80860F28:00: info: preventing DSP entering D3 state to preserve context
[  866.594413] sof-audio-acpi-intel-byt 80860F28:00: error: unexpected fault 0x600a0008 trace 0x00000000
[  866.594443] sof-audio-acpi-intel-byt 80860F28:00: error: ipc host -> DSP: pending no complete no raw 0x8000000000000000
[  866.594469] sof-audio-acpi-intel-byt 80860F28:00: error: mask host: pending no complete no raw 0x0
[  866.594493] sof-audio-acpi-intel-byt 80860F28:00: error: ipc DSP -> host: pending no complete yes raw 0x600a0008
[  866.594518] sof-audio-acpi-intel-byt 80860F28:00: error: mask DSP: pending no complete no raw 0x0
[  866.594543] sof-audio-acpi-intel-byt 80860F28:00: error: waking up any trace sleepers
[  866.594568] sof-audio-acpi-intel-byt 80860F28:00: error: hw params ipc failed for stream 0
[  866.594590] sof-audio-acpi-intel-byt 80860F28:00: error: set pcm hw_params after resume
[  866.594611] sof-audio-acpi-intel-byt 80860F28:00: ASoC: error at snd_soc_pcm_component_prepare on 80860F28:00: -110
[  866.594637]  PCM: ASoC: prepare FE PCM failed
```

With the 1.6.1 build I did not see any kernel erorrs, but I do have the .lc file and sof-logger binary for that one, so I've grabbed sof-logger output for the hang with the 1.6.1 build:

```
         TIMESTAMP              DELTA C# COMPONENT          LOCATION                      CONTENT
[ 23200152.708333] ( 23200152.000000) c0 dma-trace             src/trace/dma-trace.c:327  ERROR FW ABI 0x3011000 DBG ABI 0x5002000 tag v1.6.1 src hash 0x459cfb8c (ldc hash 0x459cfb8c)
[159749081.822917] (136548928.000000) c0 dw-dma                 src/drivers/dw/dma.c:283  ERROR dw_dma_start(): dma 1 channel 1 not ready ena 0x2 status 0x3
[159749096.562500] (       14.739583) c0 host         3.15          src/audio/host.c:317  ERROR host_copy_one_shot(): dma_copy() failed, ret = 4294967280
[159749107.708333] (       11.145833) c0 pipe         3.19      src/audio/pipeline.c:1029 ERROR pipeline_copy(): ret = -16, start->comp.id = 16, dir = 1
[159749120.989583] (       13.281250) c0 pipe         3.19      src/audio/pipeline.c:1219 ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[698705316.302083] (538956224.000000) c0 dma-trace             src/trace/dma-trace.c:494  ERROR dtrace_add_event(): number of dropped logs = 1
```

jwrdegoede, 2021-03-19:
I've tried to reproduce this while forcing pipewire to 16 bits audio, by uncommenting the:

```
audio.format            = "S16LE"
```

line in "/etc/pipewire/media-session.d/alsa-monitor.conf" and after that I can no longer reproduce the problem.

jwrdegoede, 2021-03-19:
> Can you also confirm if the same test fails on newer (than Kabylake) hardware ?

The problem does not seem to reproduce on a ThinkPad X1 carbon 8th gen with a  "Intel(R) Core(TM) i7-10610U CPU @ 1.80GHz" CPU.

lgirdwood, 2021-03-20:
@jwrdegoede ok thanks, I think we may need to restrict BYT/CHT in topology to 16bit only for then time being. The 24bit DMA settings do have some differences at the SSP port level too and IIRC there is probably a golden DMA/SSP config for this with 24 bit.

plbossart, 2021-06-10:
@lgirdwood what did you mean by "restrict BYT/CHT in topology to 16bit only"? we can restrict the DAI or the PCM or both.

lgirdwood, 2021-06-11:
Restrict the DAI - it seems SSP likes 16 bit best.

plbossart, 2021-06-11:
> Restrict the DAI - it seems SSP likes 16 bit best.

we enabled the SSP with 24bits in the past, only 16bit was used to talk to the modem on SSP0.
I'll try to cook-up a patch to change this to 16bits, we are in work-around mode here.

mengdonglin, 2022-06-21:
Move BYT platform to maintenance mode and SOF v2.2 will be the last upstream release with support for this platform. There will be no more bug fixing effort on this platform.

closed 2022-06-21 by mengdonglin