thesofproject/sof #3939: [BUG][TGL-NOCODEC][v1.7] Xrun on DMIC in multiple-pipeline test

aiChaoSONG, 2021-03-22:
**Describe the bug**
In multiple pipeline test case, an xrun is reported on DMIC pipeliine.
report ID:  2698

**To Reproduce**
run script: multiple-pipeline.sh -f c -l 200

**Reproduction Rate**
low, at 109 of 200

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {topic/sof-dev d289bfd9}
    * SOF: {stable-v1.7 47d077d9d23e}
2) Name of the topology file
    * Topology: {sof-tgl-nocodec-ci.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {TGL NOCODEC}

```
[52344094331.250000] (             nan) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1551 ERROR dmic_irq_handler(): full fifo A or PDM overrun
[52344095215.729172] (      884.479187) c0 dw-dma                 src/drivers/dw/dma.c:1101 ERROR dw_dma_get_data_size(): xrun detected
[52344095235.885422] (       20.156250) c0 dai          8.43           src/audio/dai.c:801  ERROR dai_report_xrun(): overrun due to no space available
[52344095255.572922] (       19.687500) c0 dai          8.43           src/audio/dai.c:693  ERROR comp_overrun(): sink->free = 768, copy_bytes = 0
[52344095344.270836] (       88.697914) c0 pipe         8.44      src/audio/pipeline.c:1044 ERROR pipeline_copy(): ret = -61, start->comp.id = 43, dir = 0
[52344095367.968750] (       23.697916) c0 pipe         8.44      src/audio/pipeline.c:1234 ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[52345105270.000000] (  1009902.000000) c0 dai          8.43     src/audio/component.c:209  ERROR comp_set_state(): wrong state = 1, COMP_TRIGGER_STOP
[52345105292.291672] (       22.291666) c0 pipe         8.44      src/audio/pipeline.c:899  ERROR pipeline_trigger(): ret = -22, host->comp.id = 41, cmd = 0
[52345105315.468750] (       23.177084) c0 ipc                       src/ipc/handler.c:443  ERROR ipc: comp 41 trigger 0x50000 failed -22
```

keyonjie, 2021-03-22:
This should not be bandwidth issue as the 4 capture pipelines should take < 20% of the bandwidth only.

lgirdwood, 2021-03-22:
@mwasko @slawblauciak looks like the DMIC IRQ handler again, did we not remove this since it did nothing but report an error (and consume cycles) ?

slawblauciak, 2021-03-23:
@zrombel can you try reproducing it on our RVPs?

zrombel, 2021-03-23:
Ok, will do.

mwasko, 2021-03-23:
> @mwasko @slawblauciak looks like the DMIC IRQ handler again, did we not remove this since it did nothing but report an error (and consume cycles) ?

@lgirdwood we cannot completely ignore and suppress DMIC xrun. However to avoid DMIC xrun storm of interrupts the change was introduced to stop DMIC when xrun occurs.

keyonjie, 2021-03-24:
I realize there could be this possibility for DMIC pipelines, as the DMIC is started first and DMA later. The reason we don't see this in the SSP pipelines is we are not observing xrun interrupts there.

keyonjie, 2021-03-24:
once we can reproduce this stably, maybe worth to try if the PR #3952 help.

lgirdwood, 2021-05-20:
@keqiaozhang still an issue today ?

lgirdwood, 2021-05-27:
@keqiaozhang @XiaoyunWu6666  still an issue today ?

keqiaozhang, 2021-05-31:
@lgirdwood , yes, this issue can still be reproduced on TGL-RVP-NOCODEC platform in today's CI test. Daily test ID:4290

keyonjie, 2021-06-03:
@keqiaozhang @XiaoyunWu6666 looks this is fixed according to the latest daily result, please help to double check and close it if true.

XiaoyunWu6666, 2021-06-03:
@keyonjie @keqiaozhang  I'll check it

aiChaoSONG, 2021-06-03:
run test case 500 iterations, this issue is not reproduced, close.

closed 2021-06-03 by aiChaoSONG