thesofproject/sof #3947: [BUG][TGL-012-drop-stable] DSM Multicore Stream Stall

zrombel, 2021-03-23:
**Describe the bug**
In DSM test where DSM component is run on CORE 1 and rest of topology on default CORE 0 after triggering pipelines start, playback pipeline stops copying data.

**Topology**
```
    +----------------------------------------------+
    | +------+   +---+   +- -+   +---+   +-------+ |
    | | Host |-->|Buf|-->|Dsm|-->|Buf|-->|SSP Dai|-------+
    | +------+   +---+   +---+   +---+   +-------+ |     |
    +------------------+   ^   +-------------------+     |
                       |   |   |                         |
                       | +---+ |                         |
                       | |Buf| |                         |
                       | +---+ |                         |
                       +---^---+                         |
    +----------------------|------------------------+    |
    | +------+   +---+   +-|--+   +---+   +-------+ |    |
    | | Host |<--|Buf|<--|Dmux|<--|Buf|<--|SSP Dai|<-----+
    | +------+   +---+   +----+   +---+   +-------+ |
    +-----------------------------------------------+
```

**To Reproduce**
Run Python test:
22_05_TestDsmMulticoreSsp48000Hz24b32b2ch or 22_05_TestDsmMulticoreSsp48000Hz32b32b2ch
with parameters:
--playback-iterations=100

**Reproduction Rate**
~1/10

**Environment**
1) Name of the platform(s) on which the bug is observed.
    * Platform: TGL B0 RVP
2) Firmware branch name and commit 
    * Branch: tgl-012-drop-stable
    * Hash: fa35ee0
3) Python tests branch name and commit
    * Branch: master
    * Hash: 237e3511f3787aa9b45c53b43bea33cfa2646114

**Logs**
[22_05_TestDsmMulticoreSsp48000Hz32b32b2ch FAIL.zip](https://github.com/thesofproject/sof/files/6188732/22_05_TestDsmMulticoreSsp48000Hz32b32b2ch.FAIL.zip)

slawblauciak, 2021-04-02:
I was able to bisect and find which change introduces this behavior:
https://github.com/thesofproject/sof/commit/5104f5e02df5a720f2896d8403a880cde286f657
However, the same change exists on tgl-011-drop-stable branch, and the problem does not occur there.
This leads me to believe that this behavior is a result of at least two changes. 
I suspect the problem might have its source in the scheduler again, as such I will continue debugging from that side.

keyonjie, 2021-04-06:
Hi @slawblauciak @zrombel  can you try if reverting https://github.com/thesofproject/sof/commit/1fa1001e24a8bc8db419e8ff97aaf591ae8ceee5 will help?

lgirdwood, 2021-04-13:
@keyonjie @slawblauciak any updates here ?

slawblauciak, 2021-04-13:
@lgirdwood I was only able to eliminate some possibilities: it's most likely not related to IDC or spin locks. I had some trace points set up around those and there were no oddities there. Additionally, when the issue occurs I can see that none of the pipelines actually work, the pipeline tasks are not called.
I'll be debugging it further from scheduler side.

slawblauciak, 2021-04-13:
Also, neglected to mention that this issue is only reproduced with Python tests under slim driver. No attempt of reproduction under full stack was successful.

lgirdwood, 2021-04-13:
@slawblauciak thanks, btw how is core 1 work scheduled and synchronized with core 0 in this case ?

slawblauciak, 2021-04-14:
@lgirdwood good question, I'm not exactly familiar with the way SMP is implemented in the firmware.

lgirdwood, 2021-04-14:
@slawblauciak the issue I'm thinking here if core 1 work takes too long for one iteration and how this could spill over into the core0 work (and how we could detect/report and recover)..

slawblauciak, 2021-04-14:
@lgirdwood that was my initial though as well, but I believe the work on the cores is scheduled entirely independently and without synchronization, except for the locks in the buffer access. But again, I'm not 100% sure.

slawblauciak, 2021-04-22:
Upon further debug I was able to partly determine source of the problem. There was a small discrepancy between latest changes made to the scheduler, between tgl-011 and tgl-012. Upon aligning the changes, I was able to get the scheduler to run somewhat properly now. However, the issue still occurs, which suggests that the task might be being incorrectly cancelled or erroneously handled otherwise. Unfortunately I lack the required understanding of the scheduler as a whole and the changes made to be able to tell why aligning them improves the behavior at this moment.

slawblauciak, 2021-04-23:
@zrombel please verify the effectiveness of the linked PR

closed 2021-04-26 by lgirdwood

zrombel, 2021-04-27:
Fixed by https://github.com/thesofproject/sof/pull/4088, similar bug with different root cause found: https://github.com/thesofproject/sof/issues/4101

PR #4088 (2021-04-26): scheduler: guard against subdivision underflow on domain clear
In some situations the domain clear routine might encounter
a case, when the number of clients is decremented and 0,
as such, a subsequent subdivision will result in an integer underflow.

Fixes #3950 
Fixes #3949 
Fixes #3947
PR #4089 (2021-05-13): LL scheduler domain refinement
clean up of the ll_scheduler domain clients/tasks management.

    ll_schedule: domain client/task refinement
    
    Refinement to make the domain client/task management more clear:
    
    a. update the total_num_tasks and num_clients in the helper pair
    domain_register/unregister().
    
    b. update the enabled[core] and enabled_clients in the helper pair
    domain_enable/disable().
    
    c. for the completed task, do the domain client/task update in the new
    created helper schedule_ll_task_done().
    
    d. cleanup and remove the client/task m [... cut]