thesofproject/sof #3950: [BUG][TGL-012-drop-stable] SRC Multicore Stream Stall

zrombel, 2021-03-23:
**Describe the bug**
In SRC test where SRC component is run on CORE 1 and rest of topology on default CORE 0 after triggering pipelines start, playback pipeline stops copying data.

**Topology**
```
                    pipe_plb core 0, src core 1
           +--------------------------------+
    +----+ | +---+  +---+  +---+  +-------+ |
    |Host|-->|Buf|->|Src|->|Buf|->|SSP Dai|---+
    +----+ | +---+  +---+  +---+  +-------+ | |
           +--------------------------------+ |
                    pipe_cap core 0           |
           +------------------+               |
    +----+ | +---+  +-------+ |               |
    |Host|<--|Buf|<-|SSP Dai|<----------------+
    +----+ | +---+  +-------+ |
           +------------------+
```

**To Reproduce**
Run Python test:
17_06_TestMultiCoreCompSrcPlb48000Hz32b32b4c
with parameters:
--playback-iterations=100

**Reproduction Rate**
~1/50

**Environment**
1) Name of the platform(s) on which the bug is observed.
    * Platform: TGL B0 RVP
2) Firmware branch name and commit 
    * Branch: tgl-012-drop-stable
    * Hash: fa35ee0
3) Python tests branch name and commit
    * Branch: master
    * Hash: 237e3511f3787aa9b45c53b43bea33cfa2646114

**Logs**
[17_06_TestMultiCoreCompSrcPlb48000Hz32b32b4ch FAIL.zip](https://github.com/thesofproject/sof/files/6189004/17_06_TestMultiCoreCompSrcPlb48000Hz32b32b4ch.FAIL.zip)

slawblauciak, 2021-04-02:
Possibly related to https://github.com/thesofproject/sof/issues/3947
More information in the comments of the aforementioned issue.

lgirdwood, 2021-04-13:
@keyonjie @slawblauciak any updates ?

slawblauciak, 2021-04-23:
@zrombel please verify the effectiveness of the linked PR

closed 2021-04-26 by lgirdwood

zrombel, 2021-04-27:
Fixed by https://github.com/thesofproject/sof/pull/4088

PR #4088 (2021-04-26): scheduler: guard against subdivision underflow on domain clear
In some situations the domain clear routine might encounter
a case, when the number of clients is decremented and 0,
as such, a subsequent subdivision will result in an integer underflow.

Fixes #3950 
Fixes #3949 
Fixes #3947