thesofproject/sof #3953: [BUG][TGL] XRUN issue on TGL_RVP_NOCODEC when testing multiple pipelines

keqiaozhang, 2021-03-24:
**Describe the bug**
CI observed xrun on TGL_RVP_NOCODEC when testing capture on multiple pipelines.  xrun happened on dmic pipeline.

**To Reproduce**
$ ./sof-test/test-case/multiple-pipeline.sh -f c -l 50

**Reproduction Rate**
first time CI observed this issue.

**Expected behavior**
No XRUN

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 4e30747d
SOF Branch: master
SOF Commit: 041c2ef6de3e
Platform: TGL_RVP_NOCODEC
Test ID: daily/2180

**etrace**
```
[10587035651.093750] (             nan) c0 dma-trace             src/trace/dma-trace.c:339  ERROR FW ABI 0x3012001 DBG ABI 0x5003000 tag v1.7-rc1-155-g041c2ef6de3e src hash 0xa0955020 (ldc hash 0xa0955020)
[11070118937.187500] (483083296.000000) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1551 ERROR dmic_irq_handler(): full fifo A or PDM overrun
[11070119816.875000] (      879.687500) c0 dw-dma                 src/drivers/dw/dma.c:1101 ERROR dw_dma_get_data_size(): xrun detected
[11070119837.239584] (       20.364584) c0 dai          8.43           src/audio/dai.c:801  ERROR dai_report_xrun(): overrun due to no space available
[11070119857.239584] (       20.000000) c0 dai          8.43           src/audio/dai.c:704  ERROR comp_overrun(): sink->free = 768, copy_bytes = 0
[11070119946.718750] (       89.479164) c0 pipe         8.44      src/audio/pipeline.c:1044 ERROR pipeline_copy(): ret = -61, start->comp.id = 43, dir = 0
[11070119970.781250] (       24.062500) c0 pipe         8.44      src/audio/pipeline.c:1234 ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[11071125265.781250] (  1005295.000000) c0 dai          8.43     src/audio/component.c:209  ERROR comp_set_state(): wrong state = 1, COMP_TRIGGER_STOP
[11071125288.229168] (       22.447916) c0 pipe         8.44      src/audio/pipeline.c:899  ERROR pipeline_trigger(): ret = -22, host->comp.id = 41, cmd = 0
[11071125313.020834] (       24.791666) c0 ipc                       src/ipc/handler.c:443  ERROR ipc: comp 41 trigger 0x50000 failed -22
```

**dmesg**
```
kernel: [ 5534.855583] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
kernel: [ 5534.855975] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
kernel: [ 5534.855977] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 791060
kernel: [ 5534.855998] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 4 dir 1 cmd 1
kernel: [ 5534.856525] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg=0x2054001e successful
kernel: [ 5534.856528] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
kernel: [ 5534.856930] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
kernel: [ 5535.339094] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
kernel: [ 5535.339140] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
kernel: [ 5535.360414] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 4 dir 1 cmd 0
kernel: [ 5535.360433] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
kernel: [ 5535.360641] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x60050000 size 12
kernel: [ 5535.360670] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg=0x540000 successful
kernel: [ 5535.360681] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
kernel: [ 5535.360688]  DMIC: ASoC: trigger FE cmd: 0 failed: -22
kernel: [ 5535.360721] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 4 dir 1
kernel: [ 5535.360728] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
kernel: [ 5535.360987] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
kernel: [ 5535.361076] sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 4 dir 1
kernel: [ 5535.839097] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
kernel: [ 5535.839151] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
kernel: [ 5536.339095] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
kernel: [ 5536.339145] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
```

keyonjie, 2021-03-24:
looks similar to https://github.com/thesofproject/sof/issues/3939.

It will be good if it can be reproduced not too difficult. @singalsu

keyonjie, 2021-05-20:
@keqiaozhang @XiaoyunWu6666 can you help try if https://github.com/thesofproject/sof/pull/4221 help for this?

keqiaozhang, 2021-06-01:
@keyonjie , this issue can still be seen in yesterday's daily test, Test ID 4290.

keyonjie, 2021-06-01:
> @keyonjie , this issue can still be seen in yesterday's daily test, Test ID 4290.

Thanks, it disappear in today's 4300. So looks the issue is improved, but we can still reproduce it in a lower rate.

keyonjie, 2021-06-01:
@keqiaozhang @lgirdwood one more fix needed to address this issue:
https://github.com/thesofproject/sof/pull/4269

keyonjie, 2021-06-03:
@keqiaozhang @XiaoyunWu6666 looks this is fixed according to the latest daily result, please help to double check and close it if true.

keyonjie, 2021-06-16:
The issue happen again from yesterday, looks regression, reopen it.

plbossart, 2021-06-16:
> The issue happen again from yesterday, looks regression, reopen it.

This doesn't seem the right bug, the TGL_NOCODEC issue is related to an IPC error, not an xrun.

XiaoyunWu6666, 2021-06-18:
'Strangely' it disappeared in inner daily 4705

The only failed case for TGLU_RVP_NOCODEC is multiple-pause-resume-25.sh , which cannot be manually reproduced manually , however.

Is any commit that was merged yesterday suspicious ?

lgirdwood, 2021-06-30:
@XiaoyunWu6666 @keqiaozhang any update on this ? If it's still not showing then we could close ?

plbossart, 2021-06-30:
we removed the mixer and smart amp from the topology so made this test more robust, we should close and reopen an issue when we re-add the mixer.

keyonjie, 2021-07-01:
> we removed the mixer and smart amp from the topology so made this test more robust, we should close and reopen an issue when we re-add the mixer.

to be accurate, no mixer in sof-tgl-nocodec-ci.m4, the reason we don't see this in daily basically because we are testing a different tplg now.

XiaoyunWu6666, 2022-01-24:
haven't seen this bug in CI daily test for months

closed 2022-01-24 by XiaoyunWu6666