thesofproject/sof #3958: [BUG][BDW][HSW] QEMU boot test failed with GCC 10 for hsw bdw

fredoh9, 2021-03-25:
**Describe the bug**

After upgrading GCC 10 (https://github.com/thesofproject/sof/issues/3880), QEMU boot test failed for bdw, hsw.
Current QEMU version is QEMU 4.1.93.

**To Reproduce**
```
docker pull thesofproject/sof
#20210228 has GCC 10 toolchains
docker tag thesofproject/sof:20210228 sof

cd SOF_ROOT
./scripts/docker-run.sh ./scripts/xtensa-build-all.sh -a -r
#run QEMU boot test for all or specify bdw bsw
./scripts/docker-qemu.sh ../sof.git/scripts/qemu-check.sh
```

**Reproduction Rate**
100%

**Expected behavior**
QEMU boot test should be passed for all platforms

**Screenshots or console output**
[QEMU_boot_failure_bsw_bdw.txt](https://github.com/thesofproject/sof/files/6206569/QEMU_boot_failure_bsw_bdw.txt)

lgirdwood, 2021-03-26:
@fredoh9 can you regenerate docker SDK now. It was a bug in the xtensa overlay, now fixed.

fredoh9, 2021-03-26:
I will re-generate docker image and check again. If all works fine, I will switch to GCC 10 toolchains over this weekends.

fredoh9, 2021-03-26:
I generated new Docker image and am able to verify it. New label is pushed, `thesofproject/sof:20210326`.
I will switch this image to latest in this weekends.

fredoh9, 2021-03-29:
After upgrading the docker image, I closely monitored any error. It has worked fine.

closed 2021-03-29 by fredoh9