thesofproject/sof #3965: [BUG] [TGL] [tgl-012-drop-stable] Audio corruption and unexpected fault with codec adapter

RDharageswari, 2021-03-29:
**Describe the bug**

1. SOF crashes on boot with 'unexpected fault' when playback from speakers – as a result no audio (see dmesg output in unexpected_fault.txt).
To reproduce: reboot device and start playback, for example from youtube
Reproduction rate: 1/10.

2. Audio is corrupted when playback from speakers.
To reproduce: start playback on speakers, inject HP, eject HP, restore playback. Corrupted audio sounds like a bitrate mismatch.
Reproduction rate: close to 1
**To Reproduce**
Steps to repro:
1. Play video through youtube and check the playback
2. The playback might be
    a. correct
    b. white noise only
    c. no sound can be heard
3. pause playback and close lid for 30 seconds
4. open lid, go back to step 1, the playback would be either a or b or c

The probabilities of symptoms b and c are around:
b: 50%
c: 25%

uploading white noise sample video clip:
white noise came out since 12 second.

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Showstopper
blocks integration

**Environment**
This issue is not observed without codec adapter
This issue is seen while CONFIG_DUMMY_CODEC is enabled
The above observation and attached logs are with codec adapter + waves binary

3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL
    * Branch: https://github.com/thesofproject/sof/tree/tgl-012-drop-stable

**Screenshots or console output**
Please find attached
[unexpected_fault.txt](https://github.com/thesofproject/sof/files/6218594/unexpected_fault.txt)
[crash-sof-log.txt](https://github.com/thesofproject/sof/files/6218595/crash-sof-log.txt)
[dmesg_crash.log](https://github.com/thesofproject/sof/files/6218596/dmesg_crash.log)
[crash-dmesg-log.txt](https://github.com/thesofproject/sof/files/6218597/crash-dmesg-log.txt)
[corruption-sof-log.txt](https://github.com/thesofproject/sof/files/6218598/corruption-sof-log.txt)

https://user-images.githubusercontent.com/37420451/112774625-1bb90c80-8fef-11eb-9910-b73be3e1c4d5.MOV

mrajwa, 2021-03-29:
@RDharageswari, just to clarify, are steps 5,6 (pause + close/open lid ) necessary to trigger this issue?

mrajwa, 2021-03-29:
also, you mention `CONFIG_DUMMY_CODEC` while logs clearly shows  waves_codec_configure is being used. Missing info about the topology and cores utilized

keyonjie, 2021-03-29:
@RDharageswari are you using CONFIG_DUMMY_CODEC? can you submit a PR with the tplg you are using?

lancedkhou, 2021-03-29:
> @RDharageswari, just to clarify, are steps 5,6 (pause + close/open lid ) necessary to trigger this issue?

@mrajwa , steps 5/6 are not necessary.

When system booted into desktop, the very first audio playback could be:
a. correct/normal
b. white noise only (see enclosed video)
c. no sound can be heard

There is a chance to recover by: 
step 5. close lid for around 30 secs 
step 6. open lid and play again and hear the playback
it still could be a, b or c

mrajwa, 2021-03-29:
@lancedkhou, ok thanks. Could you please show the topology settings including scheduled cores?

dbaluta, 2021-03-29:
@RDharageswari this is very strange. So, you say that with CONFIG_DUMMY_CODEC=y this bug happens but with CONFIG_DUMMY_CODEC=n this bug doesn't happen?

I think the problem is because ``init_process`` function is missing from ``waves`` codec. It is also missing with dummy, I will fix that once @cujomalainey latest patches are in.

Can you please try this patch:

```
diff --git a/src/audio/codec_adapter/codec/waves.c b/src/audio/codec_adapter/codec/waves.c
index 497c81d9c..54a465ed2 100644
--- a/src/audio/codec_adapter/codec/waves.c
+++ b/src/audio/codec_adapter/codec/waves.c
@@ -648,6 +648,19 @@ int waves_codec_prepare(struct comp_dev *dev)
        return ret;
 }
 
+
+int waves_codec_process(struct comp_dev *dev)
+{
+       int ret;
+       struct codec_data *codec = comp_get_codec(dev);
+       struct waves_codec_data *waves_codec = codec->private;
+
+       codec->cpd.consumed = 0;
+       codec->cpd.init_done = 1;
+
+       return 0;
+}
+
 int waves_codec_process(struct comp_dev *dev)
 {
        int ret;
diff --git a/src/include/sof/audio/codec_adapter/interfaces.h b/src/include/sof/audio/codec_adapter/interfaces.h
index 4787ef4d7..b4a279c96 100644
--- a/src/include/sof/audio/codec_adapter/interfaces.h
+++ b/src/include/sof/audio/codec_adapter/interfaces.h
@@ -60,6 +60,7 @@ static struct codec_interface interfaces[] = {
                .id = WAVES_ID,
                .init  = waves_codec_init,
                .prepare = waves_codec_prepare,
+               .init_process = waves_codec_init_process,
                .process = waves_codec_process,
                .apply_config = waves_codec_apply_config,
                .reset = waves_codec_reset,

```

RDharageswari, 2021-03-29:
Sure, will try the attached patch.
There is an update from waves:
SOF crashes on boot with 'unexpected fault' when playback from speakers – as a result, no audio.
There is no Waves processing, only the dummy codec (passthrough with memcpy) added to the topology.
[sof-logger_crash.log](https://github.com/thesofproject/sof/files/6223792/sof-logger_crash.log)
[dmesg_crash.log](https://github.com/thesofproject/sof/files/6223794/dmesg_crash.log)
The reproduction rate can be 1/50.
Please see the logs attached.

dbaluta, 2021-03-29:
@RDharageswari thanks! Indeed the logs you provided do not contain anything related to 'waves' codec as opposed to first logs. So, I guess it the same problem reported above but related to dummy codec. 

Dummy was renamed to passthrough so depending on what sources you have please try this:

```
--- a/src/audio/codec_adapter/codec/passthrough.c
+++ b/src/audio/codec_adapter/codec/passthrough.c
@@ -36,6 +36,7 @@ int passthrough_codec_prepare(struct comp_dev *dev)
                return -ENOMEM;
        }
        codec->cpd.out_buff_size = cd->period_bytes;
+       codec->cpd.init_done = 1;
```

dbaluta, 2021-03-29:
@RDharageswari one more thing. Can you please attach the topology used (m4 and .tplg files). And also how to test. Are they using ALSA PCM interface or ALSA Compress interface?

I guess it's ALSA PCM as support for ALSA SOF compress is not merged yet.

RDharageswari, 2021-03-29:
[0001-dummy-codec.patch.txt](https://github.com/thesofproject/sof/files/6224296/0001-dummy-codec.patch.txt)
it is PCM interface

cujomalainey, 2021-03-29:
In reference to [comment](https://github.com/thesofproject/sof/issues/3965#issuecomment-809309209) from @dbaluta 

Is this really the issue? How did we land the requirement without updating the existing adapters? Also why are not failing on setup if the mandatory callback is not installed? Also those additional flags, why are trying to run corrupted rather than failing if they are not set?

dbaluta, 2021-03-30:
@cujomalainey Cadence codec doesn't need that callback installed yet as it is handled internally. For waves / passthrough the issue should be visible but I guess there was no test with the master. Will send a patch to fix the issue.

cujomalainey, 2021-03-30:
@dbaluta we are testing on tgl-drop-012 which doesn't have the patch to enable this yet so this callback is not the issue.

stolx, 2021-03-30:
@dbaluta @mrajwa @cujomalainey 

Update on the "audio corruption" part of the issue.
It is somehow related to `CONFIG_DEBUG` and `CONFIG_TRACEV`, it is easy to reproduce when those are off.

Steps to reproduce:
1. Use tgl-012-drop-stable + this commit https://github.com/stolx/sof-custom/commit/9e9b546a649994b7a277ba5084ed7d592018d22c
commit adds all necessary changes to reproduce the issue: topology, config + info log in the process function.
2. Build SOF using xt-xcc for defconfig `tgplp_defconfig` and override config `tigerlake_chrome`
3. Build topology sof-tgl-rt1011-rt5682.tplg
4. Install both 2 and 3 on the device
5. Launch audio `$aplay -Dhw:0,0 ./48_16_2_chirp.wav` (file that I am using is nothing special, 2 seconds sine sweep, 48000 Hz, 16 bit samples in 16 bit container, 2 channels, but I can share if needed)
6. Reproduction rate for me is roughly 1/10

Logs enclosed
[sof-log.txt](https://github.com/thesofproject/sof/files/6229507/sof-log.txt)
[dmesg-log.txt](https://github.com/thesofproject/sof/files/6229509/dmesg-log.txt)

cujomalainey, 2021-03-30:
Hmm, the fact that enabling debug changes things screams race condition in my head. Thanks for update. Will report findings from todays debugging.

cujomalainey, 2021-03-30:
I think I found something related to the distortion. https://github.com/thesofproject/sof/blob/main/src/audio/codec_adapter/codec_adapter.c#L496
This function is bailing without an error which is letting the codec_adapter partially prime even though they shouldn't. This at least causes it to fail right away. I am going to try and load the config with non-data to see if that silences it and allows things to work correctly.

cujomalainey, 2021-03-31:
Ignore above link, it fails the cmd, not prepare like i originally thought. Though @stolx I did find something interesting unique to waves and not to the passthrough. If you play 32Le to passthrough you get normal playback but waves reduces rate from 48k to 32k according to alsa conformance. Firmware also vomits a bunch of xrun errors. I have been trying to hunt down the issue but no luck yet so far.

oleggtitoff, 2021-03-31:
Hi all,
Could you please suggest what is the right way to use the sof-coredump-reader tool?
Using it we would like to get more info about this 'unexpected fault'.

Our steps:
1. reproduce 'unexpected fault' using aplay

2. cat /sys/kernel/debug/sof/exception > exception.bin
This command creates an exception.bin file, but there are errors:
stdout:
cat: /sys/kernel/debug/sof/exception: Invalid argument
dmesg:
[  293.108383] sof-audio-pci 0000:00:1f.3: error: debugfs entry cannot be read in DSP D3

3. hexdump -C exception.bin

4. ./sof-coredump-reader.py -v -i exception.bin --stdout
In the coredump output, registers are all zeros and:
exccause: IllegalInstructionCause: Illegal instruction

Please see the full output attached: 
[exception.txt](https://github.com/thesofproject/sof/files/6235717/exception.txt)

We assume that this is not the expected output. Please advise if we use this tool in a not proper way?

paulstelian97, 2021-03-31:
Try enabling

CONFIG_SND_SOC_SOF_DEBUG
CONFIG_SND_SOC_SOF_DEBUG_ENABLE_DEBUGFS_CACHE

on the kernel kconfig and rebuilding the kernel. This enables caching the exception so that it is saved and readable from the /sys/kernel/debug/sof/exception file even after the DSP turned off. You should be able to get some actual useful data rather than a file filled with zeros this way.

oleggtitoff, 2021-03-31:
Thank you, the kernel was updated with both options enabled.
Now there are no errors after `cat /sys/kernel/debug/sof/exception > exception.bin`

`hexdump` gives such output:
```
localhost ~ # hexdump -C exception.bin
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000800
```

But `./sof-coredump-reader.py -v -i exception.bin --stdout` still gives output full of zeros: [exception2.txt](https://github.com/thesofproject/sof/files/6237543/exception2.txt)
Please suggest?

cujomalainey, 2021-04-01:
This is the chrome os 5.4 fork we are working on

stolx, 2021-04-01:
> Note I cannot observe this when using only core 0.

same, when only core 0 is used everything works fine

juimonen, 2021-04-01:
@lyakh FYI you worked on issues with multicore...

mrajwa, 2021-04-06:
Today I have verified tgl-012-drop-stable with DOLBY DAP lib and sof-hda-generic topology on linux kernel, the results are as follows:

1. Parallel playback and capture using aplay and arecord respectively works OK (except barely audible, low amplitude white noise in the background)
2. Playback using YT also works fine, pause/resume too (low amp noise is also present + sporadic glitches occurred)

Results are the same no matter if codec adapter is on master or slave core.
I have never experienced any crash or no audio output issue.

@stolx, @RDharageswari can you check the topology I mentioned? I will dig deeper into the white noise issue.

mrajwa, 2021-04-06:
Update: as it turns out YT playback format on my ubuntu is S16_LE while the topology was set to S32_LE - after fixing that everything is correct. @stolx @RDharageswari please adjust the format in the topology.

mrajwa, 2021-04-07:
@stolx, to address the noise issue, I believe you need to update `waves_effect_config` in waves.c according to https://github.com/thesofproject/sof/pull/4008

lgirdwood, 2021-04-13:
@mrajwa @keyonjie can you give us an update on what has been ruled out and the current state and avenues of investigation ?

cujomalainey, 2021-04-13:
I'm going to be taking a look at how this adapter handles the buffers this week to see if there is an issue there.

lgirdwood, 2021-04-13:
@cujomalainey I have a suspicion that perhaps we are not producing/consuming data on cache line size and aligned chunks. @slawblauciak @keyonjie have we double checked this ? Something could have slipped in here that means we are not alined with cache.

mrajwa, 2021-04-13:
@lgirdwood I was only able to reproduce the noise issue with dolby codec I have. After applying the patch from https://github.com/thesofproject/sof/pull/4008 (will be updated further soon) the problem is gone. I also double checked with Keyon, we did a live debug session on friday testing post processing on multicore. We also tried to stress it a bit adding more parallel pipelines - no issue observed. On the other hand, @zrombel run full scope tests and the results reveled stream stalls on multicore scenario (note he uses slim driver for test) this issue is common for all features: DSM, post processing, etc.
Is there anybody else who noticed this issue on full stack device?

mengdonglin, 2021-07-20:
@RDharageswari Do you think we can close this issue?

closed 2022-01-18 by mengdonglin